MarlinFirmware/Marlin #3179: error: 'WAIT_FOR_USER' was not declared in this scope

esenapaj, 2016-03-19:
WAIT_FOR_USER added https://github.com/MarlinFirmware/Marlin/pull/3109,
but undefined.

error message:
AppData\Local\Temp\build6674025c445f92d0b07185e4e4080804.tmp\sketch\Marlin_main.cpp: In function 'void gcode_M600()':

Marlin_main.cpp:5574: error: 'WAIT_FOR_USER' was not declared in this scope

     KEEPALIVE_STATE(WAIT_FOR_USER);

                     ^

AppData\Local\Temp\build6674025c445f92d0b07185e4e4080804.tmp\sketch\Marlin_main.cpp:447:47: note: in definition of macro 'KEEPALIVE_STATE'

   #define KEEPALIVE_STATE(n) do{ busy_state = n; }while(0)

                                               ^

exit status 1
'WAIT_FOR_USER' was not declared in this scope

esenapaj, 2016-03-20:
This error occurs when you enable FILAMENTCHANGEENABLE in Configuration_adv.h.

AnHardt, 2016-03-20:
Replace not existing state `WAIT_FOR_USER` with existing state `PAUSED_FOR_USER`.

esenapaj, 2016-03-20:
Thanks for the fix, now error has gone away.

closed 2016-03-20 by thinkyhead