MarlinFirmware/Marlin #3193: Submit PRs a little faster with this HTML form

thinkyhead, 2016-03-20:
It's a hassle to submit PRs to `RCBugFix` – or really any repository branch that isn't set as the default. For this reason we get a lot of PRs submitted for `RC` that need to be closed and redone. Wouldn't it be nice if you could choose any default target for your Pull Requests and have it stick?

In the meantime, I've built this workaround, which you can try out in your web browser. If you (like me) have a custom `localhost` start-page set in your browser, add these forms to make it easier and faster to submit a PR to Marlin. These can of course be edited as we get into fixing bugs on the release version.

[Github PR Shortcut Form githubmf.tar.gz](https://github.com/MarlinFirmware/Marlin/files/181551/githubmf.tar.gz)

It would be cool if somehow this kind of form could be hosted right on Github, inside the README.md page perhaps, or if we could somehow automatically pick up the Github user name in the form. **In the meantime you must edit the `index.html` file and put in your Github org/username before using this.**

As an alternative to the above, this HTML may be edited to add a field for the Github user name (hiding or minimizing the field once it has the information). Cookies are also an option for persistent storage, but browsers all now do autofill, so that might be redundant.

The whole crux of this HTML form is the javascript:
```js
var GITHUB_USER = "you_yes_you";
var MARLIN_ORG_HOME = "https://github.com/MarlinFirmware";
var loc = MARLIN_ORG_HOME + "/Marlin/compare/RCBugFix..." + GITHUB_USER + ":" + branchname + "?expand=1";
window.open(loc, '_blank');
```
If you can find other ways to get the same result, make a widget or a button or an accessory panel… please do! It will save a lot of time making PRs, especially if you make a lot of them.

Roxy-3DPrintBoard, 2016-03-21:
Oh?   I don't have to learn Git to do a Pull Request????

jbrazio, 2016-03-21:
You still do, knowing C++ is also helpful. :-P
This just speeds up a bit Github interface by pre-selecting the target Marlin beach for you.

thinkyhead, 2016-03-27:
Come to think of it, so much my work with `git` is in the terminal —and many of us need to do these tasks all the time— so I made a few BASH scripts specifically to help with Marlin. Described at the top ^^

thinkyhead, 2016-03-27:
> I don't have to learn Git to do a Pull Request?

Not really. But hey, `git` is super-easy once you get the gist of it.

closed 2016-05-04 by jbrazio