MarlinFirmware/Marlin #3200: Please add X2_MAX_POS to EEPROM (Dual X Carriage)

pbrgle1, 2016-03-21:
Sometimes i have to change the distance between the extruders by a fraction of a mm and it would be really helpful if this could be done by saving to EEPROM.

configuration_adv.h

#define X2_MAX_POS 257.4

jbrazio, 2016-03-25:
Why do you have to frequently adjust a value that should be a mechanical parameter of the machine ?

pbrgle1, 2016-03-25:
It's only +/- 0.2mm but enough to cause problems. Probably due to expanding and contracting of the rods and frame.

thinkyhead, 2016-03-26:
Are you using GT2 belts and pulleys? Are they over- or under-constrained? Some adjustment there can really help.

If you're doing dual-color, I think possibly the `BABYSTEPPING` options could help. You would want to add some extra GCode to move the two carriages to the same spot, one after the other. When the second carriage is active, you would babystep the X axis to get it to the right spot. Then you can start your print and the axis will continue to stay aligned.

pbrgle1, 2016-03-27:
Thanks i'll give it a try. But still some (semi)permanent solution like EEPROM setting would be nice. Once the values are set they last for quite some time.

thinkyhead, 2016-03-27:
It sounds like what is really needed is not to artificially fix this by hacking `X2_MAX_POS`, but to provide an option for "fine-tuning" the offset between the carriages. We will have to get into this more after the current release. Features like this are becoming more important as dual printing is getting more popular all the time.

closed 2016-05-04 by jbrazio