MarlinFirmware/Marlin #3205: Using FAN_KICKSTART_TIME (RCBugFix) won't let control fans anymore

Alex9779, 2016-03-22:
Hello,
I transferred the config of my machine to the current RCBugFix.
All seemed fine but doing my first print got me into a thermal runaway error.
I realised that the fans have been running at 100% though I set they only to 10% at that stage.
A little (almost 3 hours now) investigation showed that using FAN_KICKSTART_TIME = 100 is the reason for this.
If this is active I cannot control the cooling fans anymore. They are either off or at 100%.
I digged into the code but I am puzzled. I read the corresponding sections over and over again.
The last changes to that have been made in 4bbea51 but I don't see a problem with the new code, the changes regarding multiple PWM fans seem correct...

AnHardt, 2016-03-22:
Bug confirmed. I see the same here.

thinkyhead, 2016-03-24:
I was wondering about this one too. I found that if I turned the fan off, I could regain control. Anyway, looks like it is patched. Thanks for the report!

closed 2016-03-24 by thinkyhead