MarlinFirmware/Marlin #3211: extruder heater never stabilizes at set temp

mramia88, 2016-03-22:
I just finished the hardware of my printer and uploaded Marlin RC 1.1 I am not able to get extruder heaters to stabilize, I have checked wiring, board and connection. Everything seems ok but no matter what PID settings I update heaters never stabilize it usually fluctuates +/- 10C. I have used the same board, Power supply and extruders previously with Marlin successfully. 24V 40W heaters below are my settings in Conf H 

#define TEMP_SENSOR_0 5
#define TEMP_SENSOR_1 5
#define TEMP_SENSOR_2 0
#define TEMP_SENSOR_3 0
#define TEMP_SENSOR_BED 1

// This makes temp sensor 1 a redundant sensor for sensor 0. If the temperatures difference between these sensors is to high the print will be aborted.
//#define TEMP_SENSOR_1_AS_REDUNDANT
#define MAX_REDUNDANT_TEMP_SENSOR_DIFF 10

// Actual temperature must be close to target for this long before M109 returns success
#define TEMP_RESIDENCY_TIME 3  // (seconds)
#define TEMP_HYSTERESIS 3       // (degC) range of +/- temperatures considered "close" to the target one
#define TEMP_WINDOW     1       // (degC) Window around target to start the residency timer x degC early.

// Comment the following line to disable PID and enable bang-bang.
#define PIDTEMP
#define BANG_MAX  // limits current to nozzle while in bang-bang mode; 255=full current
#define PID_MAX BANG_MAX 200// limits current to nozzle while PID is active (see PID_FUNCTIONAL_RANGE below); 255=full current
#if ENABLED(PIDTEMP)
  //#define PID_DEBUG // Sends debug data to the serial port.
  //#define PID_OPENLOOP 1 // Puts PID in open loop. M104/M140 sets the output power from 0 to PID_MAX
  //#define SLOW_PWM_HEATERS // PWM with very low frequency (roughly 0.125Hz=8s) and minimum state time of approximately 1s useful for heaters driven by a relay
  //#define PID_PARAMS_PER_EXTRUDER // Uses separate PID parameters for each extruder (useful for mismatched extruders)
                                    // Set/get with gcode: M301 E[extruder number, 0-2]
  #define PID_FUNCTIONAL_RANGE 10 // If the temperature difference between the target temperature and the actual temperature
                                  // is more then PID_FUNCTIONAL_RANGE then the PID will be shut off and the heater will be set to min/max.
  #define PID_INTEGRAL_DRIVE_MAX PID_MAX  //limit for the integral term
  #define K1 .95 //smoothing factor within the PID

  // If you are using a pre-configured hotend then you can use one of the value sets by uncommenting it
  // Ultimaker
  #define  DEFAULT_Kp 6.54
  #define  DEFAULT_Ki .15
  #define  DEFAULT_Kd 73.83

thinkyhead, 2016-03-22:
Have you used `M303` to set the PID of your hot end? You might try running `M303` with a temperature target closer to what you normally use. (150°C is the default.)

jbrazio, 2016-03-22:
Try something like `M303 E0 C10 S210` (where *210* is the target temperature).

mramia88, 2016-03-22:
I have, there is no change. I have tried lowering Current from 130 to 255. M303 ranges +/- 10 as well and never closes the gap

jbrazio, 2016-03-22:
It seems you have an error on the config file you have pasted, please try the following:
```cpp
#define BANG_MAX 255 // limits current to nozzle while in bang-bang mode; 255=full current
#define PID_MAX BANG_MAX // limits current to nozzle while PID is active (see PID_FUNCTIONAL_RANGE below); 
```

If that does not work, try increasing `PID_FUNCTIONAL_RANGE` to 50.

mramia88, 2016-03-22:
I had bang max 255 and it didnt work either. Ill try changing the range to 50

jbrazio, 2016-03-22:
You should define a value to `BANG_MAX` don't leave it like you have on the pasted config file.

mramia88, 2016-03-22:
I had it defined from 130-255 none of the values worked

mramia88, 2016-03-22:
Just changed the PID_FUNCTIONAL_RANGE to 50 that seems to have done it! thank much

mramia88, 2016-03-22:
its still fluctuates a bit on the lower side. drops to 187 then up to 191 from set 190.  should I do a  M303

jbrazio, 2016-03-22:
Yes please, do the fine tuning of the PID iwth `M303`.

mramia88, 2016-03-22:
will do! thank you for all your help.

jbrazio, 2016-03-22:
You're welcome. If you feel the problem is solved please close the issue.

closed 2016-03-22 by mramia88

Justinjhcs1, 2016-08-22:
I have temp problems too I think every RAMPS board is different depending on the Moss Used to control the Heater I have set my HEATER_CURRENT =35  It was 200 and I was getting Temps as High as 300 C so every Moss is different I guess this is why there even is a Current limiter to the Extruder doing a Temp test now Current at 65 only going as high as 200C for PLA and it is cycling now Possibly too much Current my other on it is set to 255 and still it heats slow and this is with a 500Watt Power supply.... I pushed the Input voltage as high as 13.58VDC I dare not sett it higher i did that with another ramps and Burned the Board maybe I have the input Voltage set too high too the VDC was as hight as 14VDC and still the base was not getting passed 108C I could not do ABS or HIPS or even PETG I was stuck with PLA that is Ok if your making parts for a PM VE33 ( this is what I am calling it because of the Various Changes I have made to a Basic Persa Mendel.)

thinkyhead, 2016-08-22:
Good luck. These machines are always an adventure.