MarlinFirmware/Marlin #3233: RCBugFix testing Auto bed leveling - new offset in not processed

paulusjacobus, 2016-03-24:
In RCbugfix release I noticed that my M851 offsets were not taken into account. Only when I do a hard reset (press the button on my mega) then the offset is taken on board.

Prusa I2, Ramps1.4 Mega board, fixed z probe (proximity sensor).

Sequence: G28, G29, Print a object, adjust the z offset M851 Z-0.33, M500, M503, G28, G29, Print the object again (z gap remains the same...)

It's a minor bug since the user has a work around with the hardware reset

jbrazio, 2016-03-24:
You should to a `M501`.. so the correct sequence is:
```
G28 ; home
G29 ; z-probe
G1 Z0 ; or whatever you want to do
M851 Z-0.33
M500 ; save EEPROM
M501 ; reload EEPROM
G28 ; test the new offset
G29 
```

paulusjacobus, 2016-03-25:
@jbrazio Sorry I feel very stupid but I think you're right (I always relied on M503). Will close this issue.

closed 2016-03-25 by paulusjacobus

thinkyhead, 2016-03-27:
`M500` followed by `M501` should have no effect. You're only reading back the settings just written. If you require `M501` to fix an issue, then there's still something wrong.

jbrazio, 2016-03-27:
I always thought `M501` reads the EEPROM and overrides the [not saved] running settings.

paulusjacobus, 2016-03-27:
I noticed the offset issue again. A hardware reset does fix it. Will re-test this over the next weekend.

thinkyhead, 2016-03-28:
@jbrazio That is correct. But M500 writes the current settings (in RAM) to EEPROM, so M501 immediately after M500 only reads back what you just wrote.

thinkyhead, 2016-03-28:
With recent bugs I'm keeping in mind that variables, the stack, and memory in general may be getting corrupted by bad pointers or buffer overruns. These might be caused by code bugs or by compiler bugs. So tracking down things that make the machine crash or act weird is a big priority. Please let us know of anything strange, no matter how small.

thinkyhead, 2016-03-28:
Hmm, referring back to the original issue – You must do `G29` after `M851` or you get no benefit. I assume that's what you are doing – `M851` followed by `G29` – but that when you do that `G29` the offset is not being applied correctly…?