MarlinFirmware/Marlin #3234: Automatically turn the Fan on when Extruder is on

Grogyan, 2016-03-25:
I was going to suggest this as a feature, however I found that the feature is in configuration_adv

I sometimes forget to turn on the fan before I turn on the extruder, as I am now using the fan port on the RAMPS 1.4 in EFB mode.
With enabling this feature within the configuration_adv.h file, the fan is not turning on as expected when the hot end is over the EXTRUDER_AUTO_FAN_TEMPERATURE threshold, which is typically at 50 degrees C
If I set EXTRUDER_0_AUTO_FAN_PIN to FAN_PIN, the sanity check prevents this as a possibility.
Maybe i'm not using this feature correctly.

If I set the PLA_PREHEAT_FAN_SPEED to say, 200, in configuration.h, this only turns the fan on from a smart display like the Full Graphics Display.
So if I turn the Extruder on in Repetier Host, this doesn't turn the fan on automatically either.

Currently the ideal scenario is for the fan to automatically turn on when the hot end is above a certain temperature (eg 50 degrees), and stays on until the temperature drops below this threshold.
Even if the extruder is turned on with Repetier/OctoPrint/Smart Display.

Currently using RC4
RAMPS 1.4 EFB
Full Graphic Display

jbrazio, 2016-03-25:
Which fan are you talking about ? There are two types of fans: the fan which should start at 50C and aims to cool down the cold part of the hotend, and the blower fan which you start using GCODE.

Grogyan, 2016-03-25:
I do not at the moment have a blower to cool the part as it is being printed.

I will in the future will have both the blower and the active extruder cooling fan on the same Fan port on the RAMPS board

jbrazio, 2016-03-25:
So what you want to accomplish is to activate the hotend-fan when the temperature of the extruder is above 50C right ?

Grogyan, 2016-03-25:
Correct

jbrazio, 2016-03-25:
Just edit `Configuration_adv.h`:

```cpp
// Extruder cooling fans
// Configure fan pin outputs to automatically turn on/off when the associated
// extruder temperature is above/below EXTRUDER_AUTO_FAN_TEMPERATURE.
// Multiple extruders can be assigned to the same pin in which case
// the fan will turn on when any selected extruder is above the threshold.
#define EXTRUDER_0_AUTO_FAN_PIN 11
#define EXTRUDER_1_AUTO_FAN_PIN -1
#define EXTRUDER_2_AUTO_FAN_PIN -1
#define EXTRUDER_3_AUTO_FAN_PIN -1
#define EXTRUDER_AUTO_FAN_TEMPERATURE 50
#define EXTRUDER_AUTO_FAN_SPEED   255  // == full speed
```

If your board only has one FAN port then you must `#undef FAN_PIN`.

In a nutshell `FAN_PIN` is the blower fan, `EXTRUDER_X_AUTO_FAN_PIN` is the cooling fan for cold part of the hotend.

Grogyan, 2016-03-25:
That didn't work

jbrazio, 2016-03-25:
Did you disabled `FAN_PIN` ?
What you cannot do is to have `FAN_PIN` assigned **to the same pin** as `EXTRUDER_0_AUTO_FAN_PIN`.

Grogyan, 2016-03-25:
I disabled FAN_PIN, however this produced other errors with compiling

jbrazio, 2016-03-25:
Try this `#define FAN_PIN -1`.

Grogyan, 2016-03-25:
Did that, compiling error
`exit status 1
'fanSpeeds' was not declared in this scope`

jbrazio, 2016-03-25:
Where are you doing it ?

Grogyan, 2016-03-25:
In
pin_RAMPS_14.h

jbrazio, 2016-03-25:
`pins_RAMPS_14.h` has the following defenition about `FAN_PIN`

``` cpp
#if MB(RAMPS_14_EFF) || MB(RAMPS_13_EFF) || ENABLED(IS_RAMPS_EFB)
  #define FAN_PIN           9 // (Sprinter config)
  #if MB(RAMPS_14_EFF) || MB(RAMPS_13_EFF)
    #define CONTROLLERFAN_PIN  -1 // Pin used for the fan to cool controller
  #endif
#elif MB(RAMPS_14_EEF) || MB(RAMPS_14_SF) || MB(RAMPS_13_EEF) || MB(RAMPS_13_SF)
  #define FAN_PIN           8
#else
  #define FAN_PIN           4 // IO pin. Buffer needed
#endif
```

At the end of the file could you please try:
``` cpp
#undef FAN_PIN
#define FAN_PIN -1
```

Grogyan, 2016-03-25:
Same error
`
In file included from sketch\ultralcd.cpp:55:0:

sketch\dogm_lcd_implementation.h: In function 'void lcd_implementation_status_screen()':

dogm_lcd_implementation.h:286: error: 'fanSpeeds' was not declared in this scope

   u8g.drawBitmapP(9,1,STATUS_SCREENBYTEWIDTH,STATUS_SCREENHEIGHT, (blink % 2) && fanSpeeds[0] ? status_screen0_bmp : status_screen1_bmp);

                                                                                  ^

exit status 1
'fanSpeeds' was not declared in this scope
`

jbrazio, 2016-03-25:
Could you please attach here you configuration (.h and adv.h) files ?

Grogyan, 2016-03-25:
Configuration.h
http://pastebin.com/HWpG0FUx

Configuration_ADV.h
http://pastebin.com/BAe7dSqj

jbrazio, 2016-03-25:
This is a bug.

Please try if this fixes your error.
 1. Download my cfe514e33bdb436ed02b1f143cc7d05a67bd296f branch from [here](https://github.com/jbrazio/Marlin/archive/bugfix/dogm-lcd-nofan-error.zip).
 2. Copy your configuration files
 3. In `Configuration_adv.h` add `#define EXTRUDER_0_AUTO_FAN_PIN 9`
 4. Edit `pins_RAMPS_14_EFB.h` and add the following to the end:

```cpp
#undef FAN_PIN
#define FAN_PIN -1
```

Please report if it fixed your problem.

thinkyhead, 2016-03-26:
Patched!

closed 2016-03-26 by thinkyhead

Grogyan, 2016-03-26:
I will probably be testing this tomorrow.

Cheers

cemalgultekin, 2019-06-02:
I am trying to do the same thing with my board with firmware 2.0.bugfix. 
On Configuration.h I set motherboard to

	#ifndef MOTHERBOARD
	  #define MOTHERBOARD BOARD_RAMPS_14_EFB
	#endif

On my Configuration_av.h, I set FAN_PIN to -1, 

	#define FAN_PIN -1
	#define E0_AUTO_FAN_PIN 9
	#define E1_AUTO_FAN_PIN -1
	#define E2_AUTO_FAN_PIN -1
	#define E3_AUTO_FAN_PIN -1
	#define E4_AUTO_FAN_PIN -1
	#define E5_AUTO_FAN_PIN -1
	#define CHAMBER_AUTO_FAN_PIN -1
	#define EXTRUDER_AUTO_FAN_TEMPERATURE 50
	#define EXTRUDER_AUTO_FAN_SPEED 255   // 255 == full speed

and the result is 

	sketch\src\module\motion.cpp: In function 'get_cartesian_from_steppers':

	sketch\src\module\motion.cpp:253:1: internal compiler error: Segmentation fault

	 }

	 ^

	Please submit a full bug report,

	with preprocessed source if appropriate.

	See <http://gcc.gnu.org/bugs.html> for instructions.

	lto-wrapper.exe: fatal error: C:\Program Files (x86)\Arduino\hardware\tools\avr/bin/avr-gcc returned 1 exit status

	compilation terminated.

	c:/program files (x86)/arduino/hardware/tools/avr/bin/../lib/gcc/avr/5.4.0/../../../../avr/bin/ld.exe: error: lto-wrapper failed

	collect2.exe: error: ld returned 1 exit status

	exit status 1
	Error compiling for board Arduino/Genuino Mega or Mega 2560.