MarlinFirmware/Marlin #3237: Z axis always tries to home outside of bed.

thjubeck, 2016-03-25:
Hey, I have a prusa i3 with ramps 1.4 and the z axis always tries to home outside of the bed when using a G28 command but if you home each axis separate, it says not in safe position.

jbrazio, 2016-03-25:
Are you using the latest RC ?
Have you enabled SAFE Z HOMING ?
Are your restrictions to SAFE Z HOMING correctly defined ?

AnHardt, 2016-03-25:
Do you have one of these:
```
#define DISABLE_X true
#define DISABLE_Y true
```
#define DISABLE_X false
#define DISABLE_Y false
```
?

AnHardt, 2016-03-25:
Do you have configured a probe, but not AUTO_BED_LEVELING_FEATURE?

thjubeck, 2016-03-25:
#ifndef CONFIGURATION_H
#define CONFIGURATION_H

#include "boards.h"

// This configuration file contains the basic settings.
// Advanced settings can be found in Configuration_adv.h
// BASIC SETTINGS: select your board type, temperature sensor type, axis scaling, and endstop configuration

//===========================================================================
//============================= DELTA Printer ===============================
//===========================================================================
// For a Delta printer replace the configuration files with the files in the
// example_configurations/delta directory.
//

//===========================================================================
//============================= SCARA Printer ===============================
//===========================================================================
// For a Delta printer replace the configuration files with the files in the
// example_configurations/SCARA directory.
//

// User-specified version info of this build to display in [Pronterface, etc] terminal window during
// startup. Implementation of an idea by Prof Braino to inform user that any changes made to this
// build by the user have been successfully uploaded into firmware.

//#define STRING_VERSION "1.0.2"

#define STRING_VERSION_CONFIG_H __DATE__ " " __TIME__ // build date and time
#define STRING_CONFIG_H_AUTHOR "(none, default config)" // Who made the changes.

// SERIAL_PORT selects which serial port should be used for communication with the host.
// This allows the connection of wireless adapters (for instance) to non-default port pins.
// Serial port 0 is still used by the Arduino bootloader regardless of this setting.
#define SERIAL_PORT 0

// This determines the communication speed of the printer
#define BAUDRATE 115200

// This enables the serial port associated to the Bluetooth interface
//#define BTENABLED              // Enable BT interface on AT90USB devices

// The following define selects which electronics board you have.
// Please choose the name from boards.h that matches your setup
#ifndef MOTHERBOARD
  #define MOTHERBOARD BOARD_RAMPS_13_EFB
#endif

// Define this to set a custom name for your generic Mendel,
// #define CUSTOM_MENDEL_NAME "This Mendel"

// Define this to set a unique identifier for this printer, (Used by some programs to differentiate between machines)
// You can use an online service to generate a random UUID. (eg http://www.uuidgenerator.net/version4)
// #define MACHINE_UUID "00000000-0000-0000-0000-000000000000"

// This defines the number of extruders
#define EXTRUDERS 1

//// The following define selects which power supply you have. Please choose the one that matches your setup
// 1 = ATX
// 2 = X-Box 360 203Watts (the blue wire connected to PS_ON and the red wire to VCC)

#define POWER_SUPPLY 1

// Define this to have the electronics keep the power supply off on startup. If you don't know what this is leave it.
// #define PS_DEFAULT_OFF

//===========================================================================
//=============================Thermal Settings  ============================
[... 684 lines cut ...]
 * 81 - Printrboard - Uses Analog input 2 on the Exp1 connector (version B,C,D,E)
 * 301 - Rambo  - uses Analog input 3
 * Note may require analog pins to be defined for different motherboards
 **********************************************************************/
// Uncomment below to enable
//#define FILAMENT_SENSOR

#define FILAMENT_SENSOR_EXTRUDER_NUM	0  //The number of the extruder that has the filament sensor (0,1,2)
#define MEASUREMENT_DELAY_CM			14  //measurement delay in cm.  This is the distance from filament sensor to middle of barrel

#define DEFAULT_NOMINAL_FILAMENT_DIA  3.0  //Enter the diameter (in mm) of the filament generally used (3.0 mm or 1.75 mm) - this is then used in the slicer software.  Used for sensor reading validation
#define MEASURED_UPPER_LIMIT          3.30  //upper limit factor used for sensor reading validation in mm
#define MEASURED_LOWER_LIMIT          1.90  //lower limit factor for sensor reading validation in mm
#define MAX_MEASUREMENT_DELAY			20  //delay buffer size in bytes (1 byte = 1cm)- limits maximum measurement delay allowable (must be larger than MEASUREMENT_DELAY_CM  and lower number saves RAM)

//defines used in the code
#define DEFAULT_MEASURED_FILAMENT_DIA  DEFAULT_NOMINAL_FILAMENT_DIA  //set measured to nominal initially 

//When using an LCD, uncomment the line below to display the Filament sensor data on the last line instead of status.  Status will appear for 5 sec.
//#define FILAMENT_LCD_DISPLAY

#include "Configuration_adv.h"
#include "thermistortables.h"

#endif //__CONFIGURATION_H

thjubeck, 2016-03-25:
I used the Marlin-1.0.2-1 version and it works correctly so there is some difference between them that is causing issues.

Blue-Marlin, 2016-03-25:
> //#define STRING_VERSION "1.0.2"

That's not a configuration what can run with a recent RC.
There are huge differences. It's like driving with a 30 years old map.

thjubeck, 2016-03-25:
Sorry posted my config working with 1.0.2. Reposted newer config.

thinkyhead, 2016-03-26:
It's hard to glean much from the settings alone. But there is more available to help debug the issue.
- Download the latest `RCBugFix` code.
- Enable the `DEBUG_LEVELING_FEATURE` option in your config. This provides detailed output during leveling (and some homing).
- Before you do any leveling, use the command `M115 S32` to enable the extra debug output.

Test leveling, being careful not to hurt your machine. Collect the output from the console log in Pronterface (or other favorite host software) and paste it in a message here so we can see what the machine "thinks" is going on.

thinkyhead, 2016-03-26:
> Reposted newer config.

I took the liberty to edit it down to just the relevant options. The only oddity I can see is that you aren't setting `MANUAL_HOME_POSITIONS` but still setting `MANUAL_[XYZ]_HOME_POS`. To be on the safe side use the intended method instead:
```cpp
#if ENABLED(MANUAL_HOME_POSITIONS)
  #define MANUAL_X_HOME_POS 0
  #define MANUAL_Y_HOME_POS 0
  #define MANUAL_Z_HOME_POS 0
  //#define MANUAL_Z_HOME_POS 402 // For delta: Distance between nozzle and print surface after homing.
#endif
```

thjubeck, 2016-03-26:
Commenting this out works. Now to get the bed dimmensions right. Always something. Thanks for the help.
#define DISABLE_MAX_ENDSTOPS

thjubeck, 2016-03-26:
Un-commenting that is

Roxy-3DPrintBoard, 2016-03-26:
> One other thing that could be affecting it is, you have no max endstops – so you need to enable:
~~~~
#define DISABLE_MAX_ENDSTOPS
~~~~

@ThinkyHead   I'm confused on this.   It never used to be important to enable the DISABLE_MAX_ENDSTOPS.   Did something change where this needs to be set 'correctly' now?    I thought that just turned off the pull up resistors in the AVR chip ???

thinkyhead, 2016-03-27:
#Roxy-3DPrintBoard Actually, this is one of those issues that has always popped up. We see it more often around new releases because that's when people mess with their configurations, and they always forget to enable this setting when they have no max endstops. The default configuration file sets neither `DISABLE_MIN_ENDSTOPS` nor `DISABLE_MAX_ENDSTOPS`, the reasoning being that min and max endstops are both quite common, so the default is not presumptuous. It probably makes the most sense to enable `DISABLE_MAX_ENDSTOPS` by default, since the default config is otherwise targeted to a cartesian with min endstops (plus, a wrong setting in that case would be more obvious).

All that said, #3065 eliminates these settings completely and I think its approach should be adopted to replace the old system. (In that PR `SanityCheck.h` informs users of the new situation.)

closed 2016-04-04 by thinkyhead