MarlinFirmware/Marlin #3238: [RC4] Rigidbot Panel can't print from USB

agalpin, 2016-03-25:
Using RC4 and the Rigidbot LCD/Panel I am unable to print from USB.

I can navigate the menus look at the files on the USB but when selecting one of the prints it  doesn't start.
I am however able to use OctoPrint to print from the USB with the same file. 
I have an idea that it's just not triggering the load/print when the button is pressed, as everything else is as expected, as it does return me to the info screen.

thinkyhead, 2016-03-25:
Print from USB or print from SD Card?

agalpin, 2016-03-25:
The RigidPanel has a USB port instead of a SD Slot that comes standard with most other LCD Panels.

thinkyhead, 2016-03-25:
Ah, I see! Interestingly enough, I am now unable to connect via USB and Pronterface either. So, there you have it – something is conflicting that I forgot to test.

thinkyhead, 2016-03-25:
For reference, what is your `BAUDRATE`? Have you tried both 115200 and 250000?

agalpin, 2016-03-25:
just run at 115200, i can give other a try tomorrow if that would help?

thinkyhead, 2016-03-25:
Yeah, check those rates. In my own case, I had decided to try the slower baud rate but I just forgot to tell Pronterface. I believe the new default might be 115200, but that shouldn't matter if you're using your existing config files. So… I looked into recent changes but couldn't find anything other than that one – and it just happened to trip me up when I updated and installed `RC4` just now.

thinkyhead, 2016-03-25:
Still looking at recent changes–– nothing jumps out yet.

agalpin, 2016-03-25:
There's a good chance it never worked, i don't use the USB anymore after moving to OctoPrint. I was asked if it all worked so decided to actually check it and the printing from USB didn't work.

agalpin, 2016-03-26:
https://github.com/MarlinFirmware/Marlin/issues/3245 is also this issue, print filename had 20 at the front @thinkyhead

jbrazio, 2016-03-26:
As a patch has been submitted, do you confirm this issue is closed ?

agalpin, 2016-03-26:
I am still printing but yes, i confirm that it is fixed :+1:

jbrazio, 2016-03-26:
Great, do you mind closing the issue ? Thanks for your help sorting this one out.

closed 2016-03-26 by agalpin