MarlinFirmware/Marlin #3245: [RC4] Some gcode files can't print from SD Card

ted321, 2016-03-25:
So I compared the difference.I found out if manually change the gcode`M104 S200` to `M104 S200 T0` it works fine.
Is this a bug or something.

jbrazio, 2016-03-25:
You have to be a bit more specific..
What happens when you have `M104 S200` ?

ted321, 2016-03-25:
` M104 S200 `Nothing happened,the printer won't start.

jbrazio, 2016-03-25:
Could you please have a serial connection to the printer when initiating the SD print so we can see what Marlin outputs ?

ted321, 2016-03-25:
Sorry, I want to help, but I don't know how to get Marlin outputs.

jbrazio, 2016-03-25:
On the Arduino IDE just click on Serial Monitor, and then operate the printer has you would; Marlin will "say" stuff to that monitor window.

ted321, 2016-03-25:
I'm really sorry about the confuse.It has nothing to do with the M104 command.It's the file name.
The file is "10.gcode", marlin says "open failed, File: c88~1.gco"
I changed the file name to "abcdef.gcode" it works.

ted321, 2016-03-25:
Seems Marlin do not recognizea filename start with numbers。Such as "8.gcode" "8abc.gcode" "10.gcode" "10abc.gcode" etc..

Roxy-3DPrintBoard, 2016-03-25:
Just to be clear....   You can't select the file off of the SD-Memory card using the LCD Panel, right?  Because looking at the M23 code, it looks like you should be able to select the file using the USB link from PronterFace.

Can you try selecting the file using PronterFace?

thinkyhead, 2016-03-25:
We're looking at a similar (maybe the same) issue at #3229. To help me understand what's going on, can you tell me: What version of Arduino are you using to build Marlin?

thinkyhead, 2016-03-25:
Try this: In `Marlin_main.cpp` at around line 5985, change this:
```cpp
while (*current_command_args == ' ' || (*current_command_args >= '0' && *current_command_args <= '9')) ++current_command_args;
```
…to this…
```cpp
while (*current_command_args >= '0' && *current_command_args <= '9') ++current_command_args;
while (*current_command_args == ' ') ++current_command_args;
```
And see if that helps the issue.

thinkyhead, 2016-03-25:
Ok, #3246 should soon be merged. It addresses the issue and improves things generally in that part of the GCode parser.

thinkyhead, 2016-03-25:
Give `RCBugFix` a try when you have a chance. It now has this patched.

ted321, 2016-03-26:
#3246 fixed the issue.Thank you for all your hard work.

jbrazio, 2016-03-26:
@ted321 thank you for helping sitting this bug out, do you mind closing the issue ? Thanks.

commit f423716c (2016-03-25): Fix: current_command_args skips digits at the front

Closes #3245
closed 2016-03-26 by ted321