MarlinFirmware/Marlin #3247: [RC4] Marlin RC4 doesn't support simple mechanical Z Probe?

esenapaj, 2016-03-25:
for example:
[ATOM 2.0 Delta 3D Printer - Official Site](http://www.atom3dp.com/en/atom2/)
[MICROMAKE 3D打印機DIY學習套件 注塑版 三角洲 並聯臂 Kossel-淘宝网全球站] (http://world.taobao.com/item/42916612391.htm) ... my printer
[Effector for Kossel mini with Z probe by wkkpatrick - Thingiverse](http://www.thingiverse.com/thing:820768)
[Detachable Z-probe w. magnetic connection for delta printer by robkar - Thingiverse](http://www.thingiverse.com/thing:892398)

Marlin began to force that we select classification of probes,
since commit 1cc316f1fdff9b93587e48a69b595dc50e34c5bc(PR #3082).
But these simple mechanical probes aren't sled, servo, inductive sensor, or the Allen key.

If I had to choose simple probe resembles the Allen key, but it doesn't needs deploys and retracts.
Therefore if I use simple probe as the Allen key(#define Z_PROBE_ALLEN_KEY),
it causes retract error in stow_z_probe() in Marlin_main.c.

```
      #if ENABLED(Z_MIN_PROBE_ENDSTOP)
        bool z_probe_endstop = (READ(Z_MIN_PROBE_PIN) != Z_MIN_PROBE_ENDSTOP_INVERTING);
        if (!z_probe_endstop)
      #else
        bool z_min_endstop = (READ(Z_MIN_PIN) != Z_MIN_ENDSTOP_INVERTING);
        if (!z_min_endstop)
      #endif
        {
          if (IsRunning()) {
            SERIAL_ERROR_START;
            SERIAL_ERRORLNPGM("Z-Probe failed to retract!");
            LCD_ALERTMESSAGEPGM("Err: ZPROBE");
          }
          Stop();
        }
```

thinkyhead, 2016-03-26:
Essentially it is an "Allen Key" style probe, and can be configured by treating it as one. Given that more of these kinds of probes are appearing, we might decide to change the name so it isn't so "allen key" specific, but refers to probes that follow procedures for initial deployment and final stowing. This has been on the general agenda for a while, but there have been other priorities superseding it.

Since you seem keen on the subject, please make a PR that makes it work the way it should – or just make a Feature Request, proposing how it should work, and we will keep it on the To-Do list. This issue right here would suffice, but at this point you should be targeting the MarlinDev branch for this request, because with the new release imminent, we have to close this tree to new features.

esenapaj, 2016-03-26:
I bow to your wish, then I close this issue for release of RC5.
And, I challenge to make a PR for MarlinDev.

closed 2016-03-26 by esenapaj