MarlinFirmware/Marlin #3248: [RC4] Busy: Porcessing stream on serial console

Astroport, 2016-03-25:
There is a regular stream of Busy: Processing messages on the serial console when printing. Not necessarily a bug but more of an annoyance.

paulusjacobus, 2016-03-26:
@Astroport I think this was done to keep the comms channel open with repetier host and octoprint.

thinkyhead, 2016-03-26:
There's a new option in `Configuration.h` if you don't need this feature.
```cpp
//
// Host Keepalive
//
// By default Marlin will send a busy status message to the host
// every 2 seconds when it can't accept commands.
//
//#define DISABLE_HOST_KEEPALIVE // Enable this option if your host doesn't like keepalive messages.
```
I'll be submitting patches to some of the host software so that they don't reprint the same message if it repeats multiple times in a row. They should print a number next to the message (how many times it has repeated), and keep it on the same line instead.

closed 2016-03-26 by thinkyhead

thinkyhead, 2016-03-26:
#3252 relaxes the timeout period to 4 seconds. Perhaps it can even be lengthened to 10 seconds. This feature is to let host software know that Marlin hasn't "gone away" but simply has a blocked queue and is not accepting commands.