MarlinFirmware/Marlin #3256: Menu options cause crash and restart

kagouraki, 2016-03-26:
i have branch RCbugfix compiled with arduino 1.6.8. Sanguinololu board and REPRAP_DISCOUNT_SMART_CONTROLLER. 
The problem is that if i choose Level bed the printer beginns to home x for a second and then appears some letters in the screen flashing fast and then the firmware restarts. If i execute the G29 S1 command from repetier it works ok and i can level normally. 
The same thing happens when i choose babystepp x, y or z in the tune menu. The screen goes empty and then the firmware restarts. Bellow is a video of the two problems. 
[https://youtu.be/yOCrYWo_p9U](url)

jbrazio, 2016-03-26:
Try the [latest version of RCBugfix](https://github.com/MarlinFirmware/Marlin/archive/RCBugFix.zip).

kagouraki, 2016-03-26:
I have the latest version of RCBugfix... I downloaded it before the video. Another suggestion ? These are my config and adv config files. 

[config.zip](https://github.com/MarlinFirmware/Marlin/files/190606/config.zip)

emartinez167, 2016-03-27:
Try an earlier version or Arduino.

Regards,

Ernesto

kagouraki, 2016-03-27:
Which one is the latest stable and does it support fillament change, babystepping while printing and manual bed leveling ?

jbrazio, 2016-03-27:
@emartinez167 this only happens with bed leveling from the LCD ?
Auto home from the LCD is OK as an example ?

emartinez167, 2016-03-27:
That is weird behavior... No clue. Could it be a null pointer?

jbrazio, 2016-03-27:
I was asking you, not affirming.

emartinez167, 2016-03-27:
I did not raise this issue, I only suggested using an earlier version of Arduino. @kagouraki is the one who raised it.

Roxy-3DPrintBoard, 2016-03-27:
> That is weird behavior... No clue. Could it be a null pointer?

I'm just causing trouble here.   Because a bad pointer could definitely cause problems.   But my guess is a NULL pointer wouldn't cause these problems because the memory it would hammer is at 0x0000 (Havard Style!!!!)   Almost for sure, this address is quarantined and nothing important is there!

kagouraki, 2016-03-27:
Yes Auto home from LCD works normal. Sorry i was sleeping :P

thinkyhead, 2016-03-27:
We're trying to chase down of a buffer overflow or possibly a bad pointer. So far I haven't been able to reproduce the issue on my own machine, but I need to try some other versions of Arduino, because they do different things with the code and may have different optimization flags. Arduino 1.6.8 is too new for us to have a lot of experience with. Anyway, this seems like an uninitialized pointer, corrupted stack, or buffer overflow. I'm looking at changes to see if anything obvious stands out. It will be better if I can reproduce the crashing behavior and catch it in the act. I will keep you posted as that goes — hopefully fast!

kagouraki, 2016-03-27:
my problem is where i can find the RCbugfix version of ,for example, one month ago. I dont know github very well :/

thinkyhead, 2016-03-28:
@kagouraki From your command line you can use the `git` command to create a branch from any point in time. For example, to get March 20th:
```bash
git checkout 'RCBugFix@{2016-03-20 12:00:00} -b rcbugfix_march_20
```

kagouraki, 2016-03-28:
Ok i will see it probably at the weekend because this week is madness at work. I will keep you informed. Thank you

thinkyhead, 2016-03-31:
Be sure to keep checking the latest `RCBugFix`. A lot of fixes lately, some related to menu options.

closed 2016-04-28 by jbrazio