MarlinFirmware/Marlin #3257: 12864 LCD Encoder not working anymore

lavato, 2016-03-26:
I have installed 12864 LCD with RAMP 1.4 but the Encoder button does not work.
However the same HW setup works with Marlin 1.0.2

thinkyhead, 2016-03-27:
Thanks for bringing this to out attention. Check out these issues: #3007 https://github.com/MarlinFirmware/Marlin/issues/2681#issuecomment-147735504 and see if they have any connection to yours.

I'm investigating the issues discussed in these threads, but so far I see nothing at least _logically_ wrong about the code for handling "slow buttons." I am also comparing how 1.0.2 does things differently from the current version, so hopefully that will turn up some clue – or at least we can revert the behavior.

Incidentally, do you have a Viki or a Makrpanel, or just a generic one? And, what version of Arduino IDE are you compiling with?

lavato, 2016-03-27:
I think it is Makrpanel. I provided the link below.

It is definitely not #3007 & #2681, as the button does not respond at all, so nothing happens when you push or rotate. It is stuck in the home screen.I cannot even get to the menu. 
The only time the push button works is for a brief moment, 100ms, when I can push and rotate just after Marlin reboots, but after this it does not work at all.

I have compiled Marlin with 1.6.7 & 1.6.8 Arduino. The only other thing I noticed is that the 1.0.2 uses different  u8glib ()

FYI:
I used vanilla RC4 and vanilla 1.0.2 to ensure I can reproduce. The only thing I changed in the Configuration.h. is the Motherboard to 33 and LCD entry to "discounted full".

Here is the link for the LCD I use:

http://www.aliexpress.com/item/Free-shipping-3D-printer-smart-controller-RAMPS-1-4-LCD-12864-control-panel-blue-screen/1966418671.html?spm=2114.01010208.3.2.0AtDjb&ws_ab_test=searchweb201556_9,searchweb201602_1_10036_10035_301_10034_507_10032_10020_10001_10002_10017_10010_10005_10006_10011_10003_10021_10004_10022_10009_401_10008_10018_10019,searchweb201603_8&btsid=6e1c84b0-970c-4aab-9c36-1e801b560421

thinkyhead, 2016-03-27:
Be sure to use the latest `RCBugFix` so we know you have the latest code. Just in case we fixed this by mistake already.

lavato, 2016-03-27:
I tried it, unfortunately it is still the same, reverted back to 1.0.2 again and it works.

FYI: the RCBugFix has its default motherboard set to RAMP EBF

thinkyhead, 2016-03-28:
EBF? Hmm... I'll make sure that's patched ASAP. I'm going to try and track down this issue for you today. I wonder if you could try the older u8glib from 1.0.2 with `RCBugFix` and see if it helps. If it's a library conflict that would be useful to know.

Astroport, 2016-03-28:
I am sure that the default mother board is EFB
From configuration.h
#ifndef MOTHERBOARD
  #define MOTHERBOARD BOARD_RAMPS_14_EFB
#endif

This is 43 in boards.h 
Works OK here with the same controller

thinkyhead, 2016-03-28:
@lavato I'll keep staring at the code, but we may need to ask you to do some testing if we're going to find this bug. It's good to know that the bug doesn't exist in 1.0.2. How about `1.1.0-RC2` or `1.1.0-RC3`? Does the bug exist in those? If not, I can provide you with branches from different points in time, so we can narrow the bug down to a particular week or day when it started.

Roxy-3DPrintBoard, 2016-03-28:
@thinkyhead You have mail...

Blue-Marlin, 2016-03-28:
> I think it is Makrpanel.

The link shows a `REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER`

thinkyhead, 2016-03-28:
@lavato Is this the LCD you have selected in your configuration – –
`REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER`?

lavato, 2016-03-28:
Yes

lavato, 2016-03-28:
NP, I will do some testing, let me know what exactly I should test.

lavato, 2016-03-28:
OK, I recompiled below and:
- RC 1 works
- RC 2,3  & 4 does not work.

Grogyan, 2016-03-28:
I suggest looking at your Full Graphic Display board for solder shorts.
If you have a multimeter, you can do a continuity check on the encoder when the printer is switched off.
Also, try swapping the cables over, it may be possible that you have bung cable.

As I and many others use the same setup as you do, and not have any problems.

There was a brief time when RCbugfix, and the display wouldn't work, but it has since been fixed.

lavato, 2016-03-29:
Not sure I agree with what you are saying as the card works with 1.0.2 and RC 1.

jbrazio, 2016-03-29:
@Grogyan @Iavato which boards are you using ?

lavato, 2016-03-29:
RAMPS 1.4

Roxy-3DPrintBoard, 2016-03-29:
@thinkyhead  You are a popular guy!  You have more mail.

thinkyhead, 2016-03-31:
We continue to patch bugs, some related to LCD. So please try the latest `RCBugFix` when you get a chance and let us know if we fixed this issue yet.

lavato, 2016-03-31:
I did, as this is what you suggested before, however, I though you were to send cuts of code from different periods between the two RCs, which I was to test? (i.e. time periods between RC1 & RC2)

I hope this is not affecting majority of 12864s but I would imagine people don't often change the firmware, which is why this may not have been highlighted before. 
If this proves to be more common, people with same config may not be able to use the new release.

I would have though my config was pretty standard?

thinkyhead, 2016-04-01:
@Roxy-3DPrintBoard Sorry, my internet access has been somewhat limited lately. I will get to your email shortly…

thinkyhead, 2016-04-01:
@lavato I'm still unable to reproduce the issues you're seeing, but I don't have your exact display. Anyway, more bugs have been getting patched in `RCBugFix`, and some of them are around the LCD menus, so please give the latest code a try when you get a chance (. Be sure to copy the `u8glib` that comes with Marlin into your Sketchbook folder. Otherwise Arduino might be using a version we haven't tested with. (It's stored in the `ArduinoAddons/Arduino_1.6.x/hardware/marlin/avr/libraries/U8glib` folder).

Blue-Marlin, 2016-04-01:
U8GLIB has __NOTHING__ to do with the encoder.

I'm not convinced it's a error with the encoder at all. With about the same likelihood it's a ignored MINTEMP and killed therefore.

What does the display/log tell you.

thinkyhead, 2016-04-02:
> U8GLIB has NOTHING to do with the encoder.

Certainly. But it would be lax not to make sure we're all using the same libs for testing. I also want to eliminate the possibility of _any unrelated components_ stepping on stack or buffers and causing such strange behavior in the LCD menus. (Lately buffer corruption has been more of a worry on my mind, because we saw buffer corruption in the command queue recently –– but only for one user so far - whew!)

> With about the same likelihood it's a ignored MINTEMP and killed

If there's a temperature error, that is usually flagged on the LCD screen. If it isn't, then that would be another bug to look into.

Blue-Marlin, 2016-04-02:
@thinkyhead 
`REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER` is about the second most installed display out there.
`RAMPS 1.4 & Arduino Mega 2560` is about the most common combo out there.
In the encoder handling we have not had relevant changes between 1.0.2 and now. If so, the changes (to RC1) would be about a year old now.

The likelihood he is the only one who discovered that "encoder" problem is extremely small.
Much more likely is, that he's missing to tell us an important detail of his sorry.

Killed, would be a good (and likely) reason for a death knob/RAMPS/MEGA/Marlin.

thinkyhead, 2016-04-02:
Perhaps we might helpfully suggest a test that he can do to confirm or refute this possibility.

lavato, 2016-04-02:
@Blue-Marlin I agree,  the chance is small - but this particular version of the Ramps is widely sold on eBay & AliExpress so the market is folded with this stuff.

also

I have now reproduced this many times moving from 1.0.2 to RC4\RCBugFix and changing the same points in the config, the MBO and the LCD. I also removed all HW and only left the LCD. (to make sure)

Sadly RC4/RCBugFix still does not work but 1.0.2 does.

I though the idea that @thinkyhead  had to test cuts of code between 1.0.2 and RC1 is a good one but I need help obtaining the source code for these time periods.

Blue-Marlin, 2016-04-02:
Is there any reason for not telling us about your status line or the last 10 rows of the serial connection?

lavato, 2016-04-02:
@Blue-Marlin 
Not sure what you are implying here.

What do you mean by "status line" and "10 rows of the serial connection"?

Blue-Marlin, 2016-04-02:
What does the lowest line on your display show?

What does the terminal window of your host program show?

lavato, 2016-04-02:
@thinkyhead 
OK. I don't know what the problem is exactly but I can confirm that encoder does not work when all the hardware is removed. ( so the only thing connect to RAMPS is 2560, PCU & LCD, 

If the stepper motors, hot end etc are attached the encoder works. I tested with RC4, RC5 and RCBugFix.

But as per above, 1.0.2 works regardless. Not sure if you want to investigate fuhrer but I am happy with this "feature".

@Blue-Marlin 
**Last status line says: ** 3d Printer ready.

**Serial last 10 lines:**
echo:Home offset (mm):
echo:  M206 X0.00 Y0.00 Z0.00
echo:Material heatup parameters:
echo:  M145 S0 H180 B70 F0
echo:  M145 S1 H240 B110 F0
echo:PID settings:
echo:  M301 P22.20 I1.08 D114.00 C100.00 L20
echo:Filament settings: Disabled
echo:  M200 D3.00
echo:  M200 D0

Blue-Marlin, 2016-04-02:
Thanks for the feedback.
Instead of connecting all the periphery it would be completely sufficient to use 999 as thermistor type, to avoid the ERR: MINTEMP and the kill for that.

thinkyhead, 2016-04-03:
Sounds like a result of our tightening up on issues like disconnected thermistors at startup. If the encoder works (and the printer is not in `kill` state) under normal conditions, then this is closed.

closed 2016-04-03 by thinkyhead

lavato, 2016-04-03:
@Blue-Marlin 
Thanks for the suggestion.

I need to print some stuff before I take out all HW again but I will try it next week, as I will be doing some more upgrades.

So I understand, are you are saying that after the  "ERR: MINTEMP" event occurs, the framework will prevent encoder from working in RCs 1-5 but not in 1.0.2?

Blue-Marlin, 2016-04-03:
I can't remember on 1.0.2 really. But since about a year we kill the machine when we see a MINTEMP error - or any other potentially disastrous condition.. That means - you can't do anything but reset. And before you ask - yes, that's wanted to be exactly this way.

lavato, 2016-04-03:
you said it. ;)  

The only thing I would do is just add a comment in the code to clarify it (the 999 point), so when people hook up just the LCD they know if works.

lavato, 2016-04-17:
I can confirm that "Killed" event, which was caused by "MINTEMP"  is a reason for LCD encoder not working.

OmurCeran, 2016-08-07:
Hi.I have a problem which is almost similar this.There is a photo which is mine lcp smart controller.After ı upload rc7 firmware for my 3d printer, I go through this problem.Can you solve this problem ?
[image]

Blue-Marlin, 2016-08-07:
"KILLED." is printed in 3 situations
* You called it with M112
* "trying to call sub-gcode files with too many levels. MAX level is:"
* INVALID_EXTRUDER_NUM

The serial log will show you the details.

OmurCeran, 2016-08-07:
I don't understand clearly. Can you explain more clearly?

Blue-Marlin, 2016-08-07:
No.

OmurCeran, 2016-08-07:
What? Why?

Blue-Marlin, 2016-08-07:
Please look up the reason in the serial output.
Please report the output here.
Than, when we know the reason, we can thy to give you advice what to do.

thinkyhead, 2016-08-08:
@OmurCeran If you connect Pronterface or some other host, then you should see a more detailed error message in the console when this kill screen appears.