MarlinFirmware/Marlin #3261: Relative Extrusion Causes Extruder Chatter

jneilliii, 2016-03-27:
After recently upgrading firmware to Marlin 1.1 RC4 my Bulldog XL extruder began to have issues with absolute extrusion.  The same file sliced with relative extrusion in Slic3r 1.2.9 works without issue. The symptoms of the problem is that the extruder would start to chatter as if there was a hot end jam.  If you attempt to manually extrude after the issue occurred it would persist and still be a problem.  After resetting the RAMBo board it would allow you to extrude manually without issue. I've attached the 2 files.  The one appended with relative prints without issue, the other is problematic.

[Glass_7mm.gcode.txt](https://github.com/MarlinFirmware/Marlin/files/190789/Glass_7mm.gcode.txt)
[Glass_7mm_relative.gcode.txt](https://github.com/MarlinFirmware/Marlin/files/190788/Glass_7mm_relative.gcode.txt)

thinkyhead, 2016-03-27:
Thank you for including the GCode files. I haven't used any relative extrusion, so I can't comment yet, but I will give it a try. I'd assume there's very little difference in the behavior, unless the relative GCode is somehow different. But I will run it through the parser and see if there's something weird going on.

thinkyhead, 2016-03-27:
Incidentally, have you tried a few different absolute versus relative GCode files? It would help to have a larger sample size, because at the moment we're looking for something that looks like a possible buffer overrun issue.

Also, before we go too much further you should make sure this hasn't already been patched in the current `RCBugFix` code, which has already gotten some important fixes since RC4.

jneilliii, 2016-03-27:
So I have gone back and tried to print another file that was having issue, re-sliced with relative positioning and the issue is still there.  I've attached the gcode on that one here, and realize it is a much more complicated print but a good example of the problem no matter what coordinate positioning I use for the slice. I'll give RCBugFix branch a try and report back.
[Tie_Advanced_X1_Front.gcode.txt](https://github.com/MarlinFirmware/Marlin/files/190995/Tie_Advanced_X1_Front.gcode.txt)

jneilliii, 2016-03-27:
Ok, so far RCBugFix branch appears to be working with this last relatively positioned gcode file.  If problem comes back I'll re-open this issue.

closed 2016-03-27 by jneilliii

bubnikv, 2016-12-22:
Any idea what is going on? We seem to have some issues with our Marlin fork at Prusa Research when printing with relative extruder coordinates. It seems that the E axis accumulator (a floating point number) loses a lot of precision when meters of filament are printed. In absolute mode the slic3r resets the E accumulator at each retraction, but in relative mode the E accumulator grow is unlimited.

thinkyhead, 2016-12-23:
@bubnikv The Prusa fork is based on some pretty old code, at this point. I don't know where we specifically fixed this issue, but I believe it has been fixed for a while now.

Are you seeing an issue where the E value grows so large that it can no longer be expressed in stepper steps?

At any given time, whatever the high-level E axis value might be, it's always using the same steps-per-mm factor to convert into stepper steps as it goes to the planner. So the steps value should always correspond to the E axis value. The output of `M114`, which includes the stepper steps count, should bear this out.

My first suggestion to isolate the issue in your fork is to make sure that wherever maths are being done with both `long` and `float`, be sure to cast the `long` variables to `float`. Of course, examine our current `planner.*` and `stepper.*` code and compare it to yours. In the process you may see some changes that make sense and adopt them.

Blue-Marlin, 2016-12-23:
If the problem is caused by the limited exactnes of floats we also have the problem.
Roughly a float can hold a limited number of digits. If a very small number is added to a very big number it's possible to see the big number not increasing - or not enough.

Blue-Marlin, 2016-12-23:
Lets guess.
1 m is abour 1024mm = 2^10.  If we want to move by 0.001mm this is 2^-10. Togther we need 20 binary digits. A single floats mantissa is 23 bits long. -> There in not much room for larger or smaller numbers.

Can we find a way to reset E if it grows to large?

Blue-Marlin, 2016-12-23:
If current_position[E_AXIS] is always the current extruder, this
```diff
@@ -2849,10 +2849,14 @@ static void homeaxis(AxisEnum axis) {
  *  - Set destination from included axis codes
  *  - Set to current for missing axis codes
  *  - Set the feedrate, if included
  */
 void gcode_get_destination() {
+  if (fabs(current_position[E_AXIS]) > 1000.0) {
+    current_position[E_AXIS] = 0.0;
+    sync_plan_position_e();
+  }
   LOOP_XYZE(i) {
     if (code_seen(axis_codes[i]))
       destination[i] = code_value_axis_units(i) + (axis_relative_modes[i] || relative_mode ? current_position[i] : 0);
     else
       destination[i] = current_position[i];
```
should help. Even 500.0mm should be ok.

Blue-Marlin, 2016-12-23:
Or, a bit more direct, we could ask
```
if (current_position[E_AXIS] > code_value_axis_units(i) * (float) 0xffff )  // if binary magnitude differers more than 16 bit 
```
but the float multiplication is ugly.

bubnikv, 2017-01-06:
> Are you seeing an issue where the E value grows so large that it can no longer be expressed in stepper steps? (Hmm, actually I guess that's pretty unlikely.)

Ok, we found a culprit. Our print farm stuff experimented with alternative starter G-codes. The problem with relative extuder was caused by running the machine at a print farm for a number of prints in an extruder relative mode, while never resetting the E counter. The original starter G-code contained G92 E0, while the modified did not. Calling G92 E0 once per print is sufficient to keep the floating point E counter at a reasonable level. Without resetting the E counter, the E counter accumulates indefinitely and it starts losing digits after a couple of prints. The homing command (G28) does not reset the E counter in any case.

thinkyhead, 2017-02-11:
Should we reset the E counter under some condition or just let it be?

bubnikv, 2017-02-11:
For now, I am fine with the current behavior as long as it is documented.

It would be nice to reset the extruder somehow and somewhere, but it would take some time to find a safe way to do it without breaking someone's assumption.