MarlinFirmware/Marlin #3264: Method to load/unload filament

jbrazio, 2016-03-27:
This may seen a stupid question, but I would like to know how do you load/unload the filament in your printer using Marlin. Let's say you're preparing to print a red thing and have the yellow filament inserted. Thanks for your input.

Roxy-3DPrintBoard, 2016-03-27:
I turn on the M600 command.   And then, before you send your sliced GCode to the printer, you go edit in an M600 command at the right level.   Typically, I look for a G1 Z??.??? that changes the nozzle to the desired height.   I then edit in a M600 Z 50   (Z 50 gives me an extra 50 mm of clearance to work)

When it hits the M600 it will automatically raise the nozzle and retract the filament.   And then it will turn off the Z Axis motor so you can manually feed in a different color.   I typically put a piece of paper on top of the currently printed part just to keep the new filament that I'm manually extruding from getting stuck to the printed piece.   The problem is you have to extrude 100mm or maybe even 150mm of filament to get the nozzle fully purged and you don't want this getting stuck to the piece being printed.   

When everything is good...  You click the LCD Panel button and everything continues with the new color.

thinkyhead, 2016-03-28:
Sounds like it turns off the E axis motor, not the Z axis motor, is that correct Roxy?

Roxy-3DPrintBoard, 2016-03-28:
Yes.  You can control where the X,Y & Z are moved to for the filament
change.   But the Extruder motor is turned off until the LCD Button is
clicked.

jbrazio, 2016-03-28:
This is for mid-print, and in the beginning of the print ? You also use `M600` ?

Roxy-3DPrintBoard, 2016-03-28:
Oh!  No...   I tell PronterFace to turn off all the motors, And I just manually do what has to be done.

thinkyhead, 2016-04-08:
In my opinion, the first thing `M600` should do is a short retract and a very fast move in the XY plane, towards the right or back edge of the bed. I find that if the nozzle goes straight up it leaves a sturdy thread sticking out of the top of the print, but going sideways leaves only a small whisker, if anything.

Roxy-3DPrintBoard, 2016-04-08:
I guess I never thought about it, but "Yes!"  it is true.  M600 always leaves a big pimple on the part being printed.   Doing what you suggest, would fix that.   Normally, this is something I would jump in and mess around with.  But...   It will have to wait.

jbrazio, 2016-04-08:
I'm reopening the issue as it seems there is interest into discussing this feature.

What could we improve on `M600` ?

Roxy-3DPrintBoard, 2016-04-08:
> What could we improve on M600 ?

There is the Pimple issue that ThinkyHead brought up.   But also, the current M600 is dependent upon an LCD Panel being in the system.   The reason is it is looking for an LCD Click to finish the command and continue processing commands in the buffer.   The problem is, this makes it so a user without an LCD Panel can not use the command to do a filament change.    

I added a push button switch to one of my free pins and warmed over the code so I could use the M600 command without an LCD Panel.   But in hindsight, I could have just used one of the End Stops or Z-Probe to indicate to the firmware it was OK to proceed.   So, what I'm suggesting is we allow an LCD Click to be generated by the LCD Panel and/or by one of the end stops (or probe).    

The M600 code is very well written!   Other than these items...   Nothing has needed to be done to it for years.

thinkyhead, 2016-04-09:
It would be nice to be able to send a GCode that simulates a "user click" during any "Wait for user" pause. But this would require a change to the way we handle commands from serial.

We would need to keep monitoring the serial input even when the serial command buffer is full, and if certain commands arrive, execute them immediately without putting them in the command queue at all. Then we would be able to add a code (I suggest `M2`) that has the same result as clicking the LCD.

Blue-Marlin, 2016-04-09:
For all processors using `MarlinSerial` that is not a big deal. For all using `USBCON` (AT90USB) i see no way, except integrating USBCON into Marlin.

CNTRL-X for kill, CHTRL-C for breaking the current command or stop, ... - so many possibility's.

thinkyhead, 2016-04-10:
I can imagine a pure software solution, but not easily. First, is it possible to collect serial input during an interrupt instead of in the main loop? That would make for the easiest implementation. If it must be done in the main loop, then it gets kind of hairy, because we have "eddies" where the command handlers temporarily "take over" for the main loop. Obviously we wouldn't want to invoke the command dispatcher recursively.

jbrazio, 2016-05-04:
With PR #3662 and #3605 this topic is being followed-up.

closed 2016-05-04 by jbrazio