MarlinFirmware/Marlin #3266: 1 nozzle 2 extruder

yaniekarlina, 2016-03-28:
hello everyone. i have a project to make 3d printer delta with 1 nozzle and 2 extruder. can marlin make it? i try to modify marlin program but it not working.

Roxy-3DPrintBoard, 2016-03-28:
You want to feed two filaments into 1 nozzle?  The Marlin Firmware can handle that assuming the nozzle can be purged reliably.

yaniekarlina, 2016-03-29:
yes, that's what i mean. so what must i do?

Roxy-3DPrintBoard, 2016-03-29:
Configure Marlin to have 2 extruders that are 0 mm apart.   And get both of the motors running correctly.   At that point, you may need to tell the Slic3r to do some custom GCode to purge the nozzle when you do a color change, but that is about it....

thinkyhead, 2016-03-30:
Try MarlinKimbra and see if it can do what you want. It already has native support for this, which it calls SINGLENOZZLE. I have submitted a mixing extruder feature that also does this, but only to the MarlinDev branch, and it hasn't been merged yet. If you like, I can make a branch for you that has the current Marlin plus this feature. It works just like the Repetier Host mixing feature.

closed 2016-05-04 by jbrazio