MarlinFirmware/Marlin #3270: Stuck at heating bed RC4

leecuk, 2016-03-28:
Print sticks waiting for bed temperature to stabilise before turning on the extruder heater. Bed is using pid control and is very stable at requested temperature. Worked fine in RC3, anyone else had problems? - Keep up the good work guys :)

jbrazio, 2016-03-28:
Could you please post the test gcode, configuration files and if possible a log for the serial output ? Thanks.

leecuk, 2016-03-28:
Please find attached the config without pid control on the bed, there is nothing really unusual. 
[Configuration.h.txt](https://github.com/MarlinFirmware/Marlin/files/192579/Configuration.h.txt)

The start of the gcode looks like this - 

G90
M82
M106 S0
M140 S110
M190 S110
M104 S240 T0
M109 S240 T0
G28

Thanks

jbrazio, 2016-03-29:
You have some redundancy on the G-Code, this is enough:
```
G90
M82
M190 S110
M109 S240 T0
G28
```

Nevertheless is it possible to copy the serial output of Marlin ?

WheresWaldo, 2016-03-29:
I am seeing the same issue if you choose to wait for the bed heater. Just a couple of FYIs, previous versions of Marlin did not exhibit this behaviour and Simplify3D always inserts both commands for heating if you select Wait in their temperature screens. So even though it is redundant it should n't behave the way it does.

Both RC4 and RC4BugFix (pulled 27-MAR) exhibit this behaviour.

jbrazio, 2016-03-29:
Thanks for the heads up.
Is it possible to get a serial output when Marlin is exhibiting the issue ?

WheresWaldo, 2016-03-29:
It would be my pleasure, but could you please advise how do I get the serial output? My setup is controlled by OctoPi on rPi2 if that helps.

Blue-Marlin, 2016-03-29:
Switch to the Terminal-tab.

leecuk, 2016-03-29:
[output.txt](https://github.com/MarlinFirmware/Marlin/files/194080/output.txt)
Please find attached the terminal output, the bed was set at 40 degrees.

I'm also using Simplify3D.

Thanks

marvin42blue, 2016-03-29:
I'm having the same issues with my self-built kossel mini, I'm using pronterface to control my printer via USB, this is my startup sequence:

M107
M190 S50 ; set bed temperature
M104 S215 ; set temperature
G28 ; home all axes
G1 Z15 F5000 ; lift nozzle
G1 X-50 ; outside start pos
M109 S215 ; wait for temperature to be reached

I've noticed, when I manually start heating and issue a manual "M109 S215" command, it seems to work (based on the fact that the serial output W:n counts from 10 to 0 and then M109 is done).
But when I try to "print" it hangs with "W:?" - the temperature reading on the LCD display is ok and it also worked well with RC3.

marvin42blue, 2016-03-29:
I found an issue in the M190 code:

>  while (!cancel_heatup && degTargetBed() != degBed()) {

both functions degTargetBed() and degBed() return a float value - it's not good to compare two float values for equal values, this might never happen.

I decided to give the temperatur a hysteresis of 1 degree and changed the code into

> while (!cancel_heatup && (fabs(degTargetBed()-degBed()) > 1.0) )

and this part works

I also added a code to M109 to compute this:
int temp = labs(degHotend(target_extruder) - degTargetHotend(target_extruder));
and this returns pretty silly values, like 340, 12756 etc....

I'm not sure whether its a good idea to use labs() on float values without conversion to int

trnoa, 2016-03-29:
I had the same issue. With the correction from Marvin42blue it works! Tanks a lot!

jbrazio, 2016-03-30:
Thanks @marvin42blue, a PR has been submitted.

thinkyhead, 2016-03-30:
And merged… RC5 coming soon…

PR #3277 (2016-03-30): M190 bed temp is never reached
Follow up ea9fd1200bf7999bb83cb9ed2851019c87cd1c23
Fixed #3270
closed 2016-03-30 by thinkyhead