MarlinFirmware/Marlin #3281: update command

WillSmithInChina, 2016-03-30:
is there any command to update firmware ? if you know that, please tell me  ！！！thank you, guys

jbrazio, 2016-03-30:
No, you have to use Arduino IDE/avrdude.

thinkyhead, 2016-03-30:
https://www.youtube.com/watch?v=HV0X7c389cI

closed 2016-03-30 by thinkyhead

jbrazio, 2016-03-30:
@thinkyhead that one was really targeted at me :-P
I believe the question here was more if you could update the firmware using the SDCard with some G/M command.. this was my interpretation.

thinkyhead, 2016-03-30:
That would be pretty magical if an Arduino could flash itself!

It will be great when we can point to our excellent documentation. I'm putting out the call tonight!

jbrazio, 2016-03-30:
> This is a proof-of-concept dual bootloader for Atmel AVR atmega chips.
It combines a serial bootloader and a mmc/sd card based one in the 
same 2kb of flash memory that is normally used by the arduino serial 
loader alone.

https://github.com/thseiler/embedded/tree/master/avr/2boots

paulusjacobus, 2016-03-30:
That's a great feature, never thought of that before. If it works, it might
become our default upgrade path for non-technical oriented users....

Blue-Marlin, 2016-03-30:
That's unlikely.
It would need to flash a new bootloader first.
Hen - Egg problem.