MarlinFirmware/Marlin #3283: END SCRIPT MPT WORKING  echo:Unknown command: "m402"

dinizhobby, 2016-03-30:
Any solution for this problem below?

after G29  #define Z_PROBE_END_SCRIPT "m402" not working
15:29:41.580 : echo:Unknown command: "m402"

jbrazio, 2016-03-30:
Have you tried with a capital `M` ? Like `M402` ?.

Blue-Marlin, 2016-03-30:
Using M402 in an `Z_PROBE_END_SCRIPT` does not make any sense.
M402 is defined for `Z_PROBE_SLED`, `SERVO_ENDSTOPS`, `Z_PROBE_ALLEN_KEY`. If you have not defined one of the - nothing will happen.
On the other hand, if one of these is defined, the probe handling is automatically integrated into G29, so no `Z_PROBE_END_SCRIPT` is needed.

dinizhobby, 2016-03-30:
Hi

I use Servo for Z_probe. if i write m401 and m402 is working ok.
But after command G29 servo no retract.

I did insert M401 in my G-code  before G29.

dinizhobby, 2016-03-30:
I need just retract servo after G29 command.

thinkyhead, 2016-03-30:
Sorry about that. It's supposed to retract automatically after `G29`. What happens if you remove `M401` and `M402` completely?

dinizhobby, 2016-03-30:
G29 no lower and no retract servo.
I need enter command manual M401 for lower and M402 for retract it.

thinkyhead, 2016-03-30:
If `G29` is failing to raise and lower the servo, here's what you need to to:

First, be sure to start with the latest `RCBugFix` code, which has received a few important fixes since `RC4`.

You should enable the `DEBUG_LEVELING_FEATURE` option and before you do your `G29` routine, make sure to have a host or console like Pronterface connected and send the command `M111 S32` to enable detailed logging of the bed leveling routines. Then when you do `G29` you will get detailed output of what the machine "thinks" is going on.

Copy the  output from the console and paste it here so we can have a look at what's actually going on.

Blue-Marlin, 2016-03-30:
As my servo probe works as it should work - maybe your Configuration.h could help us to find the problem.

dinizhobby, 2016-03-30:
Connecting...
start
Printer is now online.
echo:Marlin 1.1.0-RC4
echo: Last Updated: 2016-03-07 12:00 | Author: (none, default config)
Compiled: Mar 30 2016
echo: Free Memory: 3140  PlannerBufferBytes: 1232
echo:V22 stored settings retrieved (392 bytes)
echo:Steps per unit:
echo:  M92 X80.00 Y80.00 Z80.00 E169.80
echo:Maximum feedrates (mm/s):
echo:  M203 X500.00 Y500.00 Z500.00 E25.00
echo:Maximum Acceleration (mm/s2):
echo:  M201 X9000 Y9000 Z9000 E10000
echo:Accelerations: P=printing, R=retract and T=travel
echo:  M204 P3000.00 R3000.00 T3000.00
echo:Advanced variables: S=Min feedrate (mm/s), T=Min travel feedrate (mm/s), B=minimum segment time (ms), X=maximum XY jerk (mm/s),  Z=maximum Z jerk (mm/s),  E=maximum E jerk (mm/s)
echo:  M205 S0.00 T0.00 B20000 X20.00 Z20.00 E5.00
echo:Home offset (mm):
echo:  M206 X0.00 Y0.00 Z0.00
echo:Endstop adjustment (mm):
echo:  M666 X-0.70 Y0.00 Z-0.50
echo:Delta settings: L=diagonal_rod, R=radius, S=segments_per_second, ABC=diagonal_rod_trim_tower_[123]
echo:  M665 L213.50 R107.80 S200.00 A0.00 B0.00 C0.00
echo:Material heatup parameters:
echo:  M145 S0 H180 B70 F255
echo:  M145 S1 H240 B100 F255
echo:PID settings:
echo:  M301 P22.20 I1.08 D114.00 C100.00 L20
echo:Filament settings: Disabled
echo:  M200 D3.00
echo:  M200 D0
echo:Z-Probe Offset (mm):
echo:  M851 Z-1.20
>>> M111 S32
SENDING:M111 S32
echo:DEBUG LEVELING ENABLED
>>> G28
SENDING:G28
gcode_G28 >>>
reset_bed_level
setup_for_endstop_move > enable_endstops(true)
>>> homeaxis(0)
> enable_endstops(false)
> enable_endstops(true)
> TRIGGER ENDSTOP > current_position: (0.00, 729.00, 729.00)
> enable_endstops(false)
> endstop_adj = -0.70 > destination: (-0.70, 729.00, 729.00)
> enable_endstops(true)
set_axis_is_at_home 0 > (home_offset[axis]==0.00) > current_position: (0.00, 729.00, 729.00)
> AFTER set_axis_is_at_home > current_position: (0.00, 729.00, 729.00)
<<< homeaxis(0)
>>> homeaxis(1)
> enable_endstops(false)
> enable_endstops(true)
> TRIGGER ENDSTOP > current_position: (0.00, 0.00, 729.00)
> endstop_adj * axis_home_dir = 0.00
set_axis_is_at_home 1 > (home_offset[axis]==0.00) > current_position: (0.00, 0.00, 729.00)
> AFTER set_axis_is_at_home > current_position: (0.00, 0.00, 729.00)
<<< homeaxis(1)
>>> homeaxis(2)
> enable_endstops(false)
> enable_endstops(true)
> TRIGGER ENDSTOP > current_position: (0.00, 0.00, 0.00)
> enable_endstops(false)
> endstop_adj = -0.50 > destination: (0.00, 0.00, -0.50)
> enable_endstops(true)
set_axis_is_at_home 2 > (home_offset[axis]==0.00) > current_position: (0.00, 0.00, 243.00)
> AFTER set_axis_is_at_home > current_position: (0.00, 0.00, 243.00)
<<< homeaxis(2)
(DELTA) > current_position: (0.00, 0.00, 243.00)
ENDSTOPS_ONLY_FOR_HOMING enable_endstops(false)
<<< gcode_G28
>>> G29
SENDING:G29
gcode_G29 >>>
G29 Auto Bed Leveling
reset_bed_level
setup_for_endstop_move > enable_endstops(true)
z_before = (before) 20.00
z_before = (before) 20.00
z_before = (before) 20.00
z_before = (before) 20.00
z_before = (before) 20.00
z_before = (before) 20.00
z_before = (before) 20.00
z_before = (before) 20.00
z_before = (before) 20.00
z_before = (before) 20.00
z_before = (before) 20.00
z_before = (before) 20.00
probe_pt >>>
> ProbeAction:0
> current_position: (0.00, 0.00, 243.00)
Z Raise to z_before 20.00
> do_blocking_move_to_z 20.00
do_blocking_move_to: (0.00, 0.00, 20.00)
prepare_move_raw > destination: (0.00, 0.00, 20.00)
echo:busy: processing
echo:busy: processing
> do_blocking_move_to_xy 37.00, -33.00
do_blocking_move_to: (37.00, -33.00, 20.00)
prepare_move_raw > destination: (37.00, -33.00, 20.00)
echo:busy: processing
run_z_probe (DELTA) 1
prepare_move_raw > destination: (37.00, -33.00, -10.00)
run_z_probe (DELTA) 2 > current_position: (37.00, -33.00, -10.00)
<<< probe_pt
z_before = (between) 10.00
probe_pt >>>
> ProbeAction:0
> current_position: (37.00, -33.00, -10.00)
Z Raise to z_before 10.00
> do_blocking_move_to_z 10.00
do_blocking_move_to: (37.00, -33.00, 10.00)
prepare_move_raw > destination: (37.00, -33.00, 10.00)
echo:busy: processing
> do_blocking_move_to_xy 20.00, -33.00
do_blocking_move_to: (20.00, -33.00, 10.00)
prepare_move_raw > destination: (20.00, -33.00, 10.00)
run_z_probe (DELTA) 1
prepare_move_raw > destination: (20.00, -33.00, -10.00)
run_z_probe (DELTA) 2 > current_position: (20.00, -33.00, -10.00)
<<< probe_pt
z_before = (between) 10.00
probe_pt >>>
> ProbeAction:0
> current_position: (20.00, -33.00, -10.00)
Z Raise to z_before 10.00
> do_blocking_move_to_z 10.00
do_blocking_move_to: (20.00, -33.00, 10.00)
prepare_move_raw > destination: (20.00, -33.00, 10.00)
> do_blocking_move_to_xy 3.00, -33.00
do_blocking_move_to: (3.00, -33.00, 10.00)
prepare_move_raw > destination: (3.00, -33.00, 10.00)
run_z_probe (DELTA) 1
prepare_move_raw > destination: (3.00, -33.00, -10.00)
Disconnected.

thinkyhead, 2016-03-30:
Incidentally, what is your `Z_ENDSTOP_SERVO_NR` setting? @Blue-Marlin It occurs to me that we probably need to sanity check that if the servo number array is defined, the `[XYZ]_ENDSTOP_SERVO_NR` also must exist. Older configs may lack these.

dinizhobby, 2016-03-30:
#define NUM_SERVOS 3 // Servo index starts with 0 for M280 command

// Servo Endstops
//
// This allows for servo actuated endstops, primary usage is for the Z Axis to eliminate calib
// Use M851 to set the Z probe vertical offset from the nozzle. Store that setting with M500.
//
#define X_ENDSTOP_SERVO_NR -1
#define Y_ENDSTOP_SERVO_NR -1
#define Z_ENDSTOP_SERVO_NR 2
#define SERVO_ENDSTOP_ANGLES {{0,0}, {0,0}, {-20,100}} // X,Y,Z Axis Extend and Retract angles

thinkyhead, 2016-03-30:
Thanks! I have to run soon, but will look into this more later. Perhaps someone else will have a solution sooner. You might as well post your whole `Configuration.h` file too.

dinizhobby, 2016-03-30:
/**
 * Marlin 3D Printer Firmware
 * Copyright (C) 2016 MarlinFirmware [https://github.com/MarlinFirmware/Marlin]
 *
 * Based on Sprinter and grbl.
 * Copyright (C) 2011 Camiel Gubbels / Erik van der Zalm
 *
 * This program is free software: you can redistribute it and/or modify
 * it under the terms of the GNU General Public License as published by
 * the Free Software Foundation, either version 3 of the License, or
 * (at your option) any later version.
 *
 * This program is distributed in the hope that it will be useful,
 * but WITHOUT ANY WARRANTY; without even the implied warranty of
 * MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the
 * GNU General Public License for more details.
 *
 * You should have received a copy of the GNU General Public License
 * along with this program.  If not, see <http://www.gnu.org/licenses/>.
 *
 */

/**
 * Configuration.h
 *
 * Basic settings such as:
 *
 * - Type of electronics
 * - Type of temperature sensor
 * - Printer geometry
 * - Endstop configuration
 * - LCD controller
 * - Extra features
 *
 * Advanced settings can be found in Configuration_adv.h
 *
 */
#ifndef CONFIGURATION_H
#define CONFIGURATION_H

#include "boards.h"
#include "macros.h"

//===========================================================================
//============================= Getting Started =============================
//===========================================================================

/**
 * Here are some standard links for getting your machine calibrated:
 *
 * http://reprap.org/wiki/Calibration
 * http://youtu.be/wAL9d7FgInk
 * http://calculator.josefprusa.cz
 * http://reprap.org/wiki/Triffid_Hunter%27s_Calibration_Guide
 * http://www.thingiverse.com/thing:5573
 * https://sites.google.com/site/repraplogphase/calibration-of-your-reprap
 * http://www.thingiverse.com/thing:298812
 */

//===========================================================================
//============================= DELTA Printer ===============================
//===========================================================================
// For a Delta printer replace the configuration files with the files in the
// example_configurations/delta directory.
//

//===========================================================================
//============================= SCARA Printer ===============================
//===========================================================================
// For a Scara printer replace the configuration files with the files in the
[... 979 lines cut ...]
 * Note may require analog pins to be defined for different motherboards
 **********************************************************************/
// Uncomment below to enable
//#define FILAMENT_SENSOR

#define DEFAULT_NOMINAL_FILAMENT_DIA 3.00  //Enter the diameter (in mm) of the filament generally used (3.0 mm or 1.75 mm) - this is then used in the slicer software.  Used for sensor reading validation

#if ENABLED(FILAMENT_SENSOR)
  #define FILAMENT_SENSOR_EXTRUDER_NUM 0   //The number of the extruder that has the filament sensor (0,1,2)
  #define MEASUREMENT_DELAY_CM        14   //measurement delay in cm.  This is the distance from filament sensor to middle of barrel

  #define MEASURED_UPPER_LIMIT         3.30  //upper limit factor used for sensor reading validation in mm
  #define MEASURED_LOWER_LIMIT         1.90  //lower limit factor for sensor reading validation in mm
  #define MAX_MEASUREMENT_DELAY       20     //delay buffer size in bytes (1 byte = 1cm)- limits maximum measurement delay allowable (must be larger than MEASUREMENT_DELAY_CM  and lower number saves RAM)

  #define DEFAULT_MEASURED_FILAMENT_DIA  DEFAULT_NOMINAL_FILAMENT_DIA  //set measured to nominal initially

  //When using an LCD, uncomment the line below to display the Filament sensor data on the last line instead of status.  Status will appear for 5 sec.
  //#define FILAMENT_LCD_DISPLAY
#endif

#include "Configuration_adv.h"
#include "thermistortables.h"

#endif //CONFIGURATION_H

thinkyhead, 2016-03-30:
As noted, that `m402` needs to be uppercase or it won't work. Looking further…

dinizhobby, 2016-03-30:
I TRIED  M402 and m402 not working.

dinizhobby, 2016-03-30:
I WANT ONLY THE SERVO LOWER AND RAISE USING G29 OR G30 AUTOMATICALLY.

Blue-Marlin, 2016-03-30:
DELTA is the main difference to my setup.

dinizhobby, 2016-03-30:
i add this function before if enable (sled)   servo lower ok!

 deploy_z_probe(); 
      
      
    #if ENABLED(Z_PROBE_SLED)
      dock_sled(false); // engage (un-dock) the Z probe
    #elif ENABLED(Z_PROBE_ALLEN_KEY) //|| SERVO_LEVELING
      deploy_z_probe();
    #endif

dinizhobby, 2016-03-30:
and 

 #if ENABLED(DELTA)
      // Allen Key Probe for Delta
      #if ENABLED(Z_PROBE_ALLEN_KEY || HAS_SERVO_ENDSTOPS)
           
        stow_z_probe();
      #elif Z_RAISE_AFTER_PROBING > 0
        raise_z_after_probing(); // ???
      #endif

problem solved  with g29

thinkyhead, 2016-03-30:
> I WANT ONLY THE SERVO LOWER AND RAISE USING G29 OR G30 AUTOMATICALLY

Yeah, I got that, yelly yeller man. I was merely pointing out that one flaw. Do you want help or not?

thinkyhead, 2016-03-30:
I've been looking at the lines you point out. But it's still possible your configuration is off. Which pins are you connecting your endstops and your z probe to?

dinizhobby, 2016-03-30:
Yes you can fix the problem correctly? I only made this change in code.

dinizhobby, 2016-03-30:
I made these changes up and ran the servo to G29.
More do not know if it is correctly

thinkyhead, 2016-03-30:
I'm not sure you should set `Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN` since you have no z min endstop.

dinizhobby, 2016-03-30:
I am using Z_min

dinizhobby, 2016-03-30:
19:54:29.340 : Reporting endstop status
19:54:29.341 : x_min: open
19:54:29.341 : x_max: open
19:54:29.341 : y_min: open
19:54:29.343 : y_max: open
19:54:29.343 : z_min: TRIGGERED
19:54:29.344 : z_max: open

thinkyhead, 2016-03-30:
Great, but you are not also using it for an endstop.

dinizhobby, 2016-03-30:
i made these changes

#if ENABLED(Z_PROBE_SLED)
      dock_sled(false); // engage (un-dock) the Z probe
    #elif ENABLED(Z_PROBE_ALLEN_KEY || HAS_SERVO_ENDSTOPS) //|| SERVO_LEVELING
      deploy_z_probe();
    #endif

and

#if ENABLED(Z_PROBE_ALLEN_KEY || HAS_SERVO_ENDSTOPS)
           
        stow_z_probe();

thinkyhead, 2016-03-30:
These are relatively new options, so I am not myself even sure of their effects. I have to scrub the code some more and see whether your fix applies universally.

dinizhobby, 2016-03-30:
Thank you for your help
Can you help me with this other issue https://github.com/MarlinFirmware/Marlin/issues/3286

thinkyhead, 2016-03-30:
Yes, thank you for repeating your changes. Again, I need to evaluate whether they are good just for you or whether they are good for everybody. Then perhaps it will be that way, or it will be slightly different. Thanks for pointing out this issue. Cheers!

closed 2016-03-30 by thinkyhead

Blue-Marlin, 2016-03-30:
The problem sounds familiar to me
https://github.com/MarlinFirmware/Marlin/issues/2980#issuecomment-181831534

@dinizhobby 
Your proposted patch may work for you, but does break about all other setups.

thinkyhead, 2016-03-30:
That `SERVO_ENDSTOPS` condition has been commented out for a long while. So I need to understand why it's ok for Cartesian but not delta, as I was expecting it to be broken for both after I looked at the code! My eyes are getting crossed. I think the fix proposed here is probably right.

dinizhobby, 2016-03-30:
can anyone help me with this another issue  https://github.com/MarlinFirmware/Marlin/issues/3286
?

thinkyhead, 2016-03-31:
#3288 should do the trick – it has exactly the fix suggested. Plus some other cleanup and correction. (I find it helps to pre-define some things in macros to gather "behaviors" together.)