MarlinFirmware/Marlin #3290: LCD on RC4

MoonshineSG, 2016-03-31:
Although all the lines relating to LCD in Configuration.h are commented out (I am not using any type of LCD on my printer), when trying to compile RC4 i get 

```.build/mega2560/src/Marlin.cpp:9:21: fatal error: LCD.h: No such file or directory```

In `~/marlin-github/.build/mega2560/src/Marlin.cpp`

```
#include <Arduino.h>
#include "Configuration.h"
#include "pins.h"
    #include <Wire.h>
    #include <LiquidCrystal_I2C.h>
    #include <Wire.h>
    #include <LiquidTWI2.h>
    #include <Wire.h>
    #include <LCD.h>
    #include <LiquidCrystal_I2C.h>
    #include <U8glib.h> // library for graphics LCD by Oli Kraus (https://code.google.com/p/u8glib/)
    #include <LiquidCrystal.h> // library for character LCD
  #include <SPI.h>
  #include <Wire.h>
  #include <SPI.h>
  #include <TMC26XStepper.h>
  #include <SPI.h>
  #include <L6470.h>

#line 1 "src/Marlin.ino"
/**
[...]
/* All the implementation is done in *.cpp files to get better compatibility with avr-gcc without the Arduino IDE */
/* Use this file to help the Arduino IDE find which Arduino libraries are needed and to keep documentation on GCode */

//#include "Configuration.h"
//#include "pins.h"

#if ENABLED(ULTRA_LCD)
  #if ENABLED(LCD_I2C_TYPE_PCF8575)
//    #include <Wire.h>
//    #include <LiquidCrystal_I2C.h>
  #elif ENABLED(LCD_I2C_TYPE_MCP23017) || ENABLED(LCD_I2C_TYPE_MCP23008)
//    #include <Wire.h>
//    #include <LiquidTWI2.h>
  #elif ENABLED(LCM1602)
//    #include <Wire.h>
//    #include <LCD.h>
//    #include <LiquidCrystal_I2C.h>
  #elif ENABLED(DOGLCD)
//    #include <U8glib.h> // library for graphics LCD by Oli Kraus (https://code.google.com/p/u8glib/)
  #else
//    #include <LiquidCrystal.h> // library for character LCD
  #endif
#endif

[...]
```

It seams that all the LCD related files are included although not needed... 

Can someone explain/help ?

oxivanisher, 2016-03-31:
I have the same problem. For two includes i found the download URL for the library in the source code,  sadly not for LCD.h.

MoonshineSG, 2016-03-31:
if you don't use LCD, the only way to temp solve this is that after it fails once you go under `.build/mega2560/src` (mega256 might be different for other boards) and erase `#include <LCD.h>` line (about 6th or 7th line) from Marlin.cpp  then build again...

oxivanisher, 2016-03-31:
But i have a LCD. So i can not use RC4 with an LCD?

jbrazio, 2016-03-31:
@MoonshineSG I do not see anywhere the source code you have posted, my `Marlin.ino` file has a condition to include `LCD.h`.

```cpp
#if ENABLED(ULTRA_LCD)
  (...)
  #elif ENABLED(LCM1602)
    #include <Wire.h>
    #include <LCD.h>
    #include <LiquidCrystal_I2C.h>
  #elif ENABLED(DOGLCD)
  (...)
#endif
```

jbrazio, 2016-03-31:
In fact have a look on the RC4 source code on Github:
https://github.com/MarlinFirmware/Marlin/blob/RC/Marlin/Marlin.ino#L40-L56

MoonshineSG, 2016-03-31:
Maybe it wasn't clear. The code above can be found in the generated file '.build/mega2560/src/Marlin.cpp'

Maybe the fact that I build on raspberry using ino makes a difference?

oxivanisher, 2016-03-31:
I try the same thing. But i did not find a LCD.h in a Windows Arduino installatin either. I looks like there is a URL missing to be checked out via git if you want to build it.

jbrazio, 2016-03-31:
Could you please post the actual error log from within the Arduino IDE ?

jbrazio, 2016-04-01:
Could you please post, as an attachment, your config and config_adv files ? Thanks.

oxivanisher, 2016-04-01:
My CFG files: [cfg_files.zip](https://github.com/MarlinFirmware/Marlin/files/199131/cfg_files.zip)

jbrazio, 2016-04-01:
If I take your configuration files and copy them into RCBugFix:

```bash
jbrazio@dagobah:~/Documents/Marlin/Marlin (RCBugFix) $ git status
On branch RCBugFix
Your branch is up-to-date with 'origin/RCBugFix'.
Changes not staged for commit:
  (use "git add <file>..." to update what will be committed)
  (use "git checkout -- <file>..." to discard changes in working directory)

	modified:   Configuration.h
	modified:   Configuration_adv.h

no changes added to commit (use "git add" and/or "git commit -a")
```

I can compile using Arduino IDE:
```
Sketch uses 93,012 bytes (36%) of program storage space. Maximum is 253,952 bytes.
Global variables use 3,710 bytes (45%) of dynamic memory, leaving 4,482 bytes for local variables. Maximum is 8,192 bytes.
```

If I try the "inotool":
```bash
jbrazio@dagobah:~/Documents/Marlin/Marlin (RCBugFix) $ ino build -f="-DUSE_AUTOMATIC_VERSIONING -ffunction-sections -fdata-sections -g -Os -w" 
No project found in this directory.
```

MoonshineSG, 2016-04-01:
For ino to work you need a 'src' folder (I symlink 'src' to 'Marlin'...

oxivanisher, 2016-04-01:
I tried it on a windows machine (fresh installed arduino software) and i was able to compile it from marlin-rc.zip with my config files. This is very strange.

I again searched for LCD.h files on the windows machine and found no such file. Is it something integrated in newer Arduino software versions?

This is a "simple" history of what i tried:
```
sudo pip install ino
sudo apt-get install arduino
cd /home/pi/
git clone https://github.com/MarlinFirmware/Marlin.git
cd /home/pi/Marlin
```
I had to checkout the following two libraries and link them to `/usr/share/arduino/libraries`:
- https://github.com/kiyoshigawa/LiquidCrystal_I2C.git
- https://github.com/lincomatic/LiquidTWI2.git
```
ln -s Marlin src
cp /where/my/configs/are/Configuration*.h /home/pi/Marlin/Marlin/
ino clean && ino build -m mega2560
```
This generates the following output:
```
pi@octopi ~/Marlin $ ino clean && ino build -m mega2560
Searching for Board description file (boards.txt) ... /usr/share/arduino/hardware/arduino/boards.txt
Searching for Arduino lib version file (version.txt) ... /usr/share/arduino/lib/version.txt
Detecting Arduino software version ...  1.0.1 (1.0.1)
Searching for Arduino core library ... /usr/share/arduino/hardware/arduino/cores/arduino
Searching for Arduino standard libraries ... /usr/share/arduino/libraries
Searching for Arduino variants directory ... /usr/share/arduino/hardware/arduino/variants
Searching for make ... /usr/bin/make
Searching for avr-gcc ... /usr/bin/avr-gcc
Searching for avr-g++ ... /usr/bin/avr-g++
Searching for avr-ar ... /usr/bin/avr-ar
Searching for avr-objcopy ... /usr/bin/avr-objcopy
src/Marlin.ino
Searching for Arduino lib version file (version.txt) ... /usr/share/arduino/lib/version.txt
Detecting Arduino software version ...  1.0.1 (1.0.1)
Scanning dependencies of src
Scanning dependencies of arduino
Scanning dependencies of LiquidCrystal_I2C
Scanning dependencies of LiquidCrystal
Scanning dependencies of SPI
Scanning dependencies of Wire
Scanning dependencies of LiquidTWI2
src/SdVolume.cpp
src/buzzer.cpp
src/M100_Free_Mem_Chk.cpp
src/planner.cpp
src/servo.cpp
src/dac_mcp4728.cpp
src/qr_solve.cpp
src/SdFile.cpp
src/mesh_bed_leveling.cpp
src/Marlin_main.cpp
src/MarlinSerial.cpp
src/stepper.cpp
src/temperature.cpp
src/Sd2Card.cpp
src/digipot_mcp4451.cpp
src/ultralcd.cpp
src/watchdog.cpp
src/blinkm.cpp
src/cardreader.cpp
src/SdBaseFile.cpp
src/stepper_dac.cpp
src/SdFatUtil.cpp
src/configuration_store.cpp
src/vector_3.cpp
src/stepper_indirection.cpp
src/Marlin.cpp
.build/mega2560/src/Marlin.cpp:9:21: fatal error: LCD.h: No such file or directory
compilation terminated.
.build/mega2560/Makefile:425: recipe for target '.build/mega2560/src/Marlin.o' failed
make: *** [.build/mega2560/src/Marlin.o] Error 1
Make failed with code 2
pi@octopi ~/Marlin $ 
```
This would then be the final command to upload the firmware:
```
ino upload -p /dev/ttyACM0 -m mega2560 #
```

To verify:
```
pi@octopi ~/Marlin $ git branch -vvv
* RC 122bdd1 [origin/RC] Merge branch 'RCBugFix' into RC
```

MoonshineSG, 2016-04-01:
@oxivanisher, I am not sure I understand... Windows or Pi ? 
Or you tried on both and on windows worked and failed on Pi ?

In my opinion, if the Config files specify no LCD, the libraries shouldn't even be needed...

jbrazio, 2016-04-01:
This seems a non-supported compilation method for Marlin.. nevertheless it should work.

oxivanisher, 2016-04-01:
The second option: Tried on both and on windows worked and failed on Pi

I have this configured in my config:
`#define REPRAP_DISCOUNT_SMART_CONTROLLER`

MoonshineSG, 2016-04-01:
@jbrazio : It's been working so far .. 
@oxivanisher : then somehow Windows has the LCD.h (or it doesnt need it) ... Strange...

oxivanisher, 2016-04-01:
@MoonshineSG Yes, it looks like that it does not need it. I did not find such a file on the Windows installation.

This is the content of the library folder on the pi:
```
pi@octopi ~/Marlin $ ls -lha /usr/share/arduino/libraries
total 48K
drwxr-xr-x 12 root root 4.0K Mar 30 17:25 .
drwxr-xr-x  6 root root 4.0K Mar 30 15:05 ..
drwxr-xr-x  3 root root 4.0K Mar 30 15:05 EEPROM
drwxr-xr-x  4 root root 4.0K Mar 30 15:05 Ethernet
drwxr-xr-x  3 root root 4.0K Mar 30 15:05 Firmata
drwxr-xr-x  3 root root 4.0K Mar 30 15:05 LiquidCrystal
lrwxrwxrwx  1 root root   58 Mar 30 17:23 LiquidCrystal_I2C -> /home/pi/git_checkout/LiquidCrystal_I2C/LiquidCrystal_I2C/
lrwxrwxrwx  1 root root   33 Mar 30 17:25 LiquidTWI2 -> /home/pi/git_checkout/LiquidTWI2/
drwxr-xr-x  4 root root 4.0K Mar 30 15:05 SD
drwxr-xr-x  3 root root 4.0K Mar 30 15:05 Servo
drwxr-xr-x  3 root root 4.0K Mar 30 15:05 SoftwareSerial
drwxr-xr-x  3 root root 4.0K Mar 30 15:05 SPI
drwxr-xr-x  3 root root 4.0K Mar 30 15:05 Stepper
drwxr-xr-x  4 root root 4.0K Mar 30 15:05 Wire
```

MoonshineSG, 2016-04-01:
@oxivanisher can you check the generated `Marlin.cpp` under `.build` folder (not sure where windows puts that ...)

oxivanisher, 2016-04-01:
This is the start of the Marlin.cpp:
```
#include <Arduino.h>
#include "Configuration.h"
#include "pins.h"
    #include <Wire.h>
    #include <LiquidCrystal_I2C.h>
    #include <Wire.h>
    #include <LiquidTWI2.h>
    #include <Wire.h>
    #include <LCD.h>
    #include <LiquidCrystal_I2C.h>
    #include <U8glib.h> // library for graphics LCD by Oli Kraus (https://code.google.com/p/u8glib/)
    #include <LiquidCrystal.h> // library for character LCD
  #include <SPI.h>
  #include <Wire.h>
  #include <SPI.h>
  #include <TMC26XStepper.h>
  #include <SPI.h>
  #include <L6470.h>

#line 1 "src/Marlin.ino"
```

MoonshineSG, 2016-04-01:
So there's `#include <LCD.h>` and it works on Windows... Grrrrr..... 
Anyway, those line shouldn't even be there... Right ?

MoonshineSG, 2016-04-01:
Hmmmm... Super strange thing.... 
I renamed Marlin.ino to Marlin.cpp and tried a clean build. It was successful, and it skipped building and linking any of the LCD related libraries so it ended up being a much faster build AND a much smaller firmware image ! 382962 vs 443722 on the .elf but of course the hex files are both 247k

Marlin.ino build (had to run twice and remove the LCD.h line from the generated LCD.h)
```
drwxr-xr-x 2 root root   4096 Apr  1 22:22 arduino
-rwxr-xr-x 1 root root 443722 Apr  1 22:22 firmware.elf
-rw-r--r-- 1 root root 253409 Apr  1 22:22 firmware.hex
drwxr-xr-x 2 root root   4096 Apr  1 22:21 L6470
drwxr-xr-x 2 root root   4096 Apr  1 22:20 LiquidCrystal
drwxr-xr-x 2 root root   4096 Apr  1 22:20 LiquidCrystal-I2C
drwxr-xr-x 2 root root   4096 Apr  1 22:21 LiquidTWI2
-rw-r--r-- 1 root root 198298 Apr  1 22:20 Makefile
-rw-r--r-- 1 root root   1707 Apr  1 22:20 Makefile.deps
-rw-r--r-- 1 root root    201 Apr  1 22:20 Makefile.sketch
drwxr-xr-x 2 root root   4096 Apr  1 22:21 SPI
drwxr-xr-x 2 root root   4096 Apr  1 22:20 src
drwxr-xr-x 2 root root   4096 Apr  1 22:20 TMC26XStepper
drwxr-xr-x 3 root root   4096 Apr  1 22:21 U8glib
drwxr-xr-x 3 root root   4096 Apr  1 22:21 Wire
```
Marlin.cpp build 
```
drwxr-xr-x 2 root root   4096 Apr  1 22:25 arduino
-rwxr-xr-x 1 root root 382962 Apr  1 22:25 firmware.elf
-rw-r--r-- 1 root root 253409 Apr  1 22:25 firmware.hex
-rw-r--r-- 1 root root  25889 Apr  1 22:24 Makefile
-rw-r--r-- 1 root root  27947 Apr  1 22:24 Makefile.deps
-rw-r--r-- 1 root root     21 Apr  1 22:24 Makefile.sketch
drwxr-xr-x 2 root root   4096 Apr  1 22:25 src
```

I guess I'm adding one line in my build script `mv src/Marlin.ino src/Marlin.cpp` :D

closed 2016-04-25 by thinkyhead