MarlinFirmware/Marlin #3295: Auto Bed Leveling not working

dinizhobby, 2016-03-31:
Hi Guys

I tried enable auto bed level function in my marlin RC4, but i have auto bed level issue. I have a Kossel Mini alt 

Check my picture below

http://imagens.abcdotecnico.com.br/images/129101121162274300473623445228454n.jpg

Roxy-3DPrintBoard, 2016-03-31:
What do you get when you give the printer a:   G29 P 4 V 4 T ????

Blue-Marlin, 2016-03-31:
@Roxy-3DPrintBoard 
This is reality here. You are not in on of your obscure playground 'Valentine's Day' or 'Black-Friday' branches.
In the reality I call 'RCBugFix' or 'RC' , G29 has no P nor T parameters for deltas.

@dinizhobby 
I'd suggest to tune parameters manually for some iterations.

dinizhobby, 2016-03-31:
did you some video or tutorial to share for tuning my delta?

thinkyhead, 2016-04-02:
> did you some video or tutorial to share for tuning my delta?

Sorry, no. Search Google and you may find something independently created.

closed 2016-04-25 by thinkyhead