MarlinFirmware/Marlin #3300: ambient temperature false

taratata2016, 2016-04-01:
as I have Autotune PID problem for 5 days ( Repetier -host gives an ambient temperature of 37 ° C instead of 23 ° C room ) . I change cards ramps - extruding - exposed the ramps with only extrude - ATX 550W 20A 20A power , nothing works. 
you can read my post here[http://www.lesimprimantes3d.fr/forum/topic/1709-prusa-i3-variations-de-temp%C3%A9rature-de-lextrudeur/](url). i hope find solution here. tks in advance for your time

jbrazio, 2016-04-01:
I'm sorry I didn't understood if you're running Marlin as the firmware of your printer.
Have you tried autotune ?
`M303 E0 S200 C8`

taratata2016, 2016-04-01:
yes, i setting marlin and update ramps and start m303 e0 s220 c8 and writing the news value in configuration.h and another update and check the temperature ....

taratata2016, 2016-04-01:
if i plug the thermistor on pins T1 or t2 the temperature is between 18°C to 24°C

Blue-Marlin, 2016-04-01:
Gentleman.
The measured temperature has nothing, absolute nothing, to do with M303.
If you see a difference between T0, T1, T2 with the same thermistor than look for different sensor types.
```
#define TEMP_SENSOR_0 1
#define TEMP_SENSOR_1 1
#define TEMP_SENSOR_2 1
```

taratata2016, 2016-04-01:
Normally, at the opening of repetier-host, the value should be the ambient temperatures. T0 gives 35 ° C. I made so many test, I am lost.

Blue-Marlin, 2016-04-01:
So ether you have a wrong sensor type now, or a defect thermistor, or the pin used by T0 is used by some other function by error, or if your board is new, there is a little chance the resistor at that channel has the wrong value. Any solder chips on your board, causing shorts?

taratata2016, 2016-04-01:
i change another ramps 1.4 - thermistor - on ramps have only thermistor. the ramps is on plank wood

taratata2016, 2016-04-01:
you can read my post herehttp://www.lesimprimantes3d.fr/forum/topic/1709-prusa-i3-variations-de-temp%C3%A9rature-de-lextrudeur/.

thinkyhead, 2016-04-01:
What version of Marlin are you using? Be sure to test with the current `RCBugFix` code to make sure we haven't already patched your issue (assuming it is software, and not hardware or configuration-related). Have you used older versions of Marlin which show a different temperature for the same thermistor?

taratata2016, 2016-04-02:
i use Marlin 1.1.0-RC3 [https://github.com/MarlinFirmware/Marlin](url). I measure the voltage (volts) at terminals T0 T1 T2 = 4.7v - I also tested the continuity multilmètre between terminals 1 3 5 (this is good). have you idea for some test mesuring board? tks in advance for your proipositions

taratata2016, 2016-04-02:
i just test the version "Release 1.1.0-RC5" and ther ambient temperature have sample trouble (35°C).
If i understand well, the " "RCBugFix" branch" is corrections inside the version RC5? i'm sur about spare parts, because i have many thermistor, 2x ramps and i test only with thermistor plug on borad. before reflash arduino mega, i reset Mega with this code `void setup(){}
void loop(){}`.

taratata2016, 2016-04-02:
[image]

taratata2016, 2016-04-02:
why Marlin writing wrong ech debug
[image]

taratata2016, 2016-04-02:
continuity test between A15 and PIN6 (T2)  -  A14 and PIN4 (T1) - A13 and PIN2 (T0) is ok!
now i must find the R7 R1 R11 on board ramps1.4 for check welds "les soudures"...

thinkyhead, 2016-04-04:
It sure looks like a hardware problem. The same thermistor should read the same resistance (hence, temperature) regardless of the connector to which it's attached. Unless there is something about the connector that's different – i.e., a different resistance. If `T0` turns out to be problematic in hardware, you can just change the pin assignments and use a different connector.

thinkyhead, 2016-04-04:
> Why is Marlin writing wrong echo debug?

A bug in the way the `M111` function was fetching a pointer to PROGMEM. This is patched in #3343.

Roxy-3DPrintBoard, 2016-04-04:
I don't like 'Harvard' architecture either!

taratata2016, 2016-04-04:
Thank you for your answer! The thermistance work well, because when i plug Melzi board (original board of printer hensine prusa i3), i have 25°C. ONly with my 2 board test ramps 1.4 have same mistake temperature 35°C.

taratata2016, 2016-04-05:
you can watch, how i do autotune pid

voci comment je fais

https://www.youtube.com/channel/UCKfChdFFkV-OBNG9AL7t2Xw

https://youtu.be/WHFz9gj46Us    0 Power Supply ON TempRoom

https://youtu.be/is8C26_kwcs      0 - Thermistor 100k B3950
    
https://youtu.be/p4X1Uoucxlk         1 - Marlin configuration.h
    
https://youtu.be/PYLKjq886fw       2   ArduinoIDE Upload Marlin Firmware
    
https://youtu.be/iAIqqbmRyTI        3 - Start AutoTune PID
    
https://youtu.be/JUlfZZftNd4          3 - Start AutoTune PID bis
    
https://youtu.be/BZmkM42QJ90    4 - New Value PID
    
https://youtu.be/ywgDQE7bvpo      5 - Extruder is Cold
    
https://youtu.be/2I3NcQ_UqwY       6 - Wrong Room Temperature
    
https://youtu.be/ifFH3e0tct4           6 - Wrong Room Temperature bis

Blue-Marlin, 2016-04-05:
Again.
As long you get false room-temperature, autotune is absolutely clueless.

Your enormous amount of noise on the temperature also indicates - there is something wrong. 
My tests with todays RcBugFix, Mega/RAMPS, sensor 11, give a stable (normal) room temperature with noise better than +- 0.2°C.

Recheck your hardware.

taratata2016, 2016-04-05:
the rcbigfix can download her [ https://github.com/MarlinFirmware](url)! is right?
i don't know what is wrong in my tests! so crazy

taratata2016, 2016-04-05:
the hardware work fine with Melzi v2.0 board with repetier firmware.

Blue-Marlin, 2016-04-05:
Try to minimize the system. Mega, RAMPS, USB-cable, one of your spare thermistors - noting else - no power supply, no stepper drivers, no heaters, no display - nothing.
If that works, readd part by part.

taratata2016, 2016-04-05:
sur i test with only thermistor and power suppy and cartridge heaters. I find the prob. I check and raplace all spare parts and use with minimze system. But i forget to check the Arduino Mega. I replace it and "Ho my god - tks so much) - work so fine now. The Arduino Mega is HS=out of order (KO)...
thank you at all people use time to help me...

closed 2016-04-05 by taratata2016