MarlinFirmware/Marlin #3302: thermal runaway false positve trigger

Sungod3000, 2016-04-01:
Hi,

take this with a grain of salt because I had the same issue in the last marlin (RC1) so I disabled it but now the default settings are triggering print stop ca 1 min into printing. I extended the parameters to 90secs and 10C but still got the same shutdown time.

thinkyhead, 2016-04-01:
If you're testing with `RCBugFix` the defaults to check fast-enough heating have been relaxed a little. It used to be 4 degrees in 16 seconds. Now it's more like 2 degrees in 20 seconds. And we've patched some bugs around the wait-for-heating routines too. Overall it should be better now, so give it a try and see if the new defaults are good enough on your machine.

maxwondercorn, 2016-04-04:
The latest `RCBugFix` commit fixed these issues for me.

closed 2016-04-04 by thinkyhead

jbrazio, 2016-04-04:
@maxwondercorn may I suggest to set `PID_FUNCTIONAL_RANGE` to `250` and repeat the tests ? The default value of `10` means that the PID is only active when the temp is `±10` from the setpoint.

maxwondercorn, 2016-04-04:
@jbrazio Okay, I was literally up all nite working on this and figured it out.  Definitely a hard and probably a stupid lesson... 

I rolled back to RC2 and the problem was still occurring even though it wasn't a problem before. While playing around I sliced a part in a different slicer (Slic3r) and the problem went away.    Tried again with normal slicer - Matter Control - problem came back. More mystery - how can a slicer make a difference?

As I was physically looking over the printer for the hundredth time I noticed my parts fan had gotten knocked aside and was pointing directly at the heater block and it was RUNNING! 

I have a very large printer and have to go behind it to even see if the part fan is running.   Since it only really runs when doing bridges I just let it point wherever until I know it's going to run.  Matter Control (normal slicer) had an update a week or so back and now it was running the fan starting on the second layer where it hadn't been before!  Same parts, different behaviour.

I've been looking at this problem for a while and can't believe I missed this.  The good part is I did learn a lot more about the inner workings of Marlin, so I guess that'll make me a better troubleshooter...

thanks.

jbrazio, 2016-04-04:
Congrats ! My next step would be tell you to increase your heater cartridge, the fan totally explains it (I would never guess it).