MarlinFirmware/Marlin #3323: Movements problems on G29

pyroboynroses, 2016-04-02:
Hello,

I'm using the last version of stable Marlin 1.1.0 RC3 with autoleveling feature activated.

I've a weird behavior of my printer when doing a G28 / G29. The problem is that it's not a always problem. It can happen sometimes, or not...

In fact, my printer is doing the G28 and sometimes the G28 on the X axis is blocked after a few centimeters, like the endstop has been hit (but the carriage is more than 10cm to the endstop).
So, I make an emergency stop and then retry, the problem is gone on the next time.

So, then after, when doing the G29, sometimes the X axis, or Y axis moves less than it has to do (I request a 9 points grid, assumes that the distance beetween each point is 12cm, sometimes it is blocked on one move, and the movements is just 4 or 5 cm, maybe less...

So, I tried to figure out why, tested with LCD screen, directly over USB, same type of problem, thinking it happens about 30 or 40 % frequency.

So, the problem talked in the issue https://github.com/MarlinFirmware/Marlin/issues/2109 made me think it's about the same so I changed my configuration.h file to deactivate the max endstop pullups but the problem is the same.

You'll find my configuration.h attached if you've any clue, should be very helpful.

BTW, thanks for the RC3, the autobed is slightly improved :-)

[Configuration.txt](https://github.com/MarlinFirmware/Marlin/files/200898/Configuration.txt)

jbrazio, 2016-04-02:
Hi, thank you for you report.

> 1.1.0 RC3

Latest version is RC5.

Have you ruled out any possible noise on the wires going to the endstop ? This seems to me a probable cause for it to be triggered without any reason.

pyroboynroses, 2016-04-03:
Hello,

Thank you for such your answer. In fact, the printer I make has a lot of metal plates, and maybe it can be an issue indeed. But difficult to test because it's part of the printer.

The problem occurs more often on the X axis, which the cable for the endstop is in a pipe with the NEMA23 stepper 4 wires. Maybe it can be a info.

Is there any chance to change the minimum detect time in Marlin, when an endstop is triggered ?

Thank you,

pyroboynroses, 2016-04-03:
Just tested to manually trigger the X endstop during the G29 process : it's not doing everything, the movement is continuing to the next position.

So, the problem should be elsewhere.

jbrazio, 2016-04-03:
False triggers is a well known problem specially when the endstop cable runs very close to the stepper cable without any protection. This is something you should try to fix in hardware.

You could try adding a ferrite bead, twisting the two cables to the end switch or in really bad scenarios introduce a filter like LC or RCL to cut noise.

jbrazio, 2016-04-03:
Btw, just for the sake of mind could you please try the latest RCBugfix instead of RC3 ?

thinkyhead, 2016-04-09:
It's possible the endtops are configured in a bad way. You will find it much easier to configure them with the current `RCBugFix` code, for sure.

pyroboynroses, 2016-04-13:
Hello,

Thank you for your answers. I think I've figured out why I'd this problem.

I've on the Mega and USB extender to plug the shielded USB cable in the back of the printer. This USB extender is not shielded and so, I've some transmission problems I guess.

I've removed this extender (like this one : http://fr.aliexpress.com/item/CY-50cm-90-Degree-Right-Angled-USB-B-Type-Male-to-Female-Extension-Cable-with-Screws/1885745703.html) and now the problem seems to be solved (45 times G28 and G29 sequence without any problem.

But, Do you have any idea why it can fail on the G28 and G29 as I never experienced problems during a print ?

Thank you,

jbrazio, 2016-04-13:
During print I believe that the "software endstops" prevail unless otherwise configured.

thinkyhead, 2016-04-14:
Electronic "noise" can lead to all kinds of strange things, so it's hard to say what caused your issues. Maybe it's because `G28` and `G29` move at "homing speed" and your stepper motors can't go that fast if there's noise, but have no problem going that fast without the noise. We can only guess without more rigorous testing and experimentation.

CONSULitAS, 2016-04-14:
@pyroboynroses 
> This USB extender is not shielded and so, I've some transmission problems I guess.

I suppose you can fix your cable with a "ferrite cable clip" like this:
http://www.ebay.co.uk/itm/UF35-3-5mm-Inner-Diameter-Ferrite-Filter-Cable-Clip-for-RFI-EMI-/191656570537?

You can get them with different diameters matching your cable. It filters noise on the line even on rather poor cables.

I have a cheap 7" tablet pc which shows strange and silly effects on the touchscreen if the ac adaptor connected. With this clip everything is fine... 😄

CONSULitAS, 2016-04-14:
@thinkyhead @jbrazio @Roxy-3DPrintBoard 
Perhaps we need a FAQ for things like this?

jbrazio, 2016-04-14:
This is more hardware domain but yes it's a enough common issue that can go into our documentation, how about inside the `Getting Started` document near the endstop configuration ?

pyroboynroses, 2016-04-14:
Thank you guys for taking importance about this "issue".

I'll test the ferrite to make sure the problem is solved this way.

BTW, as 3D printer manufacturer, I'll keep informed about new things if I find such.

Have a great day, both of you.

epatel, 2016-04-14:
I would check the pull-up resistors. Though from the config they seem to be enabled. Is it a real RAMPS board or a derivate?

pyroboynroses, 2016-04-14:
Good point epatel. It's real RAMPS we use.

I've spoken with an electronics ingeneer and he recommends for testing to add extra pullups resistors on endstops, of about 470 ohms to 1 kohms, to have more current inside the endstop, and to avoid false positive measures.

What do you think ?

epatel, 2016-04-14:
I also see there is a new RAMPS out 1.4.2 http://reprap.org/wiki/RAMPS_1.4#RAMPS_1.4.2 that seem to have added some cap for the endpoints. Well, is is also recommended to not have the cables near the motor leads. http://reprap.org/wiki/Mechanical_Endstop

closed 2016-04-15 by thinkyhead