MarlinFirmware/Marlin #3325: for those of you with the rumba board,getting a DEFAULT_SOURCE_URL error

Ky1duck, 2016-04-02:
for those of you using the rumba board and getting a DEFAULT_SOURCE_URL there is a line missing in the LANGUAGE_H file

if reads 

if MB(ULTIMAKER)|| MB(ULTIMAKER_OLD)|| MB(ULTIMAIN_2)
  #define DEFAULT_MACHINE_NAME "Ultimaker"
  #define DEFAULT_SOURCE_URL "https://github.com/Ultimaker/Marlin"
#elif MB(RUMBA)
  #define DEFAULT_MACHINE_NAME "Rumba"
#elif MB(3DRAG)
  #define DEFAULT_MACHINE_NAME "3Drag"
  #define DEFAULT_SOURCE_URL "http://3dprint.elettronicain.it/"

all I did was add a line so it looked like this 

if MB(ULTIMAKER)|| MB(ULTIMAKER_OLD)|| MB(ULTIMAIN_2)
  #define DEFAULT_MACHINE_NAME "Ultimaker"
  #define DEFAULT_SOURCE_URL "https://github.com/Ultimaker/Marlin"
#elif MB(RUMBA)
  #define DEFAULT_MACHINE_NAME "Rumba"
  #define DEFAULT_SOURCE_URL "http://reprap.org/wiki/RUMBA"   // added this line. 
#elif MB(3DRAG)
  #define DEFAULT_MACHINE_NAME "3Drag"
  #define DEFAULT_SOURCE_URL "http://3dprint.elettronicain.it/"

hope it helps ya'll

robert #

jbrazio, 2016-04-02:
Thank you for your report, a PR has been submitted.

thinkyhead, 2016-04-03:
Patched with #3331, also addressing future problems.

closed 2016-04-03 by thinkyhead