MarlinFirmware/Marlin #3332: Rumba Board selection have error

guy2b1, 2016-04-03:
Hi guys, im about to use the latest Marlin Firmware release March 2016 for my Kossel using Rumba board.  But it has error when I verify before I upload, what is this error mean?  
Marlin_main.cpp: In function 'void gcode_M115()': Marlin_main.cpp:4677: error: expected `)' before 'DEFAULT_SOURCE_URL' 
But if i change the configuration.h board back to BOARD_RAMPS_14_EFB, no errors found to verify, it seems there is an error when i select and edit into the BOARD_RUMBA

fbarcenas, 2016-04-04:
Fixed in the last RCBUGFIX, please update!

guy2b1, 2016-04-04:
Still have error problem verifying the code when I change the setting the board to RUMBA like below:

#ifndef MOTHERBOARD
  #define MOTHERBOARD BOARD_RUMBA
#endif

But if I change it back to its original, no problem verifying the code:

#ifndef MOTHERBOARD
  #define MOTHERBOARD BOARD_RAMPS_14_EFB    // BOARD_RUMBA
#endif

See screen capture attached
[image]

fbarcenas, 2016-04-04:
Arduino 1.6.8??
I have this

guy2b1, 2016-04-04:
Im using Arduino 1.0.5 to verify the code before uploading the firmware to the Rumba board.

jbrazio, 2016-04-04:
@guy2b1 Could you please post you entire `Configuration.h` and `Configuration_adv.h` ?
Please also validate if you're using the latest `RCBugfix` which can be downloaded from [here](https://github.com/MarlinFirmware/Marlin/archive/RCBugFix.zip).

guy2b1, 2016-04-04:
Sorry never tried RCBugfix will download now and test it.   I use the default downloaded configuration.h and configuration_adv.h, i only change the board to BOARD_RUMBA thats it then verify then there is an error.  Never used RCBugfix will try it now.  Thanks :)

guy2b1, 2016-04-04:
RCBugfix working now if using Rumba Board by changing below

#ifndef MOTHERBOARD
#define MOTHERBOARD BOARD_RUMBA    // BOARD_RAMPS_14_EFB
#endif

Thank you very much guys :)  Now will try to edit the configuration.h settings based from my existing Kossel Delta setup.

jbrazio, 2016-04-04:
Great, have a look at the `example_configurations\delta` folders where you'll have some kossel examples. If you feel this issue is solved, please close the ticket, thank you.

guy2b1, 2016-04-05:
I able to verify and upload the firmware successfully using RCBugFix version, but now the new problem is Auto Bed Leveling using G29, it has no response or movements should I close this ticket?  As now I am having Auto Bed Level problem, or anyone can point me to the right link?  Thank you very much guys for all your help and support :)

CONSULitAS, 2016-04-05:
@guy2b1 
perhaps this helps: https://github.com/epatel/Marlin/blob/75f9fb50c4002f6d4ba0345f5594cbac1fab1fa6/Documentation/MeshBedLeveling.md

guy2b1, 2016-04-06:
Thank you all, closing this ticket :)

closed 2016-04-06 by guy2b1