MarlinFirmware/Marlin #3333: Marlin -Ramps 1.4 remapping drivers

Ovaday, 2016-04-03:
Hello, I have one question, how to remap pins, if I have X stepper driver in Z spot and Z stepper driver in X spot? 
Now I'm just replace letters in gcode, but if it could be made by firmware, it'll be much better.

Sorry if it is in wiki, I didn't found it
Thank you

jbrazio, 2016-04-03:
This maybe a stupid questions from me but can't you switch the stepper plugs between Z and X ?

Ovaday, 2016-04-03:
@jbrazio, No, I can't. RAMPS for me - CNC controller, that is installed on the main frame. There is 9-pin sockets with special appointment for each. So I can't just unplug one and insert another
[image]

Blue-Marlin, 2016-04-03:
Have a look into your boards pin-file.

jbrazio, 2016-04-03:
As advised by @Blue-Marlin, you should go into your board pin file and switch the output pins for X and Z, this should get you in the ballpark.

As this is not an issue directly related with Marlin when possible please close this issue, thanks.

Ovaday, 2016-04-03:
@jbrazio @Blue-Marlin Thanks for you! It's helped me! Hope it'll work well

jbrazio, 2016-04-03:
You're welcome.

closed 2016-04-04 by thinkyhead