MarlinFirmware/Marlin #3334: After the LCD interface is stopped, the model fan is closed.

liangruiben, 2016-04-03:
static void lcd_sdcard_stop() {
  quickStop();
  card.sdprinting = false;
  card.closefile();
  autotempShutdown();
  cancel_heatup = true;
  fanSpeed=0;
  lcd_setstatus(MSG_PRINT_ABORTED, true);
}

thinkyhead, 2016-04-09:
I'm not sure if Marlin should disable the fan whenever anyone stops an SD Card print, necessarily. Is this what other firmware like Repetier does? If so, perhaps we should turn off the fan for consistency.

CONSULitAS, 2016-04-09:
@thinkyhead 
I am sure we shouldn't! There is a GCODE for that and if you want to disable the fan after SD print, we have to put it at the end of the file.

Everything else is a surprise for the user and leads to unpredictable results.

Automatics are good for safety (like thermal runaway protection), but if i want to disable the fans, i will do it on my own.

Just my 2¢.

thinkyhead, 2016-04-10:
@CONSULitAS Of course I agree. I mistook the call to `autotempShutdown()` above for `disable_all_heaters()`. So, my question was actually off-base. The fan should not be shut off here.

closed 2016-04-10 by thinkyhead

liangruiben, 2016-04-10:
@thinkyhead @CONSULitAS 
I think you misunderstood my meaning,
In the middle of the printing press stop printing, LCD model of fan is not closed, if the LCD select "PRINT FROM SD", the PRINT head can cause heating at a slower pace because fans did not shut down, serious will never heated to the specified temperature
I am a professional 3 d printing contract, because the problem is often caused by heating failure, previous solution is to close the heating in the LCD, the "PRINT again FROM SD"

CONSULitAS, 2016-04-10:
@liangruiben 
No problem. I try to keep my language simple for your translator.

Just put `M107` in the header of the GCODE. You can configure this. In the "Start G-code" of the printer settings. You find it in your slicer.
* If the fan is off, it does nothing.
* If the fan is on, it switches off.

Then do `M109` and heating is fast even after failure.

liangruiben, 2016-04-10:
我尝试过这么做，但是CURA和Simplify3D加入的任何代码都不会在第一行，
M190 S60.000000
M109 S215.000000
永远都在第一位

I tried to do so, but CURA and Simplify3D to join any code won't be in the first row,
M190 S60.000000
M109 S215.000000
Never in the first place

liangruiben, 2016-04-10:
Forgive my poor English, can only help you find a BUG and try to solve, and to reflect to you, Can't help you deal with the problem raised by other people, because look not to understand

CONSULitAS, 2016-04-10:
good morning (correct time)?

you are welcome

try
```
M107 ; switch off fan
M190 S60.000000
M109 S215.000000
```

look at http://reprap.org/wiki/Start_GCode_routines -> Cura 15.06
>you can still do it by editing the machine specific .json file for your printer

They use M107, too:
>G21 ;metric values
G90 ;absolute positioning
M82 ;set extruder to absolute mode
M107 ;start with the fan off
G28 X0 Y0 ;move X/Y to min endstops
G28 Z0 ;move Z to min endstops
G1 Z15.0 F9000 ;move the platform down 15mm
G92 E0 ;zero the extruded length
G1 F200 E3 ;extrude 3mm of feed stock
G92 E0 ;zero the extruded length again
G1 F9000 ; Put printing message on LCD screen
M 117 Printing...

liangruiben, 2016-04-10:
早上好，我这里是夜间4点，我因为这个问题睡不着
Good morning, I am here is at 4 o 'clock at night, I couldn't fall asleep because of the problem

我使用的是15.04，15.06在我的机器上太慢了，
What I use is 15.04, 15.06 on my machine is too slow,

我照你所说的操作过，确定M107不会出现在第一行
I do as what you said to use, to determine M107 won't appear in the first row

我尝试在Start/end-Gcode start.gcode上添加，但是没有出现我希望的结果
I tried to Start/end - Gcode Start. Gcode on to add, but not I want results

liangruiben, 2016-04-10:
Thank you, please forgive me for not using more statements, because I don't know translation software can convey my meaning

CONSULitAS, 2016-04-10:
OK, good night then. 😄 We have 22:40 in the evening.

Other ways:
* Tutorial for Simplify3D http://matthewtmead.com/blog/tweak-simplify3d-gcode-start-script/
* Have you tried Slic3r? It is easy to change the start gcode in Slic3r.
* Reset printer after abort?

liangruiben, 2016-04-10:
Simplify3D 我测试过，确认无法使用这个办法
Simplify3D I've tested and make sure can't use this way

SLIC3R还没有测试过，因为我还没有掌握正确的参数
SLIC3R haven't tested, because I haven't the right parameters

重置打印机可以，但是再次打印会增加预热的时间
Reset the printer can be, but again print will increase the time of preheat

所以我非常确认我的修改是正确的
So I am very sure my change is correct

CONSULitAS, 2016-04-10:
Slic3r is easy to configure.
Start with default parameters.
Look in Cura and copy parameters.

CONSULitAS, 2016-04-10:
@liangruiben Good night now.

liangruiben, 2016-04-10:
@CONSULitAS  Good night now.

thinkyhead, 2016-04-12:
> I am very confirm this is a BUG

I am very confident that we should not turn off the fan in this case as a general behavior of Marlin. Whether we like it or not, the defined behavior when forcing an SD print to stop "Stop SD Print" is to leave heaters on and to leave the fans on.

Preserving the "current state" of the machine.

Maybe the LCD should ask "Disable heaters and fan?" afterward. That would be friendly. But I think it is correct to leave the machine in the same state (heaters and fans on) by default.

jbrazio, 2016-04-12:
Agreed, if you want to stop them just use the available G-Codes.

liangruiben, 2016-04-12:
那能否在按下PRING FROM SD打印文件时，先热行关闭风扇？
That can press the "PRING FROM SD" print file, the first hot line close the fan?

还原初始状态
Restore the initial state

jbrazio, 2016-04-12:
Fan, heaters status *before* and *after* each print shall be customized on the slicer start and stop g-code section.

liangruiben, 2016-04-12:
问题是，切片软件不能通过加入启动代码先关闭风扇再加热
The problem is that the slicing software does not start by adding code to turn off the fan heating again

CURA15.4和simplify3D都不可以
CURA15.4 and simplify3D can't

@jbrazio

CONSULitAS, 2016-04-12:
>那能否在按下PRING FROM SD打印文件时，先热行关闭风扇？
>That can press the "PRING FROM SD" print file, the first hot line close the fan?

I think @liangruiben tries to propose a `abort.g` on the sd card like `auto0.g`. This would be executed automatically after every abort from sd printing. A nifty and flexible idea!

But then we would perhaps want a `start.g` and `end.g` executed before and after the selected SD file, too.

>Fan, heaters status before and after each print shall be customized on the slicer start and stop g-code section.

This won't work for an abort in mid print. The machine never reads the tail of the GCODE.

>CURA15.4 and simplify3D can't

According to
>http://reprap.org/wiki/Start_GCode_routines -> Cura 15.06

there seems to be indeed a restriction to edit the header and tail in Cura. :disappointed:.

So the  `start.g` and `end.g` files would work arround this slicer issue.

liangruiben, 2016-04-12:
@CONSULitAS 
是的，PRINT ABORT是不会关闭风扇的，start.g不能自动关闭风扇，然后加热
Yes, the PRINT ABORT will not close the fan, start. G cannot automatically shut off the fan, and heating

问题就在于这里
Here lies the problem

如果我在SD打印，中止打印，我再重新SD打印，加热的时候，风扇还在启动，影响加热
If I were in SD print, stop printing, SD again I print, heating time, fans also at startup, affect heating

liangruiben, 2016-04-14:
@CONSULitAS Cura 15.06 Cannot download, back to the 15.4.5

liangruiben, 2016-04-14:
我购买了一个树莓派 3，“Repetier-Server” 真的很难使用，我想开发一个开源的3D打印服务器，该如何开始
I bought a raspberry pi 3, "Repetier Server -" it's really hard to use, I want to develop an open source 3 d print Server, how to start

thinkyhead, 2016-04-15:
> I want to develop an open source 3 d print Server, how to start

It will be easier to start from an existing project and make your own changes, rather than trying to build your own. At least study the code of existing hosts to see how they work.

Start with the code of [Printrun / Pronterface](https://github.com/kliment/Printrun) or [Cura](https://github.com/Ultimaker/Cura) if you like writing Python.
Start with the code of [Repetier-Host](https://github.com/repetier/Repetier-Host) if you like writing C#.

liangruiben, 2016-04-15:
@thinkyhead Octoprint Can also control how 3d printer?

I'll c #, python I can learn

thinkyhead, 2016-04-15:
@liangruiben [Check this link](http://lmgtfy.com/?q=octoprint)