MarlinFirmware/Marlin #3344: Inconsistent endstop results with USE_XMIN_PLUG and X_MIN_ENDSTOP_INVERTING = false

samwyze, 2016-04-04:
With latest rcbugfix I am getting inconsistent triggers when using normally closed style microswitches on my x and y axis. Z axis, which uses a proximity sensor in normally open configuration works fine. I replace x and y with 3-wire normally open mech endstop (as available at makerbot) results are fine. Am I missing a setting in the new config (now that config is using use_xmin_plug layout)?

thinkyhead, 2016-04-04:
@samwyze Thanks for asking about this. I haven't given probe configurations a test with the new options, so your feedback will help to get these new settings right. At the moment, how many switches are attached, and what are they attached to? What kind of results do you get from testing with `M119`?

thinkyhead, 2016-04-04:
I think I have it. The `Conditionals.h` section for pullups should be using `ENABLED` instead of `DISABLED`. The double-negatives apparently confused me…
```cpp
  /**
   * Set ENDSTOPPULLUPS for unused endstop switches
   */
  #if ENABLED(ENDSTOPPULLUPS)
    #if ENABLED(USE_XMAX_PLUG)
      #define ENDSTOPPULLUP_XMAX
    #endif
    #if ENABLED(USE_YMAX_PLUG)
      #define ENDSTOPPULLUP_YMAX
    #endif
    #if ENABLED(USE_ZMAX_PLUG)
      #define ENDSTOPPULLUP_ZMAX
    #endif
    #if ENABLED(USE_XMIN_PLUG)
      #define ENDSTOPPULLUP_XMIN
    #endif
    #if ENABLED(USE_YMIN_PLUG)
      #define ENDSTOPPULLUP_YMIN
    #endif
    #if ENABLED(USE_ZMIN_PLUG)
      #define ENDSTOPPULLUP_ZMIN
    #endif
    #if ENABLED(DISABLE_Z_MIN_PROBE_ENDSTOP)
      #define ENDSTOPPULLUP_ZMIN_PROBE
    #endif
  #endif
```

samwyze, 2016-04-04:
Hi Scott,
I'll have a look at this later.
When I was troubleshooting, holding the endstop closed and using m119 would
report open most of the time, randomly swapping to triggered so long as I
hadn't release the end stop in a while, but would then return to open on
further m119 requests. There were no false positives though (shorting
signal and positive with a jumper always reports open as expected.)

samwyze, 2016-04-04:
Just to confirm, changing conditionals.h did the trick, thanks.

closed 2016-04-04 by samwyze