MarlinFirmware/Marlin #3345: G2/G3 return to origin

drmike8888, 2016-04-04:
I am using Marlin version: ROBO3DR1V3_06112015 on the ROBO3D+.  I created my own gcode using G2/G3 commands, and while the execution of the arcs appears correct, at the end of every arc the print head goes to the origin.  I do the following:
`G0 Z0.5 F2000
G90
G0 X64.300  Y41.000 
G91
G0 Z-0.5 F720
G2 X5.400   Y0.000   I2.7 J0 E0.665  
G2 X-5.400  Y0.000   I-2.7 J0 E0.665  
G1 X-.4 E0.031  
G2 X6.200   Y0.000   I3.1 J0 E0.764  
G2 X-6.200  Y0.000   I-3.1 J0 E0.764  
G0 Z0.5 F2000
`
The arcs them selves appear smooth, so the execution of the arc is correct.  However, the return to origin makes no sense.  Since it is in relative mode, the G1 move should go back for the second circle.  This code should generate two concentric circles, in relative mode.  Instead, the second arc is drawn at the origin, and the first arc is over shot until a line that intersects the circle and the origin is executed.  I do not see anything obvious in the Marlin_main.cpp or motion_control.cpp which would cause this.  

What can I do to help debug this problem?
Thanks!
Mike

thinkyhead, 2016-04-04:
@drmike8888 Thanks for the report. I don't know how many people are using `G2`/`G3` arcs, but I guess this is an area that needs more testing. My guess is that the arc functions are forgetting to account for relative mode, but it might be something more subtle, as you suggest. I'll take a look at the arc code and see if there's anything obviously strange about it.

thinkyhead, 2016-04-04:
I notice that there is an option `SF_ARC_FIX` that is intended to deal with "bad" arc gcode generated by SkeinForge. It apparently just always treats destination coordinates as relative, so it would probably have no effect on your issue to enable it. But I thought it was worth mentioning, in case you wanted to try it out.

thinkyhead, 2016-04-04:
Incidentally, are you using a Cartesian, CoreXY, or Delta machine?

thinkyhead, 2016-04-04:
I have some time tonight, so I will mess with `G2`/`G3` and see if anything shakes out.

drmike8888, 2016-04-04:
>     <reply+005b83cb0161b8c03adefe9f9f2b611819a862a3f7fe8bc892cf000000011319a3c
>     092a169ce08ad3268@reply.github.com>
>
> Incidentally, are you using a Cartesian, CoreXY, or Delta machine?
>
> ---

Howdy Scott,

I'm using a RoBo3d+, it uses machine code 33 in the Configuration.h file.

I did see the comment in the code about the gcode generation problem, but 
I also saw that was fixed some time last year on another wiki page.  I'm
definitely flipping back and forth between relative and absolute modes - I 
use absolute to drive to the edge of the circle, then flip to relative to
do the circle (well, two half circle arcs).

I very much appreciate your looking into it!
Mike.

drmike8888, 2016-04-05:
I tried a newer version and when sending Gcode by minicom, every G2 executed perfectly as expected.  I then loaded my full file using MatterControl, and it did the return to zero on every arc.
So there is nothing wrong with the Marlin code - some how MatterControl is screwing up the gcodes.  I will abandon MatterControl and either send the codes using a simple serial driver or find something that does not interfere with the transmitted data.

Thank you for taking the time to investigate - I think other people have seen similar problems, but failed to isolate where the problem really lies.

You can close this ticket - with my code anyway the Marlin on Robo does what I expect when the codes are properly sent.
Mike

thinkyhead, 2016-04-09:
Would you mind just posting the gcode samples so we can compare them? (If the Mattercontrol code is what you already posted, then we just need the working code to look at.) I'm not sure if it matters, but slicers should avoid using `G0` and stick with `G1`. The `G0` code is for "raw, uninterpolated" moves and should never be used in regular printing.

drmike8888, 2016-04-09:
The gcode is generated by my own C program, the above is straight from the text file I create.  The G0 is used for fast move to a place where I will start printing.   I do not have retraction yet - you can see that in the above sample.  This is now a problem which I need to address.  But I can certainly change G0->G1 everywhere, that is one command!

I found that Ocotoprint can send the gcodes with no problems.  The Robo3D executes every move precisely with all of G0, G1, G2 and G3.  The Marlin code on the printer is working fine (and I am using the RC4 (or RC5 according to "WheresWaldo" on the RoBo3d forums).  

It is interesting that Octoprint can not draw G2 and G3, which is similar to MatterControl.  But it does not screw up the transmission of the code, which MatterControl some how does.  In fact to debug my gcode generator program I had to write another program which would create png files of the trace just so I could see what the nozzle would do. 

I reiterate, this is NOT a Marlin problem.  The G2/G3 codes are performed flawlessly on the latest RC when sent with Octoprint or minicom.  I will change G0->G1 and not worry about it, but it does not seem to cause problems.  Lack of retraction - that is now my problem!  

The control code you guys wrote is working fine - you can close this ticket.   
Mike

closed 2016-04-10 by thinkyhead