MarlinFirmware/Marlin #3346: Garbled Output

maxwondercorn, 2016-04-04:
Just updated from RC5 to BugFix to see if it fixes all the thermal faults I'm having.  Now I'm getting garbled output in the Repitier log window.  Also, I starting seeing the "echo: " prefix in RC5 - I'm assuming that's a response to a request from the host software?

>   10:49:02 PM: Attempting to connect to printer
  10:49:02 PM: Connection opened
< 10:49:11 PM: start
< 10:49:11 PM: echo: External Reset
< 10:49:11 PM: Marlin 1.1.0-RC5
< 10:49:11 PM: echo: Last Updated: 2016-04-01 12:00 | Author: (GAM, Bug Fix)
< 10:49:11 PM: Compiled: Apr  3 2016
< 10:49:11 PM: echo: Free Memory: 2807  PlannerBufferBytes: 1232
< 10:49:11 PM: echo:Hardcoded Default Settings Loaded
< 10:49:11 PM: echo:Steps per unit:
< 10:49:11 PM: echo:  M92 X160.00 Y160.00 Z396.00 E94.50
< 10:49:11 PM: echo:Maximum feedrates (mm/s):
< 10:49:11 PM: echo:  M203 X300.00 Y300.00 Z5.00 E25.00
< 10:49:11 PM: echo:Maximum Acceleration (mm/s2):
< 10:49:11 PM: echo:  M201 X3000 Y3000 Z100 E10000
< 10:49:11 PM: echo:Accelerations: P=printing, R=retract and T=travel
< 10:49:11 PM: echo:  M204 P3000.00 R3000.00 T3000.00
< 10:49:11 PM: echo:Advanced variables: S=Min feedrate (mm/s), T=Min travel feedrate (mm/s), B=minimum segment time (ms), X=maximum XY jerk (mm/s),  Z=maximum Z jerk (mm/s),  E=maximum E jerk (mm/s)
< 10:49:11 PM: echo:  M205 S0.00 T0.00 B20000 X20.00 Z0.40 E5.00
< 10:49:11 PM: echo:Home offset (mm):
< 10:49:11 PM: echo:  M206 X0.00 Y0.00 Z0.00
< 10:49:11 PM: echo:Material heatup parameters:
< 10:49:11 PM: echo:  M145 S0 H180 B70 F0
< 10:49:11 PM: echo:  M145 S1 H240 B110 F0
< 10:49:11 PM: echo:PID settings:
< 10:49:11 PM: echo:  M301 P10.65 I0.37 D76.40 C100.00 L20
< 10:49:11 PM: echo:Filament settings: Disabled
< 10:49:11 PM: echo:  M200 D3.00
< 10:49:11 PM: echo:  M200 D0
< 10:49:11 PM: echo:Z-Probe Offset (mm):
< 10:49:11 PM: echo:  M851 Z-1.20
< 10:49:13 PM: Error:Line Number is not Last Line Number+1, Last Line: 0
< 10:49:13 PM: Resend: 1
  10:49:13 PM: Resend: N1 M110 *2
  10:49:13 PM: Resend: N2 M115 *4
  10:49:13 PM: Resend: N4 M111 S6 *67
< 10:49:13 PM: FIRMWARE_NAME:Marlin 1.1.0-RC5 From Archive SOURCE_CODE_URL:https://github.com/MarlinFirmware/Marlin PROTOCOL_VERSION:1.0 MACHINE_TYPE:Cobblebot Basic EXTRUDER_COUNT:1 UUID:00000000-0000-0000-0000-000000000000
< 10:49:13 PM: echo:DEBUG: _­§N_g_¯N¯g©®­¿­©ÎWßOèùÂXÐ@®À¯
< 10:49:13 PM: ¾
< 10:49:13 PM: $ÉWßOÇXÐ@¥N­_­¥NO0
< 10:49:13 PM: ^0
< 10:49:13 PM: $k­­k¡¯¯¡J®î­ÿ­®êû
< 10:49:13 PM: ÅWßOùèËXÐ@å.­?­åmN­_­m$5ÃWßO9(ÍXÐ@®.­?­®	ôÁ¡®­¿­¡½'wý/|§§«§¼§×Æ°X«¼ÕÄ°X«¼Ñ
< 10:49:13 PM: ¼ -§«©«º«/©8­I­Z­¼Íök|Ñ
< 10:49:13 PM: ¼ -««­«¾«+­<­M­^­¼ÍöK\§iz3ú+<¥iz3ú¬Ã²
< 10:49:13 PM: ö+<¥iz3úK\§iz3ú¬Å´ö£øök| à0à©^÷ÿ
< 10:49:13 PM: À+í?àIìPäÇ¶
< 10:49:13 PM: ök|#Qð+í?àIìPäÇ¶ök|/¥8©I©Z©m~¡^÷À+©<©M©^©i¡z¡¡¡^÷À à0à©Ç¶^÷
< 10:49:13 PM: À+í?àIìPäÇ¶
< 10:49:13 PM: ök|#,­=­N­_­#Ç¶3ú-¡>¡O¡X¥_wnùK\/æ2áCèZã^÷ýÂÁÅ´;øÒ÷~mûï+!ô!à0à>-­¾½ààþ÷K\¬Ç¶e÷+<¥m¡~¡¡¥e÷o§x«««)¥:¥K¥\¥'l­}­­­'ö¥e÷k«|««« à0à@à_ãÃ²3ú£3ú¬`àpàèãöm£~££§¯¸«¼¥¥«¥¼¥¯¸`6p7'wý/
< 10:49:13 PM: echo:DEBUG: _­§N_g_¯N¯g©®­¿­©ÎWßOèùÂXÐ@®À¯
< 10:49:13 PM: ¾
< 10:49:13 PM: $ÉWßOÇXÐ@¥N­_­¥NO0
< 10:49:13 PM: ^0
< 10:49:13 PM: $k­­k¡¯¯¡J®î­ÿ­®êû
< 10:49:13 PM: ÅWßOùèËXÐ@å.­?­åmN­_­m$5ÃWßO9(ÍXÐ@®.­?­®	ôÁ¡®­¿­¡½'wý/|§§«§¼§×Æ°X«¼ÕÄ°X«¼Ñ
< 10:49:13 PM: ¼ -§«©«º«/©8­I­Z­¼Íök|Ñ
< 10:49:13 PM: ¼ -««­«¾«+­<­M­^­¼ÍöK\§iz3ú+<¥iz3ú¬Ã²
< 10:49:13 PM: ö+<¥iz3úK\§iz3ú¬Å´ö£øök| à0à©^÷ÿ
< 10:49:13 PM: À+í?àIìPäÇ¶
< 10:49:13 PM: ök|#Qð+í?àIìPäÇ¶ök|/¥8©I©Z©m~¡^÷À+©<©M©^©i¡z¡¡¡^÷À à0à©Ç¶^÷
< 10:49:13 PM: À+í?àIìPäÇ¶
< 10:49:13 PM: ök|#,­=­N­_­#Ç¶3ú-¡>¡O¡X¥_wnùK\/æ2áCèZã^÷ýÂÁÅ´;øÒ÷~mûï+!ô!à0à>-­¾½ààþ÷K\¬Ç¶e÷+<¥m¡~¡¡¥e÷o§x«««)¥:¥K¥\¥'l­}­­­'ö¥e÷k«|««« à0à@à_ãÃ²3ú£3ú¬`àpàèãöm£~££§¯¸«¼¥¥«¥¼¥¯¸`6p7'wý/

`

thinkyhead, 2016-04-04:
A patch #3343 was submitted to `RCBugFix` a couple of hours ago to address this, but maybe it didn't fix the issue. How long ago did you acquire the code?

maxwondercorn, 2016-04-04:
Downloaded about 90 minutes ago.  I believe download indicated it was updated an hour before I downloaded it, so the fix probably came after this rev.

jbrazio, 2016-04-04:
Submitted PR #3374 which ardesses this issue.

jbrazio, 2016-04-04:
Submitted PR #3374 which addresses this issue.

PR #3374 (2016-04-04): M111: Add static keyword to PROGMEM debug variables
Followup: #3343 
Fixes: #3346

Wrong memory address (text garbled) when serial outputting the active DEBUG flags.
commit d819a4af (2016-04-04): Fixes #3346, followup #3343: variables must defined with the static keyword to work with PROGMEM when defined inside functions
closed 2016-04-09 by thinkyhead