MarlinFirmware/Marlin #3347: Missing Aragonese translation

jbrazio, 2016-04-04:
We need your contribution to translate the following missing messages:

```cpp
#define	MSG_BED_Z
#define	MSG_A_TRAVEL
#define	MSG_HEATING_FAILED_LCD
#define	MSG_ERR_REDUNDANT_TEMP
#define	MSG_THERMAL_RUNAWAY
#define	MSG_ERR_MAXTEMP
#define	MSG_ERR_MINTEMP
#define	MSG_ERR_MAXTEMP_BED
#define	MSG_ERR_MINTEMP_BED
#define	MSG_HEATING
#define	MSG_HEATING_COMPLETE
#define	MSG_BED_HEATING
#define	MSG_BED_DONE
```

Have a look on the file [language_en.h](https://github.com/MarlinFirmware/Marlin/blob/RC/Marlin/language_en.h) for the original English translation.

jbrazio, 2016-05-16:
Closed by #3772.

closed 2016-05-16 by jbrazio

jbrazio, 2016-05-18:
I've submitted a PR to prune unmaintained language files, @danielmartinez as you're one of the contributors to this file would you care to submit an update to it ? Thanks.

danielmartinez, 2016-05-26:
I'm don't have enough time to update the translation right now... maybe I can translate it in a future, so, if you want to prune it for now, I can file a PR when I translate it.

jbrazio, 2016-05-26:
If you plan to maintain it then no need for pruning.