MarlinFirmware/Marlin #3350: Missing Chinese translation

jbrazio, 2016-04-04:
We need your contribution to translate the following missing messages:

```cpp
#define MSG_BED_Z
#define MSG_ERR_MINTEMP_BED
#define MSG_HEATING_COMPLETE
#define MSG_BED_HEATING
#define MSG_BED_DONE
#define MSG_BED_Z
```

Have a look on the file [language_en.h](https://github.com/MarlinFirmware/Marlin/blob/RC/Marlin/language_en.h) for the original English translation.

liangruiben, 2016-04-04:
# define  MSG_BED_Z     热床Z
# define  MSG_ERR_MINTEMP_BED   热床温度错误
# define  MSG_HEATING_COMPLETE  加热完毕
# define  MSG_BED_HEATING   加热热床
# define  MSG_BED_DONE    热床加热完毕

thinkyhead, 2016-04-05:
Also needed…
```cpp
#define MSG_LEVEL_BED_HOMING                "Homing XYZ"
#define MSG_LEVEL_BED_WAITING               "Click to Begin"
#define MSG_LEVEL_BED_DONE                  "Leveling Done!"
#define MSG_LEVEL_BED_CANCEL                "Cancel"
. . .
#define MSG_HOME_OFFSETS_APPLIED            "Offsets applied"
```

jbrazio, 2016-04-05:
@thinkyhead what is the encoding used by the LCD for CN chars ?

AnHardt, 2016-04-05:
@jbrazio 
There is no. There is no way to map some thousand symbols to 128.  What we have is a very special selection of symbols mapped one by one to the strings.

jbrazio, 2016-04-05:
Then this one will be a hard one to kill..
So you mean that hex values on the file is kinda of "custom mapping", nothing like big5 not utf8 ?

jbrazio, 2016-04-05:
@AnHardt Then this one will be a hard one to kill..
So you mean that hex values on the file is kinda of "custom mapping", nothing like big5 not utf8 ?

liangruiben, 2016-04-07:
#define MSG_LEVEL_BED_HOMING                "归位 XYZ"
#define MSG_LEVEL_BED_WAITING               "单击开始"  or 点击开始
#define MSG_LEVEL_BED_DONE                  "矫平完成!"  or 调平完成
#define MSG_LEVEL_BED_CANCEL                "取消"
. . .
#define MSG_HOME_OFFSETS_APPLIED            "偏移补尝"   or 应用偏移补尝

jbrazio, 2016-04-07:
Thank you @liangruiben.
But we're having another issue, we do not have "all" the Chinese charset available on the LCD.

liangruiben, 2016-04-10:
”全部“
I don't know how to help you.

CONSULitAS, 2016-04-10:
@liangruiben 
please look at https://github.com/MarlinFirmware/Marlin/blob/RC/Marlin/language_cn.h#L36

* The characters/symbols used are from `DISPLAY_CHARSET_ISO10646_CN`.
* Find https://github.com/MarlinFirmware/Marlin/search?utf8=✓&q=DISPLAY_CHARSET_ISO10646_CN
* leads you to https://github.com/MarlinFirmware/Marlin/blob/cf9a171aa92a4f43f0f61309c7145e46233b1c7c/Marlin/dogm_lcd_implementation.h#L72
* https://github.com/MarlinFirmware/Marlin/blob/RC/Marlin/dogm_font_data_ISO10646_CN.h has all chinese characters Marlin knows
* You find https://github.com/MarlinFirmware/Marlin/tree/RC/Marlin/fonts with https://github.com/MarlinFirmware/Marlin/blob/RC/Marlin/fonts/ISO10646_CN.fon which contains the pictures.
* There is a batch for it in https://github.com/MarlinFirmware/Marlin/blob/RC/Marlin/fonts/make_fonts.bat
* read the realm

So we have to
* use a character defined in dogm_font_data_ISO10646_CN.h
* or define a new one in dogm_font_data_ISO10646_CN.h and use it then
  * paint with Fony.exe
  * save and create new `.h` file
  * use new character in language_cn.h
  * BUT:

    >Currently we are limited to 256 symbols per font.

Since we all can not read any chinese it is impossible for us to do this work. Sorry.

liangruiben, 2016-04-10:
好的，明天我会尝试操作，现在我需要睡觉了，我这边的时间是早上4点
Ok, tomorrow I'll try to operation, I need to sleep now, the time is 4 o 'clock in the morning on my side

liangruiben, 2016-04-10:
paint with Fony.exe
fony.exe那里可以下载，我用的是WINDOWS 7 X64
Fony. Exe can download there, I am using WINDOWS 7 X64

CONSULitAS, 2016-04-10:
Please make a `>` before your citations. Tutorial: https://guides.github.com/features/mastering-markdown/

Fony should work on your system.

liangruiben, 2016-04-10:
好的，没有问题，我会尽快给你弄好，请原谅我需要时间学习
Ok, no problem, I'll give you as soon as possible repair, please forgive me I need time to study

jbrazio, 2016-05-16:
@liangruiben did you had any luck updating the language file ?

liangruiben, 2016-05-19:
Where can I download Foxy.exe

liangruiben, 2016-05-19:
OK, I found FONT.EXE downloaded

Blue-Marlin, 2016-05-19:
Do you mean Fony.exe (http://hukka.ncn.fi/?fony) what was used for the fonts?

boelle, 2016-11-05:
So we are still stuck with missing china?

Blue-Marlin, 2016-11-05:
No.
We are missing an update of the Chinese language files.

Blue-Marlin, 2016-11-05:
Even that is not the full truth.
We got a PR 'UTF-8 language translation supports #4704' that would make updating the Chinese language files much easier and this includes an update of the Chinese language files.
But:
The result on the screen is currently too poor to consider a merge.
While simplifying the update for Chinese, it complicates updates for the other languages.
It's outdated now.

thinkyhead, 2016-11-06:
> It's outdated now.

…and @yhfudev has been too busy to follow up with patches. We might just have to abandon that PR for the time being.

yhfudev, 2016-11-06:
@thinkyhead The current UTF-8 patch is a big change into the source code, and the most changes are related to the language files which cause the problem of 'outdated'.

I would suggest a two-step merging for the UTF-8 patch if you need more time to consider the merging and avoiding the 'outdated' problem. That would be easier for the two branches to be merged later.

First step is to merge the language files, which won't impact the current source code, to be shared with the two branches.

```C++
// define a macro in the language_en.h
#define _UxGT(a) a
```

and
```C++
// use _UxGT() to define the strings for all of utf-8 language files
#define WELCOME_MSG                         MACHINE_NAME _UxGT("已就绪.")  //" ready."
#define MSG_SD_INSERTED                     _UxGT("存储卡已插入")  //"Card inserted"
...
```
for those non-utf8 files, we don't need to change any thing, except asking the language maintainers to prepare two version files -- one for current solution and another one is for UTF-8.
So that we can update the UTF-8 branch without worrying about the conflicts until merging.

The second step is continue to update the UTF-8 branch according to feedback, and share the same language files with the main branch.

thinkyhead, 2016-11-07:
All good. And don't forget, there's still the issue of the messed-up vertical position of some glyphs.

thinkyhead, 2016-12-09:
Attempting to resurrect the UTF-8 support by @yhfudev with #5441.

boelle, 2017-04-26:
i think this one should have the 1.1.2 milestone on it.

it has gon a bit dead for a few months and i think we will get a very hard time finding any good chinamen that also speaks english as good as chinesse

closed 2017-11-01 by thinkyhead