MarlinFirmware/Marlin #3351: Missing Danish translation

jbrazio, 2016-04-04:
We need your contribution to translate the following missing messages:

```cpp
#define MSG_BED_Z
#define MSG_A_TRAVEL
#define MSG_HEATING_FAILED_LCD
#define MSG_ERR_REDUNDANT_TEMP
#define MSG_THERMAL_RUNAWAY
#define MSG_ERR_MAXTEMP
#define MSG_ERR_MINTEMP
#define MSG_ERR_MAXTEMP_BED
#define MSG_ERR_MINTEMP_BED
```

Have a look on the file [language_en.h](https://github.com/MarlinFirmware/Marlin/blob/RC/Marlin/language_en.h) for the original English translation.

DavidBjerreBjoerklund, 2016-04-06:
#define MSG_BED_Z			"Plade Z"
#define MSG_HEATING_FAILED_LCD		"Opvarmning mislykkedes"
#define MSG_ERR_REDUNDANT_TEMP		"Fejl: reserve temperatur"
#define MSG_THERMAL_RUNAWAY		"Temperatur løber løbsk"
#define MSG_ERR_MAXTEMP			"Fejl: Maks temperatur"
#define MSG_ERR_MINTEMP			"Fejl: Min temperatur"
#define MSG_ERR_MAXTEMP_BED		"Fejl: Maks temperatur plade"
#define MSG_ERR_MINTEMP_BED		"Fejl: Min temperatur plade"

jbrazio, 2016-04-07:
Thank you @DavidBjerreBjoerklund.
Could you please also translate the following:

```cpp
#define MSG_LEVEL_BED_HOMING                "Homing XYZ"
#define MSG_LEVEL_BED_WAITING               "Click to Begin"
#define MSG_LEVEL_BED_DONE                  "Leveling Done!"
#define MSG_LEVEL_BED_CANCEL                "Cancel"
#define MSG_HOME_OFFSETS_APPLIED            "Offsets applied"
```

DavidBjerreBjoerklund, 2016-04-07:
You still miss #define MSG_A_TRAVEL
I don't know how to tranlaste it

jbrazio, 2016-04-07:
Instead of translating "travel" you could try "distance" ?
Dunno it's a tricky one.

DavidBjerreBjoerklund, 2016-04-07:
When i A Travel used? What does it do?

jbrazio, 2016-04-07:
It allows to set travel acceleration.

DavidBjerreBjoerklund, 2016-04-07:
"A-bevægelse" maybe

DavidBjerreBjoerklund, 2016-04-07:
or "A-flyt"

DavidBjerreBjoerklund, 2016-04-07:
"A-rejse"

benj4560, 2016-04-21:
(When) are you done translating it? and where can i download it?

jbrazio, 2016-04-21:
@benj4560 it shall be merged soon.

commit fe4fc884 (2016-04-21): Closes #3351: missing Danish translation
closed 2016-04-21 by thinkyhead