MarlinFirmware/Marlin #3352: Missing German translation

jbrazio, 2016-04-04:
We need your contribution to translate the following missing messages:

```cpp
#define MSG_EXTRUDE
#define MSG_BED_Z
#define MSG_A_TRAVEL
#define MSG_HEATING_FAILED_LCD
#define MSG_ERR_REDUNDANT_TEMP
#define MSG_THERMAL_RUNAWAY
#define MSG_ERR_MAXTEMP
#define MSG_ERR_MINTEMP
#define MSG_ERR_MAXTEMP_BED
#define MSG_ERR_MINTEMP_BED
```

Have a look on the file [language_en.h](https://github.com/MarlinFirmware/Marlin/blob/RC/Marlin/language_en.h) for the original English translation.

jbrazio, 2016-04-07:
@AnHardt  @Blue-Marlin anyone ? ;-P

CONSULitAS, 2016-04-07:
@jbrazio 
I am german, too. But a little bit busy now. :smile: 

If no one is faster i will care for it later.

jbrazio, 2016-04-07:
I forgot you @CONSULitAS soz !
On Marlin the force is strong with the Germans.. :-P

closed 2016-04-14 by jbrazio