MarlinFirmware/Marlin #3354: Missing Basque-Euskera translation

jbrazio, 2016-04-04:
We need your contribution to translate the following missing messages:

```cpp
#define MSG_BED_Z
#define MSG_A_TRAVEL
#define MSG_HEATING_FAILED_LCD
#define MSG_ERR_REDUNDANT_TEMP
#define MSG_THERMAL_RUNAWAY
#define MSG_ERR_MAXTEMP
#define MSG_ERR_MINTEMP
#define MSG_ERR_MAXTEMP_BED
#define MSG_ERR_MINTEMP_BED
#define MSG_HEATING
#define MSG_HEATING_COMPLETE
#define MSG_BED_HEATING
#define MSG_BED_DONE
```

Have a look on the file [language_en.h](https://github.com/MarlinFirmware/Marlin/blob/RC/Marlin/language_en.h) for the original English translation.

jbrazio, 2016-05-16:
Closed by #3772.

closed 2016-05-16 by jbrazio

jbrazio, 2016-05-18:
I've submitted a PR to prune unmaintained language files, @koldoartola, @MatixYo as you're one of the contributors to this file would you care to submit an update to it ? Thanks.

MatixYo, 2016-05-19:
I submitted an update but I was not able to test if all of the strings show correctly on LCD

jbrazio, 2016-05-19:
That's not an issue, thank you.