MarlinFirmware/Marlin #3363: Missing Russian translation

jbrazio, 2016-04-04:
We need your contribution to translate the following missing messages:

```cpp
#define MSG_BED_Z
#define MSG_A_TRAVEL
#define MSG_HEATING_FAILED_LCD
#define MSG_ERR_REDUNDANT_TEMP
#define MSG_THERMAL_RUNAWAY
#define MSG_ERR_MAXTEMP
#define MSG_ERR_MINTEMP
#define MSG_ERR_MAXTEMP_BED
#define MSG_ERR_MINTEMP_BED
#define MSG_HEATING
#define MSG_HEATING_COMPLETE
#define MSG_BED_HEATING
#define MSG_BED_DONE
```

Have a look on the file [language_en.h](https://github.com/MarlinFirmware/Marlin/blob/RC/Marlin/language_en.h) for the original English translation.

gans-AD, 2016-04-08:
MSG_HEATING_FAILED_LCD - "Разогрев не удался"
MSG_ERR_REDUNDANT_TEMP - "Ошибка:Слишком горячо"
MSG_ERR_MAXTEMP - "Ошибка:Макс.темп."
MSG_ERR_MINTEMP - "Ошибка:Мин.темп."
MSG_ERR_MAXTEMP_BED - "Ошибка:Макс.темп.стола"
MSG_ERR_MINTEMP_BED - "Ошибка:Мин.темп.стола"
MSG_HEATING - "Нагреваю сопло..."
MSG_HEATING_COMPLETE - "Нагрев выполнен"
MSG_BED_HEATING - "Нагреваю стол"
MSG_BED_DONE - "Стол разогрет"

MSG_BED_Z ,  MSG_A_TRAVEL , MSG_THERMAL_RUNAWAY  - for translation you need the steps or menu items

gans-AD, 2016-04-11:
another option with a smaller number of characters
MSG_ERR_MAXTEMP - "Ошибка: Т макс."
MSG_ERR_MINTEMP - "Ошибка: Т мин."
MSG_ERR_MAXTEMP_BED - "Ошибка:Т макс.стол"
MSG_ERR_MINTEMP_BED - "Ошибка:Т мин.стол"

jbrazio, 2016-04-11:
Thank you @gans-AD.

closed 2016-04-12 by thinkyhead