MarlinFirmware/Marlin #3368: While is printing, the Extruder unload the material!?!??!!?

fbarcenas, 2016-04-04:
This is very rare...
After work some hours and all is very well, the Extruder 0 move around -100 aprox and the filament go out and continue working....
Of course continue working in the air, because the filament is totally unload...

Also before of this, the Axi X move to search the Sensor home X-, move to X- and after return to work...

Some idea?

jbrazio, 2016-04-04:
Could you please post the g-code (as an attachment) where you saw this happening ?

fbarcenas, 2016-04-04:
[Fan.gcode.zip](https://github.com/MarlinFirmware/Marlin/files/202317/Fan.gcode.zip)

This happen also with another files.

jbrazio, 2016-04-04:
Do you have more or less an indication of when it fails, i.e. they all fail around the same print time ?

fbarcenas, 2016-04-04:
Not.
For say something, maybe in 20%-30% but a thing is sure, ALWAYS before of this:
The axi X move in X- to sensor Home and after return to continue working, some lines after the extruder unload material and after continue working....

fbarcenas, 2016-04-04:
Sorry, i don't answer well.

> they all fail around the same print time ?
Not.
For say something, maybe in 20%-30% but a thing is sure,

> Do you have multiple extruders ?
yes, two. But in this jobs only use the first

jbrazio, 2016-04-04:
I understood.
Did you tried any other slicer ? I am not familiar with simplify 3D but looking at the g-code I see some dual extruder manipulation on the header which should not happen on a single extruder robot, nevertheless this is no excuse to unload the filament middle of print but I just would like to remove the slicer from the equation.

fbarcenas, 2016-04-04:
I will try another slicer, Cura and i say you comments
I am also having another problem
(Maybe is better do another new Issues ticket?)

In the Display VIKI2, when i go to menu to move the extruder, i move the extruder yes in positive and negative correctly, but i can't go exit the menu!, 
Normally i press the button and i go out the menu.... now i must wait the seconds to go automatically to first menu...

jbrazio, 2016-04-04:
> In the Display VIKI2, when i go to menu to move the extruder, i move the extruder yes in positive and negative correctly, but i can't go exit the menu!, 
Normally i press the button and i go out the menu.... now i must wait the seconds to go automatically to first menu...

Please open another ticket for his issue.

fbarcenas, 2016-04-05:
Today happen me again... this time in a file than use the 1 and 2 extruder, while use the 2 extruder, the machine unload the material...

Some idea???

thinkyhead, 2016-04-05:
Are you printing from a host like Octoprint or Repetier over USB? Do you see the same results if you print directly from SD card?

fbarcenas, 2016-04-06:
This problem happen me in 2 machines diferently, with boards diferently
But
Always from sd card.
I never use usb or octorpint

Maybe i can do some change with sdcard?

fbarcenas, 2016-04-06:
A question: in that format must have the sd card?
Exfat?
Fat32?
Fat?

jbrazio, 2016-04-07:
FAT32.

jbrazio, 2016-04-07:
We had in the past couple of weeks at least two issues that were solved by using a fresh new SDCard, you could also try that if after formatting the old one the same behavior appears.

fbarcenas, 2016-04-07:
Actually i use Transcend and Sandisk in FAT32...

jbrazio, 2016-04-07:
Did you already try re-formatting (full) them ?

fbarcenas, 2016-04-07:
Yes,
moreover in severals PCs

thinkyhead, 2016-04-07:
This problem has also happened on my machine due to a loose connection between the SD Card / LCD Controller unit and the RAMPS board. Is your SD Card directly on the electronics board, or is it in a separate unit? If it is separate, make sure that the plugs are nice and tight. It may even help to bend the pins a little and re-plug-in the unit.

fbarcenas, 2016-04-08:
The most rare thing is than happen me in two machines different, with wires different, SD cards deferents, boards differents...

Regard to the question, is in separate unit, display VIKI2.

By the way i will recheck everything.

thinkyhead, 2016-04-09:
I don't want to presume that it's your hardware, but so far I haven't seen this behavior with recent code updates. On the other hand, the longest prints I have tested are only 20-30 minutes.

One pretty sure way to find out is to start a print (just a simple one, maybe even remove your filament for the test) and while it's in progress, wiggle the wires of your LCD controller. If "weird things" happen when you do this, it's a clear indication of a connection issue.

fbarcenas, 2016-04-09:
i use the SD slot than have directly the board and also fail, is very rare, the "unload material" not happen around some days, but than the axi Y search the home yes...

thinkyhead, 2016-04-10:
This "axis Y search the home" you refer to is not a case of Marlin interrupting your print job to search for home position. It is an error in evaluating the GCode that I have seen before — when there are connection issues — that causes a strange movement. When I have seen this, there is a slow movement towards an edge, and then back over to the print, and then the print continues normally.

I think you may have some electronic noise issue going on with your board. Are you able to improve grounding and isolation?

fbarcenas, 2016-04-10:
than i see is than always happen in the first or second layer...
But is very rare than happen in the >RC3 version, in the last never happen!!
From Rc3 version some change than can affect to this?

Today i have a new and super Transcend miniSD card, i am testing...

What brands use yours?

Maybe some brands work wrong!?

Regard to the electronic noise and isolation i think than not, and i can have in one machine, but in two machines totally differently?

thinkyhead, 2016-04-11:
Well I suppose it's possible that you are seeing something that is both logically wrong and spurious, but these kinds of bugs will usually show up for a lot of users.

Ideally, the code of Marlin is totally "deterministic." That is, Marlin should do exactly the same thing every time you run the same GCode file. If it does not, then there has to be some random factor being introduced.

It would be helpful if you were able to do some diagnostics and narrow down and isolate the conditions required to produce the error.

In the meantime, which version of Arduino are you using to build Marlin?

fbarcenas, 2016-04-11:
1.6.8, but i go to install 1.6.7.
Because with 1.6.7 i remember than not have this problems, maybe not have importance...
With the new sdCard transcend also fail, this time stop the bed heat... is like read a commando M140 S0.
I check the file and not have M140 S0.

thinkyhead, 2016-04-12:
I look forward to seeing more test results.

fbarcenas, 2016-04-12:
Today more info.
WITH the last RCBUGFIX in this moment.
1.6.7->Work File perfectly, not do movements rates the printer BUT the LCD display CRASH and the machine STOP
1.6.8->The display not crash, but the machine search the home...unload material....

I am talking in works of many hours...happen in the firstly 2-3hours

jbrazio, 2016-04-13:
@fbarcenas sorry to bother you.. could you try with 1.6.0 ? To see if the behavior is a new third one or matches one of the existing.

thinkyhead, 2016-04-14:
@fbarcenas I'd like to try to eliminate or isolate the possibility that electronic noise or SD reading functions are involved. Can you try disconnecting the display and controller and printing over USB to see if any of these same issues occur? Also, when you print with the LCD controller attached, try wiggling the wires and see if any of these behaviors occur.

fbarcenas, 2016-04-19:
This days i am really confused, i continue testing and fail when the system want...
i found exactly when happen!!!!!
at first i think than is version of arduino, sd card brand, wires.... but not...!

I explain:
 i do a file than have a lot CIRCLES, when do this CIRCLES the marlin search home X or home Y.
is horrible, each 2-3 circles search a home position!!
i attach file.
And ALWAYS HAPPEN in the FIRST layer

Happen with SDCARD and USB

Unload material not happen
Tested with 18 april RCBUGFIX and arduino 1.6.0, 1.6.8,1.6.7

Ideas?
[11198-01.gcode.zip](https://github.com/MarlinFirmware/Marlin/files/226106/11198-01.gcode.zip)

a-sprenger, 2016-04-19:
I believe I have the same problem.
After nearly 2/3 of the job, the extruder stopps extruding. I don't know if it's unloading the filament, but all other axis are working until the job has been finished!

I changed the nozzles, the filament (different ABS materials), but nothing helps.

Attached pictures:
[image]
[image]

The good extruder carriage has been printed with my mendel (Filament and print settings are the same).
The bad one with my Prusa with RC5. The heart, too

Slic3r. honeycomb 3D, different heights(0.1/0.2mm), different infills(40/20%), Octoprint

fbarcenas, 2016-04-19:
a question.

What name you put to the file?.
Please say exactly the name, exactly letter to letter and mays

a-sprenger, 2016-04-19:
WP_20160419_21_35_36_Rich.jpg
WP_20160419_21_35_45_Rich.jpg

both attached in a zip-file
[pictures.zip](https://github.com/MarlinFirmware/Marlin/files/226799/pictures.zip)

fbarcenas, 2016-04-19:
The name of the file gcode i want say, the file name than you send to Octoprint.

Thanks bye

thinkyhead, 2016-04-20:
I will test my own printer with your GCode later on and see what happens. If it exhibits similar issues, maybe I can isolate a cause.

a-sprenger, 2016-04-20:
Yesterday evening, my daughter told me, that she and her girlfriends like it to play with the different coloured "plastic". I'm not sure if that is the reason!
Yesterday evening I got the same problem with the old firmware!

I have to clean the nozzles and take new filament!

fbarcenas, 2016-04-20:
I do a new test....
SAME FILE, i change file name from "11198-01.gcode" to "Test.gcode" and... FAIL
i change to "fail.gcode" (all minus) and NOT FAIL

I am totally confused, problem of circles or with the name?!?!?!?!

thinkyhead, 2016-04-21:
@fbarcenas We had a problem in the past with numbers in filenames, and it's fixed now, so it's weird to see the "opposite problem" where a filename with numbers works, but a filename without them doesn't.

So it's also strange that `Test.gcode` fails while `fail.gcode` doesn't fail.

Are you certain that your SD Card Reader is actually okay? Is it formatted with FAT32 format? Are you able to test using a different SD Card and see if the results are the same?

It would help if, before you do any printing or testing, you could connect a console and gather all the output from Marlin so we can see what commands it's issuing. Maybe the `M23` command is mangled, or something else weird is happening.

You can use the `M111 S1` command to enable logging of every command that goes into the buffer.

thinkyhead, 2016-04-21:
@a-sprenger

Since your model is extremely large I was unable to test it on my system. The GCode is over a million lines long, and the object is arranged from around X=300 to around X=900 on the build plate. I tried just running in "Dry Run" mode, which skips all E moves and leaves the heaters off. Of course, this will conceal any problem that might exist with the E axis, but the math with E going forward is probably okay because E is reset to zero for every new set of moves.

Since this file uses `G92` a lot, I decided to do a little cleanup on the `gcode_G92` function. I did have to make a small change to prevent the E axis being synchronized with the planner more than once in some cases. I don't think this minor issue in `G92` is contributing to the problem you are seeing, so don't expect anything new after I merge that change.

We're also looking at a potential acceleration bug right now, so something we figure out there might help... #3520

fbarcenas, 2016-04-21:
I don't explain very well:
Filenames:
11198-01.gcode     FAIL
Test.gcode             FAIL
test.gcode              NOT FAIL

Tested with USB wire and different sdcards
I go to read around M111 because i don't use this never

thinkyhead, 2016-04-21:
> Tested with USB wire and different sdcards

So when you say "fail" in what way do they fail? Do they never start to print? Or does the printer quit extruding in the middle of the print?

fbarcenas, 2016-04-21:
When i say fail i want say than the machine do rare things.
Time before the rare things are: unloading material, but now the fails is always the same, the machine search home X or Y

The fail of unload material you fix in some rcbugfix version i think (maybe without you know than are fixing this...)

fbarcenas, 2016-04-21:
if the filename have a mayus or numbers, the work fail (home search X or/and Y) sure!

thinkyhead, 2016-04-21:
Well then we should look at what's going into the command buffer. Please use the `M111 S1` command to enable logging of every command that goes into the buffer, connect Pronterface, and try printing from SD as usual, then watch the output for strange things. Paste the output here if you're not sure.

fbarcenas, 2016-05-01:
Sorry for the delay, still i dont test the command M111, i will test tomorrow.
But witj the last version the marlin RESTART,
Like if you press reset button...

thinkyhead, 2016-05-02:
Looking forward to the results of your testing.

fbarcenas, 2016-05-03:
Sorry i don't understand Pronterface... 
What software i can use for log the and save the log in a file?

thinkyhead, 2016-05-03:
@fbarcenas After Pronterface (or other host) is connected, all messages from Marlin appear in its "console" region. You can use copy-and-paste as-usual to copy that text and paste it into another file, or into the browser edit field. You don't need me to explain copy-and-paste, I hope!

fbarcenas, 2016-05-03:
i am user of simplify3d and when i use pronterface i don't see like connect to the printer, i can't change the port COM... i know like do, but not work .(
For this i ask if is possible use another softs?
You know another softs?

thinkyhead, 2016-05-03:
@fbarcenas If you can't get a serial connection to your printer, there must be something wrong with your printer. You should be able to connect any host software without a problem. You can try: Cura, Repetier Host, Pronterface, and Octoprint (runs on Raspberry Pi). They should all be able to connect.

fbarcenas, 2016-05-03:
Yes yes i know like use this,
i use octoprint, cura and repetier and everything is ok, but Pronterface not work well, i am thinking than my pc windows 10 is the problem.

By the way i am printing now things and everything is ok...
But i do a change, i disable SDsupport and disable my display viki2

a-sprenger, 2016-05-03:
Hi,
well, I have no problems here. The stepper motor was damaged and the filament full of candy (from my kids). When the stepper motor became a little bit warmer it stopped totally!

thinkyhead, 2016-05-03:
@fbarcenas Ok, the only reason I brought up "Pronterface" at all was just as a suggestion, a way to get the output from Marlin. If your other host software is working, then you can use that for the same purpose. It doesn't matter which one.

Roxy-3DPrintBoard, 2016-05-03:
Also, make sure you don't have anything else connected to the printer.   This is the opposite problem you are having but I was banging my head on the table because I couldn't get Arduino to update the firmware on my printer yesterday.   The reason was because I had PronterFace connected to the printer.

fbarcenas, 2016-05-04:
Continue testing...
Disable SDCARD and VIKI2 lines...everything is ok...

fbarcenas, 2016-05-04:
Hello!
The problem happen and i have the log!!
The fail happen when:
RECEIVED: echo:busy: processing

Importance to: WARNING: Firmware unresponsive.  Attempting to force continue...
after of this axi X go to search HOME but crash and crash.... like the sensor not work.... after i turn off the machine...
[Fail1.txt](https://github.com/MarlinFirmware/Marlin/files/248942/Fail1.txt)

thinkyhead, 2016-05-05:
@fbarcenas You can disable the busy feedback with `M113 S0` or by enabling `DISABLE_HOST_KEEPALIVE` if it's bothersome or causing additional problems.

fbarcenas, 2016-05-05:
So Now at the first of each file i write:
M111 S1            ;activation log
M113 S0           ;desactivation busy:processing 

i will say something....

fbarcenas, 2016-05-06:
Hi again, the problem happen but happen some hours before and the log don't save this...
Only save some lines and i don't find.
But, why if i put: M113 S0 at start work, the machine continue say: busy: processing

I upload the new rcbugfix with DISABLE_HOST_KEEPALIVE

A idea...?
is possible than the problem will be the 250000 baud rate?? i never touch this
Is a PC with windows 10

thinkyhead, 2016-05-06:
M113 is pretty new. You might find it actually works with the latest bug fix code. But I will peek at it to make sure it's doing the right thing.

[50 more replies]

closed 2017-10-31 by thinkyhead