MarlinFirmware/Marlin #3373: Problems of MBL menu on RepRapDiscount Full Graphic Smart Controller

esenapaj, 2016-04-04:
problem 1:
The upper part of menu strings to be missing(2 lines?).
If I rotate a encoder then those missing lines are displayd.
But if I stop rotation, lines disappear again.

problem 2(or it's in the specifications?):
60 second after displaying "Home XYZ",
it switchs to "Click to Begin" automatically whether I rotate a encoder or not.

I've took a video clip of those.
https://www.youtube.com/watch?v=lsxpxf0iaVU

epatel, 2016-04-04:
What version/commit are you using?

esenapaj, 2016-04-04:
I'm used RC5(RCBugFix) as of about GMT April 4, 22h in this video clip, 
But newest RCBugFix as of GMT April 4, 12h produced same result.

esenapaj, 2016-04-04:
And, this is my configs.
[Configuration.txt](https://github.com/MarlinFirmware/Marlin/files/202618/Configuration.txt)
[Configuration_adv.txt](https://github.com/MarlinFirmware/Marlin/files/202619/Configuration_adv.txt)

These configs are made for only testing.
Because I've only delta printer, therefore I can't use MBL in the normal.

jbrazio, 2016-04-04:
@epatel From the video seems that the LCD is redrawing every cycle, could it be ?

jbrazio, 2016-04-04:
@esenapaj offtopic: your encoder "feels" to be reversed, you can now play with `#define REVERSE_MENU_DIRECTION` to get it right.

esenapaj, 2016-04-04:
I'm grateful for your instruction.

esenapaj, 2016-04-04:
Additional infomation 2:
If I test with RepRapDiscount Smart Controller(ASCII LCD),
"problem 1" doesn't occurs, but "problem 2" occurs.

thinkyhead, 2016-04-05:
> it displays strange strings

That is now patched thanks to @jbrazio.

thinkyhead, 2016-04-05:
So, the only problem then is that after a 60 second delay there's an automatic exit from the bed leveling? That is an interesting behavior, and it should be disabled. I will check to see that the appropriate flag is being set, in case it's missing there.

thinkyhead, 2016-04-05:
Ok, so the only reason that the `Homing XYZ` screen should appear is that `Bed Leveling` was clicked (instead of `Cancel`), which starts the homing procedure. The only reason that it ever proceeds to the `Click to Begin` screen is because
```cpp
(axis_known_position[X_AXIS] && axis_known_position[Y_AXIS] && axis_known_position[Z_AXIS])
```

thinkyhead, 2016-04-05:
> the LCD is redrawing every cycle, could it be?

Normally it updates at 10Hz. But if the display memory is being changed at any time this can show up immediately, before a proper display update.

esenapaj, 2016-04-05:
An interim report:
I tested with [latest RCBugFix](https://github.com/MarlinFirmware/Marlin/commit/1220dfd5d6af93d316cd53a3865939d9460fc0d3), Strange strings on Repetier-Host is gone.
Problem 1 and 2 still occurs.

WheresWaldo, 2016-04-05:
I just tested this with the latest RCBugFix (Apr-5-2016) the two bugs pictured in the video still exist. Also if you actually follow the on screen direction to `Click to Begin` it appears to jump right to the second probe location skipping the first location completely.

thinkyhead, 2016-04-05:
The jump after "Click to Begin" is probably caused by the next handler receiving an extra click. Try this and see if it fixes the problem. Change the line…
```cpp
lcd_goto_menu(_lcd_level_bed_procedure);
```
…to this…
```cpp
lcd_goto_menu(_lcd_level_bed_procedure, true);
```
In the long run the menu manager needs to capture all clicks and apply a proper debounce, then pass the click state on to the handlers. Currently, handlers need to do their own debounce management.

esenapaj, 2016-04-06:
An interim report 2:
I tested with [latest RCBugFix](https://github.com/MarlinFirmware/Marlin/commit/d5b19a49f626e387447b09357c060e17f8d1772b), problem 1 and 2 still occurs.
I can't test the @WheresWaldo's issue,
because I'm testing about these problems in this thread with a delta machine(+ configs for cartesian) .

esenapaj, 2016-04-08:
And I found a another problem(or specifications?) with [latest RCBugFix](https://github.com/MarlinFirmware/Marlin/commit/4e17187c9650eb4683d6347bfc8cb139346985db),
"Kill" button doesn't works in scene of "Home XYZ".
But if in a next scene that it's "Click to Begin", it works.

WheresWaldo, 2016-04-08:
I went back and tested with today's RCBugFix 2016-APR-08 and the MESH menu moving directly to point 2 appears fixed. But all the notification strings on the LCD are missing the top row of pixels. When you move the encoder the row appears, you stop and the row disappears. So the issues 1 & 2 still remain.

thinkyhead, 2016-04-09:
> "Kill" button doesn't works in scene of "Home XYZ"

@esenapaj I should probably add an `idle()` call to that screen while it's waiting for homing to complete. I presume that the kill button works okay when you issue a `G29` command from the console, yes?

> notification strings on the LCD are missing the top row of pixels

@WheresWaldo I'm surprised to hear that a character-based LCD display is doing weird things in the update. I'd be less surprised if it was a graphical display. Anyway, I will look and see if there's something additional that can be done to fix that. Maybe an additional screen clear or redraw is all that's needed…

thinkyhead, 2016-04-09:
Ok, so I've added a little more control over redraws, in case we need to do an extra redraw (without clear) on character changes. The PR #3435 may not have any effect on this issue (yet), but I think it should be tested ahead of further changes. You can just download the whole branch at https://github.com/thinkyhead/Marlin/tree/rc_lcd_leveling_home_etc and test it. As long as the behavior is no worse, I will be happy. Then we can see if tweaking the handlers to do an extra redraw truly helps.

esenapaj, 2016-04-09:
I tested PR #3437 in haste, I think the PR needs update.
In all of the menu, whole screen disappear If I didn't rotate the encoder.
[video clip](https://www.youtube.com/watch?v=GMxdqkHbV_o)

thinkyhead, 2016-04-09:
@esenapaj Thanks for giving it a try. Perhaps it's worth it to try changing some instances of `LCD_DRAW_UPDATE_CALL_REDRAW_NOW` to `LCD_DRAW_UPDATE_CALL_REDRAW_NEXT` just to see how it affects redraws – especially on the graphical display. I can test my own `G3D_PANEL` but I don't have a `REPRAP_DISCOUNT_SMART_CONTROLLER` to test. (Apparently it behaves differently from mine.)

WheresWaldo, 2016-04-09:
Sorry, I should have been more clear. I am also using the Full Graphics LCD Smart Controller. I will ask the few brave fellow Robo users that are testing RC5 or newer if any have the 2004 or XXL to see if they see the same symptoms.

esenapaj, 2016-04-09:
@thinkyhead
>to try changing some instances of LCD_DRAW_UPDATE_CALL_REDRAW_NOW to LCD_DRAW_UPDATE_CALL_REDRAW_NEXT

Well, it need to be tested in equal or over the 2^5 cases?
I hasten to set the numeric constants as with the undermentioned, 
it's similar non-patched, but Kill button doesn't work around of all menues.

```
line 560
      lcdDrawUpdate = LCD_DRAW_UPDATE_CALL_REDRAW_NEXT;
line 1135
    lcdDrawUpdate = LCD_DRAW_UPDATE_CALL_REDRAW_NEXT;
line 1164
     lcdDrawUpdate = LCD_DRAW_UPDATE_CALL_REDRAW_NEXT;
line 1654
      lcdDrawUpdate = LCD_DRAW_UPDATE_CALL_REDRAW_NEXT;
line 2189
          lcdDrawUpdate = LCD_DRAW_UPDATE_CALL_REDRAW_NEXT;
```
          
If I set the constants as with the undermentioned, I'm feeling that it's similar to non-patched.
But, kill button doesn't works until the Homing is finished, when I issue a G28 from the command-line.
```
line 560
      lcdDrawUpdate = LCD_DRAW_UPDATE_CALL_REDRAW_NOW;
line 1135
    lcdDrawUpdate = LCD_DRAW_UPDATE_CALL_REDRAW_NOW;
line 1164
     lcdDrawUpdate = LCD_DRAW_UPDATE_CALL_REDRAW_NOW;
line 1654
      lcdDrawUpdate = LCD_DRAW_UPDATE_CALL_REDRAW_NOW;
line 2189
          lcdDrawUpdate = LCD_DRAW_UPDATE_CALL_REDRAW_NEXT;
```

>I presume that the kill button works okay when you issue a G28 command from the console, yes?

Kill button works fine with non-patched,
when I issue a G28 command from the command-line in Repetier-Host.
But with PR #3435 and set the constants as with similar to the above
```
line 560
      lcdDrawUpdate = LCD_DRAW_UPDATE_CALL_REDRAW_NOW;
line 1135
    lcdDrawUpdate = LCD_DRAW_UPDATE_CALL_REDRAW_NOW;
line 1164
     lcdDrawUpdate = LCD_DRAW_UPDATE_CALL_REDRAW_NOW;
line 1654
      lcdDrawUpdate = LCD_DRAW_UPDATE_CALL_REDRAW_NOW;
line 2189
          lcdDrawUpdate = LCD_DRAW_UPDATE_CALL_REDRAW_NEXT;
```
, kill button doesn't works until the Homing is finished, when I issue a G28 from the command-line.

esenapaj, 2016-04-09:
##### About problem 2:
##### I become aware that this problem occurs in only case of AUX power(it's 12V usually) is being cut off, after such a long time...
So, This isn't a problem, or no need to fix?

thinkyhead, 2016-04-10:
If "Problem 2" is not software then I guess we can't fix it here. But why is the AUX power being cut off? And, to what "long time" do you refer? The lengthy duration of doing a `G28` command? The "Homing XYZ" screen should only be visible until `G28` is completed, and that should take < 10 seconds.

thinkyhead, 2016-04-10:
> it need to be tested in equal or over the 2^5 cases

"5^2" cases, you mean?

Anyway, thank you for reporting your results. I got to thinking about the changes I've made, and realize there is some alteration of behavior. I'm inclined to go back to the start and try this fix again, only more gradually.

esenapaj, 2016-04-10:
>to what "long time" do you refer?
 
Sorry, it's my very bad english...
I've rewritten the sentence.

Blue-Marlin, 2016-04-10:
Problem 1 can be caused if you tried to call one of the relevant menus outside `lcd_update()`.
Then probably you forgot the 'display loop' for the DOGM-displays.
```cpp
        u8g.firstPage();
        do {
...
          (*currentMenu)();
        } while (u8g.nextPage());
```

thinkyhead, 2016-04-10:
The display loop for DOGM hasn't been removed, so that isn't the proximal source of the issue. But, the changes I made to allow more "fine-grained" control over menu handlers is missing some key points that the old code managed to capture.

- The old code evaluates `lcdDrawUpdate` _after_ the "menu handler" calls
- The old code clears the screen _after_ the "menu handler" calls and draws _on the next loop_
- The old code always decrements `lcdDrawUpdate` after evaluating it

- The new code evaluates `lcdDrawUpdate` _before_ the "menu handler" calls
- The new code may clear the screen _before_ the "menu handler" calls, then draw _in the same loop_
- The new code uses `lcdDrawUpdate` as a state machine instead of a "draw counter"

My intention was to get the behavior as close to the original as possible, but add some more options for menu handlers (so they can get a callback to test conditions without being required to redraw, for example).

Since my "clever" new implementation manifests the differences itemized above, I need to keep tweaking it until it does exactly what the old implementation did. For example, if there's a screen clear, then return and wait till the next loop to redraw. Things like that.

"Starting over" —if needed— isn't a big deal, as this isn't a huge change to the "menu manager," and I can do more logging to make sure it's all happening in the same order as before.

AnHardt, 2016-04-10:
What happened to my donation from 14.mar.2016?
Not received?
Ignored?
Not enough?
Used for coffee?
Not ordered yet?
Not received yet?
Not in use yet?

Blue-Marlin, 2016-04-10:
Just a rough guess because the symptom (only refreshing the last stripe) would fit.

esenapaj, 2016-04-10:
Oh... does development team of Marlin haven't RepRapDiscount Full Graphic Smart Controller?
If it's true, I send the actual thing to the base of developers, or I offer purchase fund. [(here?)](https://www.gofundme.com/zmh9h9vc)
I'm regretting that release schedule of the Marlin is delayed because of these problems in this thread,
because this thread was opened by me.

emartinez167, 2016-04-10:
@thinkyhead, have you received your display?

Regards,

Ernesto

thinkyhead, 2016-04-10:
> What happened to my donation from 14.mar.2016?

I received your donation which is much appreciated, thank you! Although I praised you by name on social media at that time, my internet access has been very limited, so I haven't had time to do my next update post on Patreon and GoFundMe, where I will be thanking you once again, perhaps even writing a poetic verse about it!

thinkyhead, 2016-04-10:
@emartinez167 I have been watching my mailbox but it has not arrived. I was looking for the tracking number, which you sent me in an image, but I couldn't figure out where the image is! Can you re-send the tracking number?

thinkyhead, 2016-04-10:
Please help clear this up for me. Are you, @AnHardt and @Blue-Marlin the same individual?

thinkyhead, 2016-04-10:
> development team of Marlin haven't RepRapDiscount Full Graphic Smart Controller?

I'm building and testing with my one and only RepRap, a custom-built Prusa i3.

- Mega2560 + RAMPS 1.4
- G3D_PANEL
- Mk2 PCB Heated bed
- No added sensors

emartinez167, 2016-04-10:
The image is in the Google hangout

esenapaj, 2016-04-10:
Well, what shall I do...
I've thought that Marlin Project is operated by a corporation, but it seems to be wrong.
I think that Marlin is most powerful about developing in RepRap world,
by contrast, backbone of development system is extremely vulnerable.

Anyway, It looks like that RepRapDiscount Full Graphic Smart Controller was sent out by martinez167[(this?)](https://github.com/MarlinFirmware/Marlin/issues/3129#issuecomment-195334307),
I've offered $30 by [GoFundMe](https://www.gofundme.com/zmh9h9vc) for the present, for @thinkyhead's food.

And this is meddling entirely,
I think that development team of Marlin open an official account for donation(it needs to be incorporated?),
and maybe the link should be put the top page of the site or README.md, like the [Blender nation](https://www.blender.org/).
OSS isn't synonymous "don't make money".

Chaplin said "All it takes is courage, imagination and a little dough",
but to put it the other way around, minimum dough is needed for everything.

emartinez167, 2016-04-10:
@thinkyhead is still waiting for the graphic controller to arrive (snail
mail so expect delays).

esenapaj, 2016-04-10:
I see...
Then, I think that it's better to halt fixing about the problems in this thread until @thinkyhead recieve the GLCD. 
He is walking blindfolded now, and it's waste his limited online time in vain.
It's not what I want.

thinkyhead, 2016-04-11:
Whew! My pessimism has been at an all-time high, so thank you for brightening my day with reassurances. A lot of equipment has been promised over the year, and this will be the first to arrive! Very excited to get into the graphical display side and start breaking things there too....

esenapaj, 2016-04-11:
That's good news, I’m so happy for you.

esenapaj, 2016-04-23:
final report:
I've tested newest [RCBugFix](https://github.com/MarlinFirmware/Marlin/commit/d28db67ca9952ca8f315c1020c9708893cdc4074)(PR #3596 has been merged), and I confirmed "Problem 1" has been solved.
Now, all of the problems in this thread  that it was reported by me has been solved.

I think that this thread can close, but we might as well wait a moment for a report by @WheresWaldo.

WheresWaldo, 2016-04-23:
I will take time to test this today.

closed 2016-04-25 by thinkyhead