MarlinFirmware/Marlin #3375: Imposible exit of menu Move extruder

fbarcenas, 2016-04-04:
In the Display VIKI2, when i go to menu to move the extruder, i move the extruder yes in positive and negative correctly, but i can't go exit the menu!, 
Normally i press the button and i go out the menu.... now i must wait the seconds to go automatically to first menu...
Only happen in the actual RCBUGFIX

esenapaj, 2016-04-04:
I also confirmed that issue with the RepRapDiscount Full Graphic Smart Controller.
It's not only extruder but also x, y, z.

fbarcenas, 2016-04-04:
Yes, VIKI2 is based in RepRapDiscount Full Graphic Smart Controller.
I test X Y and Z and also happen, in the rest of menu is ok

esenapaj, 2016-04-04:
In addition, this issue become more worse with delta machine.
If I select some of x, y, z and rotate a encoder,
all of the axis(x and y and z) raise about 85mm and completely freeze in any case.
Kill switch on the LCD module can't even work, and reset button on RAMPS besides.
To escape from freezing, I've no choise except take off a power plug.
[video clip](https://www.youtube.com/watch?v=N9pjVsjHXNQ)

If I select e(extruder) and rotate a encoder,
It doesn't freeze but somehow xyz axis raise. 
[video clip](https://www.youtube.com/watch?v=XTSfd-sZaVg)

I used [latest RCBugFix branch](https://github.com/MarlinFirmware/Marlin/tree/0433df383556c37085929b442615083dd0bcf094) for these test.

jbrazio, 2016-04-04:
@thinkyhead did some rework on the LCD lately let's wait for his feedback as there seems to be a couple of LCD/encoder related issues lately.

fbarcenas, 2016-04-04:
thinkyhead _show us the light of the way_

esenapaj, 2016-04-04:
Additional infomation:
When I tested with RepRapDiscount Smart Controller(ASCII LCD), strange phenomenon has occurred.
[video clip](https://www.youtube.com/watch?v=pASv8Sb1jTw)

...something overflow?

jbrazio, 2016-04-04:
That's the hidden game "snake" :-P
Just kidding, indeed something looks wrong. @thinkyhead is now online he can comment more accurately than I can.

thinkyhead, 2016-04-04:
Recently I adjusted the menus to use a proper "history" to go back to previous items. It's possible this got muddled in the edit items, which (I'm guessing) is probably doubling-up on the "previous position" saving, (because edit items used to do that separately). If this is the case it should be a simple patch, removing the extra `lcd_save_previous_menu`. I will have a look and follow up shortly with some kind of patch.

thinkyhead, 2016-04-04:
Hmm, that LCD menu crash when using `Move X` => `Move 0.1mm` is pretty weird. Looks like a buffer overflow, but the `Move` items don't use any buffer. After that I would think stack overflow – but I think that would fill your LCD display from right-to-left instead of left-to-right. I will have to test later and see if I can reproduce this.

Was the `Move X` command in the video done before doing any `G28` homing?

esenapaj, 2016-04-05:
>Was the Move X command in the video done before doing any G28 homing?

Yes it is. It didn't occur to me until you referred so. 

case of delta machine:
If I use ''Move X" => "Move 0.1mm" before G28, critical freezing occurs.
If I use it after G28, freezing doesn't occurs.

case of cartesian machine(config for cartesian + delta machine. Because I've only delta machine.):
If I use ''Move X" => "Move 0.1mm" then freezing doesn't occurs whether before G28 or after.

Well, Initial value of variables for delta has problems?

fbarcenas, 2016-04-05:
In my case, not freeze but the problem continue exactly!

G-Forge, 2016-04-05:
RAMPS 1.4, Discount full Graphic Smart Controller, RC5bugfix @ now
Prepare > Move axis
move any axis OK it doesn't freeze but can't exit this menu option however it does timeourt to front screen after 15 seconds.

esenapaj, 2016-04-05:
About "Imposible exit" problem, I tested a bit more.

case of RepRapDiscount Full Graphic Smart Controller:
It can't exit at first sight, but can exit infrequently.
[video clip
](https://www.youtube.com/watch?v=ggAB2_NxffY)

case of RepRapDiscount Smart Controller:
It can exit, but it needs click once,twice or thrice... needs multiple.
Response isn't as good.
[video clip](https://www.youtube.com/watch?v=z-g72lccHlM)

thinkyhead, 2016-04-05:
I think the "can't exit" behavior might actually be "exits but re-enters again immediately." If I add a click feedback sound to the edit item it should fix it, if that is what's happening. After #3397 is merged give `RCBugFix` another try.

thinkyhead, 2016-04-05:
Perhaps it would make sense to hide "Move Axis" items for any axis that hasn't been homed. I don't have a delta to test with, so I can only try to figure out the issue by studying the code. At the beginning, the delta should "think" it is at 0,0,0 – in the middle and touching the bed. But the steppers could be anywhere, and it's probably a bad thing to arbitrarily move a delta when it doesn't know where its steppers are positioned.

esenapaj, 2016-04-05:
>Perhaps it would make sense to hide "Move Axis" items for any axis that hasn't been homed.

I think that it's good idea.

thinkyhead, 2016-04-05:
Ok, so #3397 adds click feedback to all items that don't automatically get it, and should resolve this issue.

I'll make another patch shortly to hide the move options until homing is done (for Delta, but leave them in place for Cartesian where it's less potentially damaging). Also, pretty soon the menus will get a cleanup, so clicks will be handled in one place instead of every handler needing to manage its own "debounce."

In the process, hopefully I can figure out why in some cases the display goes weird and fills up with squares.

esenapaj, 2016-04-06:
I've tested with [newer(not newest) RCBugFix branch](https://github.com/MarlinFirmware/Marlin/commit/90f6f02aca493c0174998241b64377d47806876c), and confirmed "Imposible to exit" problem is gone
both RepRapDiscount Full Graphic Smart Controller and RepRapDiscount Smart Controller.
I'm grateful for the fix.

there are video clips of verifications with delta.
[RepRapDiscount Smart Controller, without homing, endstop to carriage > 84mm](https://www.youtube.com/watch?v=Muo3k8xVhJE)
[RepRapDiscount Smart Controller, without homing, endstop to carriage < 84mm](https://www.youtube.com/watch?v=kBUD9Q_GmUQ)
[RepRapDiscount Smart Controller, with homing](https://www.youtube.com/watch?v=ZaHchpkNMAw)
[RepRapDiscount Full Graphic Smart Controller, without homing, endstop to carriage > 84mm](https://www.youtube.com/watch?v=N4qrSoPt_GI)
[RepRapDiscount Full Graphic Smart Controller, without homing, endstop to carriage < 84mm](https://www.youtube.com/watch?v=zQVTWf9fCnQ)
[RepRapDiscount Full Graphic Smart Controller, with homing](https://www.youtube.com/watch?v=MF8BTmNT5aE)

If endstop to carriage < about 84mm(it's about 134400steps in my printer), machine doesn't freeze and ASCII display isn't filled up with squares albeit without homing.
In this case, effector(of hotend) touch the endstop before riseing come up to 84mm.
I think that it means something overflow or break at about 134400steps?

from my Configuration.h
```
#define XYZ_FULL_STEPS_PER_ROTATION 400 //original is 200
#define XYZ_MICROSTEPS 128
#define XYZ_BELT_PITCH 2
#define XYZ_PULLEY_TEETH 16
#define XYZ_STEPS (double(XYZ_FULL_STEPS_PER_ROTATION) * double(XYZ_MICROSTEPS) / double(XYZ_BELT_PITCH) / double(XYZ_PULLEY_TEETH)) //=1600steps/mm
#define DEFAULT_AXIS_STEPS_PER_UNIT   {XYZ_STEPS, XYZ_STEPS, XYZ_STEPS, 476.55 * (XYZ_MICROSTEPS / 16)}
```

emartinez167, 2016-04-06:
That sounds like a data type for the variable holding that value needs to be larger...

Regards,

Ernesto

esenapaj, 2016-04-06:
And I've tested with [newest RCBugFix](https://github.com/MarlinFirmware/Marlin/commit/d5b19a49f626e387447b09357c060e17f8d1772b) lightly just now,  the freezing still alive.
But it can be concealed from user, by hiding "Move" menu until homing...but not good for fix the potential bug?

Roxy-3DPrintBoard, 2016-04-06:
> But it can be concealed from user, by hide the "Move" menu until homing...but it's not good for fix the potential problem?

My thinking is we should go fix the root cause of the problem and not 'Hide' the problem.   Give us a little time esenapaj.  This will get fixed...

esenapaj, 2016-04-08:
Well...the freezing issue was concealed from users now, by PR #3426.
But, it needs more to conseal.
Because if I use the "Extruder"(it means "Move E") without homing with delta machine,
XYZ axes rise leisurely for some reason.
[video clip](https://www.youtube.com/watch?v=kmGGmcHllQs)

It was tested with [newest RCBugFix](https://github.com/MarlinFirmware/Marlin/commit/4e17187c9650eb4683d6347bfc8cb139346985db).

thinkyhead, 2016-04-09:
Hiding the move functions before homing on Delta/SCARA is still an important change that I think we should keep.

Anyway, @esenapaj might be onto something concerning the value being too large for the variable.

It should be noted that `double(XYZ_FULL_STEPS_PER_ROTATION)` is meaningless on AVR. The `double` type is just an alias for `float`.

thinkyhead, 2016-04-09:
>  I use … "Move E" … without homing with delta machine … axes rises leisurely for some reason

How interesting! So with only E changing, the Z is also changing in some way… This is turning out to be quite bizarre indeed.

thinkyhead, 2016-04-09:
Ok, so it might not do a damned thing, but #3437 attempts to address possible value conversion bugs. See if it makes any difference at all (you might try with "unhomed delta allowed" also).

esenapaj, 2016-04-09:
I tested the PR #3437(error in line 447 was fixed by myself),
but result is the same as https://github.com/MarlinFirmware/Marlin/issues/3375#issuecomment-207502131.
if I use the "Move E" with unhomed delta machine, XYZ axes rises leisurely.

thinkyhead, 2016-04-09:
@esenapaj Good to know. So it doesn't seem to be a "precision" issue with the encoder value, at least. Now I have to look at the code to see what could possibly cause this behavior. Perhaps it has something to do with the conversion between cartesian and delta coordinates — but in that case, I would expect the same behavior with a homed delta.

When you see this behavior (with an unhomed delta) are all the steppers at the same height, and the nozzle in the middle? Or does this behavior only occur if the nozzle is off-center when you try to move E?

thinkyhead, 2016-04-09:
Here's something you can do to help diagnose the problem, which might be due to precision issues. Un-comment the debug lines in the function `calculate_delta` in `Marlin_main.cpp` (around line 6855). This will show the cartesian input and the delta result. It might show that something "interesting" is going on.

esenapaj, 2016-04-10:
I've tested with newest RCBugFix(and uncomment lines in calculate_delta), Result is here.
```
18:17:03.176 : wait
18:17:03.237 : N26 M105*19
18:17:03.244 : ok T:25.5 /0.0 B:25.7 /0.0 B@:0 @:0
18:17:03.578 : cartesian x=0.00 y=0.00 z=0.00
18:17:03.578 : delta a=195.43 b=195.43 c=195.43
18:17:03.677 : cartesian x=0.00 y=0.00 z=0.00
18:17:03.680 : delta a=195.43 b=195.43 c=195.43
18:17:04.241 : wait
```
>When you see this behavior (with an unhomed delta) are all the steppers at the same height,
and the nozzle in the middle?

Yes, it is.
If I try to move E with an unhomed delta, XYZ axes rises leisurely at the same height.
Those took about 340 seconds to raised 195mm(about 304000steps), and It doesn't freeze. 

...oh, 195mm approximate 195.43 in the result...

esenapaj, 2016-04-10:
And, this is result of testing with homed delta. 
```
19:22:03.837 : wait
19:22:04.169 : cartesian x=0.00 y=0.00 z=338.00
19:22:04.169 : delta a=533.43 b=533.43 c=533.43
19:22:04.272 : cartesian x=0.00 y=0.00 z=338.00
19:22:04.272 : delta a=533.43 b=533.43 c=533.43
19:22:04.840 : wait
```

esenapaj, 2016-04-10:
I've tested more.
on condition that: G28 from Repetier-Host -> G1 X0 Y0 Z0 f1000 from Repetier-Host -> tried to move E from LCD
```
20:31:59.559 : wait
20:32:00.390 : cartesian x=0.00 y=0.00 z=0.00
20:32:00.394 : delta a=195.43 b=195.43 c=195.43
20:32:00.563 : wait
```
numerical values looks like same as testing with unhomed,
but E moved normally. XYZ didn't rised.

sarf2k4, 2016-04-10:
@thinkyhead , does this issue still around? I have the same problem on the last rcbugfix (probably last Tuesday).

After moving like the z axis> press to exit (success)> cursor on z axis (again)> press> move (success)> press to exit (unsuccessful). 

What I gotta do is to wait till the timeout has been reached to the info screen. This same goes for the babystepping z as well.

thinkyhead, 2016-04-11:
>this issue still around?

Not any more. The "Press to exit" issue was patched in `RCBugFix` a few days ago.

thinkyhead, 2016-04-11:
@esenapaj So it looks like an unhomed delta is presuming that the ABC steppers are all at a certain position. Maybe these defaults fail to correspond to the Cartesian XYZE position, which should be the middle of the bed (0, 0, 0, 0) at Z height zero (which is good, since it shouldn't move farther down before homing). Marlin should be using `calculate_delta` to initialize the ABC stepper positions, but only *after* it loads settings from EEPROM, which may include specially-configured rod lengths.

I'll have to just go through the delta initialization code and see how it gets those initial values.

thinkyhead, 2016-04-11:
So… What happens if you add this line to the `setup()` function in `Marlin_main.cpp`?
```cpp
sync_plan_position_delta();
```

esenapaj, 2016-04-11:
>So… What happens if you add this line to the setup() function in Marlin_main.cpp?
```
>sync_plan_position_delta();
```

I'm feeling that it's magical fixing, and You've a marvelous discerning eye.
I've tested a unhomed delta with [forked RCBugFix](https://github.com/esenapaj/Marlin/tree/now-xyz-does-not-rizing-with-unhomed-delta), E moved normally. XYZ didn't rised.
```
15:02:55.900 : wait
15:02:56.736 : cartesian x=0.00 y=0.00 z=0.00
15:02:56.736 : delta a=195.43 b=195.43 c=195.43
15:02:56.834 : cartesian x=0.00 y=0.00 z=0.00
15:02:56.838 : delta a=195.43 b=195.43 c=195.43
15:02:56.903 : wait
```
Moreover, X and Y and Z moved normally, too.
Now, "rising" issue is gone thoroughly.

esenapaj, 2016-04-11:
And about ["display goes weird and fills up with squares"](https://github.com/MarlinFirmware/Marlin/issues/3375#issuecomment-205534373),
This phenomenon is no longer with Moving XYZ,
but it seems that it still occur with ultra high stepping rates + USE_WATCHDOG + G29(and G28?).
I'm still searching about it.

thinkyhead, 2016-04-12:
> "rising" issue is gone thoroughly

As we go along, the bugs are becoming more "indirect" and tricky to figure out. But this is a common problem – forgetting to initialize related values. This is the second bug related to converting stepper ABC positions back into XYZ positions that I've had to fix in the last two days!

PR #3388 (2016-04-05): Fix: menu edit items saving position twice
The menu system got some updates recently to make it easier to go back to the previous position. As a result, menu **edit items** no longer need to save the previous menu position, but this change was not included in the recent patch. This PR removes the extra `lcd_save_previous_menu` to fix the issue.

This should fix #3375 and #3382
PR #3397 (2016-04-05): Revert #3388, clean up lcd_move code
This reverts #3388 (a blind leap). Also cleans up some of the code around the lcd_move functions. If Github has gotten through the push backlog, there should also be a commit adding audio feedback to some items that were getting two button pushes in quick succession (and failing to exit).

This should resolve #3375 and part of #3373.
closed 2016-04-25 by thinkyhead