MarlinFirmware/Marlin #3378: Using a LCD you can't print sd files with a number as first character

StephenRC, 2016-04-04:
Using a LCD, if you select a file that begins with a number, nothing happens.  The print never starts, no  bed temp is set and nothing else occurs.  Remove the number and the print starts.

0_PLA_something.gcode doesn't work.
_PLA_something.gcode does work.

Works fine in RC3 and RC1.

Roxy-3DPrintBoard, 2016-04-04:
I believe this was broken in RC-4 but should be working in RC-5.  Please download RC-5 and report back if you still have a problem.

CONSULitAS, 2016-04-04:
see #3246
merged 10 days ago

StephenRC, 2016-04-04:
It's working in RC5.  Didn't see the small gray comment RC5 released in the file list.  The section that shows the readme, only shows RC4.

closed 2016-04-04 by StephenRC

thinkyhead, 2016-04-06:
RC6 is coming soon. I'll remember to update the README this time.