MarlinFirmware/Marlin #3381: Issue with more than 1 extruder on RUMBA board

jlcdi, 2016-04-04:
Hi,

The following problem occurs when declaring more than 1 extruder on "configuration.h". This issue is not new and is lasting from various past versions:

In file included from Configuration_adv.h:601:0,
from Configuration.h:859,
from Marlin.h:26,
from blinkm.cpp:5:
SanityCheck.h:305: error: #error TEMP_1_PIN not defined for this board.
#error TEMP_1_PIN not defined for this board.
^
SanityCheck.h:313: error: #error HEATER_1_PIN not defined for this board.
#error HEATER_1_PIN not defined for this board.
^
SanityCheck.h:319: error: #error TEMP_SENSOR_1 is required with 2 or more EXTRUDERS.
#error TEMP_SENSOR_1 is required with 2 or more EXTRUDERS.
^
#error TEMP_1_PIN not defined for this board.

Thanks for your help.

thinkyhead, 2016-04-04:
I can't see any obvious issue in the code. These pins are defined in the `pins_RUMBA.h` file (under appropriate conditions). So before I go hunting further, a couple of questions…
- What value are you using for `TEMP_SENSOR_1`?
- What version of Arduino IDE are you using to compile?

fbarcenas, 2016-04-05:
I am using RUMBA and not problem, use the last RCBUGFIX and the last arduino 1.6.8

jlcdi, 2016-04-05:
I did not affect a value at TEMP_SENSOR_1. Should I ?

Blue-Marlin, 2016-04-05:
Yes, you should.
Head - Wall - bang - bang - bang.

jlcdi, 2016-04-05:
@Blue-Marlin

You are right.
I now found out. There was no values on TEMP_SENSOR_1, TEMP_SENSOR_2, TEMP_SENSOR_3.
Problem solved. I know what I will have to say to the ones who makes the same mistake, and I've seen many...

Thks a lot everybody for your help, especially @thinkyhead

closed 2016-04-05 by jlcdi