MarlinFirmware/Marlin #3382: Cannot exit the "Move Axis" Menu

jlcdi, 2016-04-04:
Hi,

On Marlin 1.1.0-RC5, with a 12864 Reprap Discount Graphic LCD after entering, it is not possible to get out of the "Move Axis" menu.

Thanks for your help.

thinkyhead, 2016-04-04:
@jlcdi Does this problem still exist in `RCBugFix`? We've been fixing bugs and updating code related to the LCD menus.

jbrazio, 2016-04-04:
@thinkyhead this may be a duplicate of #3375.

thinkyhead, 2016-04-05:
This one is probably fixed by #3388. But #3375 has some additional features that need a closer look.

fbarcenas, 2016-04-05:
In my case the problem continue with the last rcbugfix

jlcdi, 2016-04-05:
@thinkyhead I will give it a try. But, how to download the RCBugFix ? I can find manu files, but I do not see the configuration.h. Which files shall I use to test the last RCBugFix ? Is there a complete package I can download at a given moment ?

esenapaj, 2016-04-05:
@jicdi, link of newest RCBugFix complete package is here.
[image]
https://github.com/MarlinFirmware/Marlin/tree/RCBugFix
https://github.com/MarlinFirmware/Marlin/archive/RCBugFix.zip

jlcdi, 2016-04-05:
> [image: marlinfirmware_marlin at rcbugfix -
> github_com_marlinfirmware_marlin_tree_rcbugf - 027]

Understood !!!

Thks a lot.

*Jean-Luc COLONNA d'ISTRIA*
*Director*

*LE MONACO Residences*

*35 Soi Pahonyothin 11, Pahonyothin RoadSamsen Nai, PhayathaiBangkok 10400,
ThailandCell: (66) 8 7972 1828Email: jlcolonna.idts@gmail.com
<jlcolonna.idts@gmail.com>Web: www.le-monaco.com <http://www.le-monaco.com>*

* <http://www.le-monaco.com>SKYPE: bureaubkk*

jlcdi, 2016-04-05:
@thinkyhead

Just tried with the newest RCBugFix.

It is the same.

When in "Move X", no line of menu appears to go back. We have to wait the effect of the temporisation to bring us back to the main menu.

thinkyhead, 2016-04-05:
Sorry about the slow resolution. I'm still looking into this behavior. Let me look into it closer and maybe I will have some better questions and ideas to help figure this out.

esenapaj, 2016-04-06:
@jlcdi, the problem was fixed by @thinkyhead a while ago.
Please try to newest RCBugFix.

jlcdi, 2016-04-06:
@esenapaj
@thinkyhead
I confirm that this issue is still present with the last version of RCBugFix, downloaded 8 hours ago.

esenapaj, 2016-04-06:
Well, fixing done about 4 hours ago...

thinkyhead, 2016-04-06:
@jlcdi Sadly, I only have internet access for a few hours a day right now. So I'm doing a lot of "hit and run" posting of fixes. Hopefully I will have better access soon and be more available in more time-zones.

jlcdi, 2016-04-06:
@thinkyhead You are already doing great. I just tested with the last updated version (30 mns ago). Issue is still there.

fbarcenas, 2016-04-08:
i check with the last RCBUGFIX and everything is correct and solved!

thinkyhead, 2016-04-09:
Good news. I think it was fixed with the addition of clicky sounds. In the long run, this will be addressed in a better way (by implementing a proper "debounce" for LCD clicks). In the meantime, clicky sounds and the associated delay seem to do the trick!

closed 2016-04-09 by thinkyhead