MarlinFirmware/Marlin #3394: M851, does it work with MESH Leveling?

WheresWaldo, 2016-04-05:
This is more of a question, rather than a bug, per se. I am trying to figure out is M851 Z_Offset actually works with MESH. If I need to do a slight adjustment to the leveling grid is M851 does not work, how would you accomplish this?

On a side note, I noticed with MESH when you execute M851 all by itself your response is just an 'ok' and not any number you may have stored in EEPROM for that particular M code. This behavior is different than previous versions of RC's with ABL set. I have only tested MESH on RC4, RC5 and the current RCBugFix and M851 does not appear to do anything.

Blue-Marlin, 2016-04-05:
```
// Custom M code points
#define CUSTOM_M_CODES
#if ENABLED(CUSTOM_M_CODES)
  #if ENABLED(AUTO_BED_LEVELING_FEATURE)
    #define CUSTOM_M_CODE_SET_Z_PROBE_OFFSET 851
    #define Z_PROBE_OFFSET_RANGE_MIN -20
    #define Z_PROBE_OFFSET_RANGE_MAX 20
  #endif
#endif
```
No AUTO_BED_LEVELING_FEATURE -> No M851

epatel, 2016-04-05:
Mesh leveling has its own Z tuning offset (maybe not in RC yet)

epatel, 2016-04-05:
Looks like it is available in both RC and RCFix right now.

epatel, 2016-04-05:
In G-code it is `G29 S4 Z0.1` 

Write up https://github.com/epatel/Marlin/blob/75f9fb50c4002f6d4ba0345f5594cbac1fab1fa6/Documentation/MeshBedLeveling.md

WheresWaldo, 2016-04-05:
Perfect, thanks for the quick response. I will test it today.

closed 2016-04-05 by WheresWaldo

thinkyhead, 2016-04-05:
We might want to make "Set MBL Z Offset" into its own code (`M422`?) instead of piggybacking on `G29`. I think we should probably add a menu item for this too, since we now have a sub-menu for `Bed Leveling` anyway. And, as long as we're at it, we should have an "On/Off" switch in that sub-menu too.

WheresWaldo, 2016-04-06:
My thoughts exactly, Offset is offset is offset, why have different ways to express the same value just because a few configuration items are different. It was odd to see MESH use G29 when I expected it would use the already implemented M851. Please pick one and stay with it.

thinkyhead, 2016-04-07:
@WheresWaldo As I understand it, the intent of the "Z offset" in Mesh Bed Leveling is different from the "Z Probe Z Offset" as set by `M851`. As I understand it, the MBL "Z Offset" is meant to be a fine-grained correction across the whole mesh. In contrast, the `M851` offset is the Z distance measured from the tip of the nozzle to the "tip" of the deployed probe. So normally the MBL "Z offset" is zero. But the `M851` offset is usually a negative number, at least 2 (mm) or more. So really, we can't just use `M851` here.

WheresWaldo, 2016-04-07:
Perhaps programmatically they are all different. But to the end user they may appear the same as they perform similar functions. Thus end users might want to treat them the same with the same G/M-code. It was just a suggestion.

As an FYI my view is from the perspective of a Robo3D user. It's probe design triggering below the top surface of the bed causes the M851 value to always be positive not negative. Regardless, I am sure the community will be fine with whatever the final decision is.

thinkyhead, 2016-04-07:
> Its probe design triggering below the top surface of the bed...

Indeed, positive `M851` values are becoming more common due to these kinds of "probes." The `MECHANICAL_PROBE` (once added) will always have a positive offset, for example.