MarlinFirmware/Marlin #3395: Working on a documentation regarding configuration

sarf2k4, 2016-04-05:
Hi, I am working slowly on the documentation regarding the configuration in here https://github.com/MarlinFirmware/Marlin/wiki/Marlin-Configuration

Core teams please verify that the documentation are good to go or not and I personally think that someone has to start doing the documentations for a better explanation on how to configure marlin for their 3d printer

I'm looking forward to complete the documentation of marlin configuration

jbrazio, 2016-04-05:
Thank you for your contribution.
Please have a look at #3088 and [here](http://test.marlinfw.org) which is temporary hosted [here](https://github.com/jbrazio/MarlinDocumentation).

We do plan to gather a couple of keen users to bring the existing documentation up to date and build missing ones as an example one of the ideas is to use "more visual" queues like images and diagrams.

sarf2k4, 2016-04-05:
@jbrazio , I'm sorry that I commit the changes on the wiki documentation first before asking, I was somewhat eager on this one, should that be removed?

jbrazio, 2016-04-05:
Oh please do not ask sorry for ! The documentation situation is not yet crafted in stone, I just wanted to update you as of the current discussion status. You may proceed with your contributions on the wiki or use the "new temporary" system.

sarf2k4, 2016-04-05:
Thank you for the feedback, I will continue on the documentation
development and update on the one that I updated

thinkyhead, 2016-04-09:
Thanks so much @sarf2k4 – At this point just having the documentation in any form is a good start. We can always paste it into our final location later.

sarf2k4, 2016-04-12:
I practically finished most of the documentation, although there might be not covered because some of the parts are somewhat 'rarely' used so I didn't cover that part

jbrazio, 2016-04-12:
@sarf2k4 have a look at #3447, are you willing to help ?

sarf2k4, 2016-04-12:
I already updated the wiki page yesterday actually, I might be able to but unsure about the one on the new page, I might be going to disrupt the formatting  that you guys might prefer though

jbrazio, 2016-04-13:
Hello @sarf2k4 I did a migration into the new system, have a look:
http://www.marlinfw.org/articles/getting-started/configure.html

The file from which that page is generated is located [here](https://github.com/MarlinFirmware/MarlinDocumentation/blob/master/_articles/getting-started/configure.md).

jbrazio, 2016-05-04:
Thank you for your interest making Marlin better.
You should re-open this topic on [MarlinFirmware/MarlinDocumentation](https://github.com/MarlinFirmware/MarlinDocumentation) for proper followup from the development team. We suggest you to use the `MarlinFirmware/Marlin#<topic id>` markdown to link back to this original topic.

closed 2016-05-04 by jbrazio