MarlinFirmware/Marlin #3396: Auto bed leveling - insufficient adjustment

maxwondercorn, 2016-04-05:
My axis seem to be turning in the correct directions to compensate for bed level but not sufficiently to lay down a good first layer. As the photo shows in the Y+ direction the nozzle is too close and the X- is a bit further than it should be, Y- is perfect. I've moved the part around a bit just in case it's a flatness issue and the problems remain in the same directions.  This is plate glass so it should not be a flatness issue.

Cartesian bot and bed is 320x320. RCBugFix downloaded 13:00 EST

G29 P 5 V4 T output below.  Didn't include all the M111 output

< 1:43:52 PM: clean_up_after_endstop_move > ENDSTOPS_ONLY_FOR_HOMING > endstops_not_homing()
< 1:43:52 PM: Eqn coefficients: a: -0.00578718 b: 0.00621478 d: 0.92887535
< 1:43:52 PM: Mean of sampled points: 1.00353550
< 1:43:52 PM: planeNormal x: 0.005787 y: -0.006215 z: 1.000000
< 1:43:52 PM: set_bed_level_equation_lsq > current_position: (313.90, 248.11, 16.52)
< 1:43:52 PM:  
< 1:43:52 PM: Bed Height Topography: 
< 1:43:52 PM:    +--- BACK --+
< 1:43:52 PM:    |           |
< 1:43:52 PM:  L |    (+)    | R
< 1:43:52 PM:  E |           | I
< 1:43:52 PM:  F | (-) N (+) | G
< 1:43:52 PM:  T |           | H
< 1:43:52 PM:    |    (-)    | T
< 1:43:52 PM:    |           |
< 1:43:52 PM:    O-- FRONT --+
< 1:43:52 PM:  (0,0)
< 1:43:52 PM:  +1.39798 +1.08737 +0.75152 +0.39545 +0.23889
< 1:43:52 PM:  +0.92828 +0.63283 +0.50657 -0.01616 -0.23586
< 1:43:52 PM:  +0.71364 +0.14040 -0.12727 -0.53131 -0.75354
< 1:43:52 PM:  +0.29697 -0.11970 -0.48586 -0.85960 -0.93535
< 1:43:52 PM:  +0.28182 -0.30404 -0.58939 -1.10202 -1.31162
< 1:43:52 PM:  
< 1:43:52 PM: Corrected Bed Height vs. Bed Topology: 
< 1:43:52 PM:  +0.21389 +0.23894 +0.23873 +0.21832 +0.39741
< 1:43:52 PM:  +0.09842 +0.13862 +0.34800 +0.16093 +0.27688
< 1:43:52 PM:  +0.23800 +0.00042 +0.06839 +0.00000 +0.11343
< 1:43:52 PM:  +0.17555 +0.09454 +0.06403 +0.02594 +0.28583
< 1:43:52 PM:  +0.51463 +0.26442 +0.31472 +0.13774 +0.26380
< 1:43:52 PM:  
< 1:43:52 PM: Bed Level Correction Matrix:
< 1:43:52 PM: +0.999983 +0.000000 -0.005787 
< 1:43:52 PM: +0.000036 +0.999981 +0.006214 
< 1:43:52 PM: +0.005787 -0.006215 +0.999964 
< 1:43:52 PM: > BEFORE apply_rotation_xyz > z_tmp  = 16.52
< 1:43:52 PM: > BEFORE apply_rotation_xyz > real_z = 16.24
< 1:43:52 PM: > AFTER apply_rotation_xyz > z_tmp  = 16.63
< 1:43:52 PM: > corrected Z in G29: (313.90, 248.11, 16.59)
< 1:43:52 PM: <<< gcode_G29
< 1:43:52 PM: X:313.90 Y:248.11 Z:16.59 E:0.00 Count X: 50240 Y:39680 Z:6460

[image]

maxwondercorn, 2016-04-05:
This is RC3 which is better.  The nozzle is still a bit too close in Y+/X- corner but this will work.  Some part, same location as above. Forgot to run G29 with all the debug info.

< 2:05:03 PM: G29 Auto Bed Leveling
< 2:06:12 PM: Eqn coefficients: a: -0.00541023 b: 0.00644488 d: 1.29835617
< 2:06:12 PM: planeNormal x: 0.005410 y: -0.006445 z: 1.000000
< 2:06:12 PM:  
< 2:06:12 PM: Bed Level Correction Matrix:
< 2:06:12 PM: +0.999985 +0.000000 -0.005410 
< 2:06:12 PM: +0.000035 +0.999979 +0.006445 
< 2:06:12 PM: +0.005410 -0.006445 +0.999965 
  2:06:28 PM: Printing layer 1 of 48

[image]

CONSULitAS, 2016-04-05:
@maxwondercorn 
Did you see, that your extruder extrudes far to much material, so that you have huge "noses" on the flat plane? This will cause loose of steps and other problems!

The shifted layers show that your extruder nozzle crashes into material of the last layer and then the stepper motors loose steps, because they do not have the torque to move further. BTW: If they had, it could break your printer.

You have to calibrate your extruder urgently before you proceed finding other problems first:
* http://reprap.org/wiki/Triffid_Hunter%27s_Calibration_Guide (first to start)
* http://www.sailfishfirmware.com/doc/tuning.html
* https://github.com/alexrj/Slic3r/wiki/Calibration

Roxy-3DPrintBoard, 2016-04-05:
~~~~
1:43:52 PM: +1.39798 +1.08737 +0.75152 +0.39545 +0.23889
1:43:52 PM: +0.92828 +0.63283 +0.50657 -0.01616 -0.23586
1:43:52 PM: +0.71364 +0.14040 -0.12727 -0.53131 -0.75354
1:43:52 PM: +0.29697 -0.11970 -0.48586 -0.85960 -0.93535
1:43:52 PM: +0.28182 -0.30404 -0.58939 -1.10202 -1.31162
~~~~

2.7 mm of difference between the back left and front right corner is a lot.   You really ought to get that better leveled out before you hand the rest of the bed leveling to the firmware.

maxwondercorn, 2016-04-05:
I see that's it's trying to put too much material down - which I need to work on.  If you are talking about the shifting I think that you are just seeing my dirty bed.  The layer further away is another print I did earlier today.  I moved my print to see if that made a difference with the auto leveling.

maxwondercorn, 2016-04-06:
I can work on the leveling a bit.  So how do you read the matrix?  I've always been a bit fuzzy on the output.  

I see how you got 2.7 mm, so each of the points is the difference between 0 and the probe?  Looking at the points, that would mean the bed is higher in the back left, sloping forward and toward the right front corner?

thinkyhead, 2016-04-06:
What kind of results do you get with `M48`? Is the probe very reliable?

Roxy-3DPrintBoard, 2016-04-06:
> I see how you got 2.7 mm, so each of the points is the difference between 0 and the probe? Looking at the points, that would mean the bed is higher in the back left, sloping forward and toward the right front corner? Almost looks like it's not flat - is that an artifact?

The numbers in the matrix are relative to the mean of the probed points.  The display shows how much a given area of the bed is above or below the mean (of a perfectly leveled bed).

Probably your M48 numbers are OK.  But like ThinkyHead is asking...  It's important that you can trust them.  We need a Standard Deviation below .025 here for the qr_solve() algorithm to converge on a solution.

maxwondercorn, 2016-04-06:
M48 results. 3 test edge of bed toward center.  Center between the two z-screws so it is tighter as expected.

< 10:44:10 PM: Mean: 1.176263
< 10:44:10 PM: Standard Deviation: 0.092981

< 10:49:59 PM: Mean: 1.022222
< 10:49:59 PM: Standard Deviation: 0.034786

< 10:54:42 PM: Mean: 0.919444
< 10:54:42 PM: Standard Deviation: 0.018811

Roxy-3DPrintBoard, 2016-04-06:
That first point you measured is kind of high for a Standard Deviation.   Is there some 'slop' in your mechanicals when the the nozzle is positioned there?   

Are you using a micro-switch on a kick down probe leg?   Some thing is giving you a bigger number than you want here.

maxwondercorn, 2016-04-06:
So this is a 320 mm square bed guided by four corner rails and the screws
are in the center on each side. So the first point would be the furthest
from the z-screw.  Effectively it would a point on lever arm far from its
pivot point. The measurement point was probably 270 mm from the z-screw

This is a moving bed gantry and I'm using the BLTouch sensor which
incorporates the servo and sensor switch in one package.  It's really a
nice piece of kit.

The design is overconstrained and I believe the guide rails (v-slot) is
probably "holding back" the gantry as it rises to home.  I might be able to
loosen the wheel sets up a bit to reduce this.  Nice thing is I can use M48
to test my changes.

ydkimnz, 2016-04-08:
I also have same issue. I built my custom corexy printer and it's not working with RC4 and RC5. something wrong with x y coordinate but works fine with RC3.

maxwondercorn, 2016-04-08:
So I played with the wheel sets trying to improve the StDev and was able to improve it a bit.  Before making adjustments, I ran M48 several times at the same point and found that the results were not consistent.  StdDev ranged from the 0.09 down to 0.06 and generally showed improvement each time I ran M48.

The mechanics of this system far outweigh any effect that "warming" the bearing grease would have on the improvement,  so not sure what really was going on there.  Adjusting the wheel sets I was able to get StDev down to 0.04.  Each adjustment I made improved the StDev, so I kept adjusting until I lost the the low StDev.  Now I can't find that sweet spot to get it back down.

In the end, releveling the gantry as suggested by @Roxy-3DPrintBoard and raising my Z-offset by 0.1mm has me back to a good first layer in RC3.  I'm trying to catch up on prints since my printer has been down for two weeks (multiple issues) and will try reloading the latest RCBugfix next week.

closed 2016-04-25 by thinkyhead