MarlinFirmware/Marlin #3399: Auto Bed Level Probe Position Sanity Check Issue

controlsfreek, 2016-04-06:
It is possible that I am confused, so can someone please verify that this is not a bug:

Conditionals.h line 281 reads
 #define MAX_PROBE_X (min(X_MAX_POS, X_MAX_POS + X_PROBE_OFFSET_FROM_EXTRUDER))

I do not believe the minimum should be taken here. It is possible that the X axis could be at or very near maximum position, but the zprobe could be offset to the right of the extruder location. In this case the MAX_PROBE_X would simply be X_MAX_POS + X_PROBE_OFFSET_FROM_EXTRUDER. Otherwise the compiler will complain for a legitimate configuration.

I push my probe locations as far to the outside of the bed/axis range of motion as I can, and have run into this compile error.

This of course would apply to the Y axis as well.

Thanks for taking a look!

jbrazio, 2016-04-07:
I do believe the min() is correct as the probe offset can have negative values, in this case your bed X-max is no longer the probe limit but instead the X-max (minus) probe-offset.

thinkyhead, 2016-04-08:
@jbrazio is correct. Also remember the `X_MAX_POS` can be outside the bed. Useful if you have a probe to the left of the nozzle and want it to reach the edge of the bed.

closed 2016-04-08 by thinkyhead

controlsfreek, 2016-04-08:
Min may be useful if you have a negative probe offset- but in the case of positive offset, doesn't it prevent the probe x max from being right of X_MAX?

thinkyhead, 2016-04-09:
The `MAX_PROBE_X` is the right-most point on the bed that your probe can probe. So let's have a look at the two possible cases.
#### 1. Probe on the right
```cpp
#define X_MIN_POS 0 // left-most point that the nozzle can reach
#define X_MAX_POS 200 // right-most point that the nozzle can reach
#define X_PROBE_OFFSET_FROM_EXTRUDER 10
```
In this case `min(200, 210)` gives us `200`. So, this probe can reach and probe the right of the bed where X = 200. For the lower limit we have `max(0, 10)` which gives us `10`. So this probe can probe as far left as X = 10.
#### 2. Probe on the left
```cpp
#define X_MIN_POS 0 // left-most point that the nozzle can reach
#define X_MAX_POS 200 // right-most point that the nozzle can reach
#define X_PROBE_OFFSET_FROM_EXTRUDER -10
```
In this case `min(200, 190)` gives us `190`. Since the right-most point that the nozzle can reach is `200` and the probe is `10` mm to the left of that point, with these settings the probe can only reach and probe as far as X = 190. For the lower limit we have `max(0, -10)` which gives us `0`. So this probe can probe as far left as X = 0.

Remember that `X_MIN_POS` and `X_MAX_POS` do not refer to the left and right edges of your bed, but rather to the farthest points that your carriage can physically move. If a carriage can actually move farther to the left or right, past the edges of the bed, then `X_MIN_POS` and `X_MAX_POS` should be adjusted to permit that range of movement (at least, if a probe exists).

Following this guide, in order to be able to probe all the way on the right side of the bed (X = 200) with a probe `-10` mm from the nozzle, `X_MAX_POS` should be set to `210` instead of `200`. Likewise, if the probe is right of the nozzle, then it makes the most sense to put a Max X Endstop on the right and allow movement past the edge of the bed on the left with `#define X_MIN_POS -10`.