MarlinFirmware/Marlin #3401: one nozzle dual extruder

Ky1duck, 2016-04-06:
not sure if this is the right place but dose marlin support one nozzle dual extruder yet. 

thanks

Roxy-3DPrintBoard, 2016-04-06:
I don't have one....  But "Yes!".   I think if I had one, I would put some extra GCode to purge the nozzle (in a safe place) every time I switched colors.

Ky1duck, 2016-04-08:
yea but thr problem is I cant find anywhere to tell marlin to use the same heater and temp sensor for both extruders.

thinkyhead, 2016-04-08:
@Ky1duck You should use the MarlinKimbra version of Marlin, which has a `SINGLENOZZLE` option to do what you want. We have a "mixing extruder" feature pending in our development branch which accomplishes the same thing, but it's not yet merged.

woodsmoke, 2016-04-29:
This is a feature I'd love to see too.

woodsmoke, 2016-04-29:
Is it as simple as duplicating the pin number for both TEMP_0_PIN/HEATER_0_PIN and TEMP_1_PIN/HEATER_1_PIN? It has compiled and uploaded ok. But T1 just reads 0.00

woodsmoke, 2016-04-29:
Not quite duplicated, and the heaters won't respond. So no it isn't as simple as that.

closed 2016-05-04 by jbrazio