MarlinFirmware/Marlin #3402: Which REPRAPWORLD_KEYPAD code is good REPRAPWORLD_KEYPAD code?

thinkyhead, 2016-04-06:
There are two places in the code where things are being defined for `REPRAPWORLD_KEYPAD`. Which code is the "most correct" code?

- https://github.com/MarlinFirmware/Marlin/blob/RC/Marlin/ultralcd.h#L97-L115
- https://github.com/MarlinFirmware/Marlin/blob/RC/Marlin/ultralcd_implementation_hitachi_HD44780.h#L100-L127

esenapaj, 2016-04-06:
I haven't knowledge about [the Keypad](http://reprap.org/wiki/Keypad), but I tried to trace commit log.

ultralcd_implementation_hitachi_HD44780.h:
https://github.com/MarlinFirmware/Marlin/commit/6a81291c570c356dc0a4154d75efc204cf9f6b04 ...initial commit? on 8 Jun 2013, by @kiyoshigawa(merged by @ErikZalm)
https://github.com/MarlinFirmware/Marlin/commit/df4d50579a07eb61159e80eea30637b72103e494 ...settings was changed on 5 May 2015, by @brupje

 ultralcd.h:
https://github.com/MarlinFirmware/Marlin/commit/a5be0c52c6eade234253d26991b87f7f3e43460b ...initial commit? on 14 Jul 2013, by @ErikZalm

thinkyhead, 2016-04-08:
#3428 moves everything into the `ultralcd.h` file. As with the "Move Axis" commands on LCD, I've also disabled XYZ movement for Delta/SCARA until the machine has been properly homed.

closed 2016-04-25 by thinkyhead